LORD-MicroStrain/microstrain_inertial #2: 3DM-GX5-25 logging issues

UnalA, 2020-03-17:
I am trying to use the new mscl driver to log 3DM-GX5-25 data. I try to bag the data at 10 Hz yet the logged is nowhere near 10 Hz (more like 3-4 Hz). When I try to log faster I get even worse results. Older IMUs (3DM-GX3-25) worked really well the previous interface and drivers. Is this something you have noticed? Is there something in the launch file I need to be setting to get consistent timestamps?

rdslord, 2020-04-27:
Thank you for reaching out about this. If you update MSCL and get the latest version of this driver, you should be able to log data much faster.

rdslord, 2020-05-05:
I am closing this issue as I believe the problem has been addressed in the update. In testing, I have been able to log data at 10 Hz with no problems now. If you have any additional problems, please reach out to us or open a new issue. Thanks!

closed 2020-05-05 by rdslord