LORD-MicroStrain/microstrain_inertial #4: 3DM-CV5-25

isValid, 2020-04-16:
In the process of using, I encountered two problems.

(1) I need quaternion data , The data output by this driver seems to be the original data?  How to modify can ooutput filtered quaternion?  I made some modification to the program,  then i can get data, but I'm not sure it is true.
### if (publish_imu_)
      {
        ROS_INFO("Setting AHRS active channel fields");
        mscl::SampleRate imuRate = mscl::SampleRate::Hertz(imu_rate_);

        mscl::MipTypes::MipChannelFields ahrsChannels{
            mscl::MipTypes::ChannelField::CH_FIELD_SENSOR_SCALED_ACCEL_VEC,
            mscl::MipTypes::ChannelField::CH_FIELD_SENSOR_SCALED_GYRO_VEC,
            //mscl::MipTypes::ChannelField::CH_FIELD_SENSOR_ORIENTATION_QUATERNION        
      mscl::MipTypes::ChannelField::CH_FIELD_ESTFILTER_ESTIMATED_ORIENT_QUATERNION
        };
        mscl::MipChannels supportedChannels;
        //for (mscl::MipTypes::ChannelField channel : msclInertialNode->features().supportedChannelFields(mscl::MipTypes::DataClass::CLASS_AHRS_IMU))
        for (mscl::MipTypes::ChannelField channel : msclInertialNode->features().supportedChannelFields(mscl::MipTypes::DataClass::CLASS_ESTFILTER))
        {
          if (std::find(ahrsChannels.begin(), ahrsChannels.end(), channel) != ahrsChannels.end())
          {
            supportedChannels.push_back(mscl::MipChannel(channel, imuRate));
          }
        }
        //inertialNode.setActiveChannelFields(mscl::MipTypes::DataClass::CLASS_AHRS_IMU, supportedChannels);
        inertialNode.setActiveChannelFields(mscl::MipTypes::DataClass::CLASS_ESTFILTER, supportedChannels);
        if (features.supportsCommand(mscl::MipTypes::Command::CMD_EF_DECLINATION_SRC))
        {
          ROS_INFO("Setting Declination Source, %d", declination_source);
          inertialNode.setDeclinationSource(mscl::GeographicSourceOptions(static_cast<mscl::InertialTypes::GeographicSourceOption>(declination_source_u8),declination));
        }
        //inertialNode.enableDataStream(mscl::MipTypes::DataClass::CLASS_AHRS_IMU);
        inertialNode.enableDataStream(mscl::MipTypes::DataClass::CLASS_ESTFILTER);
      }

### void Microstrain::parseMipPacket(const mscl::MipDataPacket &packet)
       {
         switch (packet.descriptorSet())
           {
             case mscl::MipTypes::DataClass::CLASS_AHRS_IMU:
               parseSensorPacket(packet);
               print_packet_stats();
               break;

             case mscl::MipTypes::DataClass::CLASS_ESTFILTER:
               parseSensorPacket(packet);
               parseEstFilterPacket(packet);
               print_packet_stats();
               break;

             case mscl::MipTypes::DataClass::CLASS_GNSS:
               parseGnssPacket(packet);
               print_packet_stats();
               break;

             default:
               break;
             }
         }

### Orientation Quaternion
    //case mscl::MipTypes::CH_FIELD_SENSOR_ORIENTATION_QUATERNION:
    case mscl::MipTypes::CH_FIELD_ESTFILTER_ESTIMATED_ORIENT_QUATERNION:
    {
      hasOrientationQuat = true;
      if (point.qualifier() == mscl::MipTypes::CH_QUATERNION)
      {
        mscl::Vector quaternion = point.as_Vector();
        curr_filter_quaternion_ = quaternion;
        // put into ENU - swap X/Y, invert Z
        imu_msg_.orientation.x = quaternion.as_floatAt(2);
        imu_msg_.orientation.y = quaternion.as_floatAt(1);
        imu_msg_.orientation.z = -1.0 * quaternion.as_floatAt(3);
        imu_msg_.orientation.w = quaternion.as_floatAt(0);
      }
    }
    break;
    }

(2) I need to turn off the magnetometer, i change launch file, but this setting does not take effect.
### 
    <param name="declination_source" value="1" type="int" />
    <param name="declination" value="0.23" type="double" />
How to modify?

rdslord, 2020-04-22:
Hello! If you download the latest version of the driver, filtered quaternion data should be published to the /filtered/imu/data topic as it was in the microstrain_mips driver. 

The declination and declination source parameters do not change the heading source. I will add a parameter for heading source.

Thank you for letting us know!

isValid, 2020-04-23:
Okay, thanks, looking forward to the version update

isValid, 2020-04-23:
> Hello! If you download the latest version of the driver, filtered quaternion data should be published to the /filtered/imu/data topic as it was in the microstrain_mips driver.
>
> The declination and [...]

Sorry, I downloaded the latest driver. 
However, during the compilation process, many problems occurred, such as 

"microstrain_3dm.cpp: 756: 29: error: expression cannot be used as a function
biasVector.x (), biasVector.y (), biasVector.z ()); "

microstrain_3dm.cpp:2402:37: error: ‘const class mscl::MipNodeFeatures’ has no member named ‘supportedStatusSelectors’
if(msclInertialNode->features().supportedStatusSelectors().size() > 1)

isValid, 2020-04-23:
In file included from /opt/ros/kinetic/include/ros/ros.h:40:0,
                 from /home/jack/catkin_ws/src/ROS-MSCL-master/ros_mscl/include/ros_mscl/microstrain_diagnostic_updater.h:1,
                 from /home/jack/catkin_ws/src/ROS-MSCL-master/ros_mscl/src/microstrain_3dm.cpp:16:
/home/jack/catkin_ws/src/ROS-MSCL-master/ros_mscl/src/microstrain_3dm.cpp: In member function ‘bool Microstrain::Microstrain::set_accel_bias(ros_mscl::SetAccelBias::Request&, ros_mscl::SetAccelBias::Response&)’:
/home/jack/catkin_ws/src/ROS-MSCL-master/ros_mscl/src/microstrain_3dm.cpp:552:29: error: expression cannot be used as a function
                biasVector.x(), biasVector.y(), biasVector.z());
                             ^
/opt/ros/kinetic/include/ros/console.h:346:165: note: in definition of macro ‘ROSCONSOLE_PRINT_AT_LOCATION_WITH_FILTER’
     ::ros::console::print(filter, __rosconsole_define_location__loc.logger_, __rosconsole_define_location__loc.level_, __FILE__, __LINE__, __ROSCONSOLE_FUNCTION__, __VA_ARGS__)
                                                                                                                                                                     ^
/opt/ros/kinetic/include/ros/console.h:379:7: note: in expansion of macro ‘ROSCONSOLE_PRINT_AT_LOCATION’
       ROSCONSOLE_PRINT_AT_LOCATION(__VA_ARGS__); \
       ^
/opt/ros/kinetic/include/ros/console.h:561:35: note: in expansion of macro ‘ROS_LOG_COND’
 #define ROS_LOG(level, name, ...) ROS_LOG_COND(true, level, name, __VA_ARGS__)
                                   ^
/opt/ros/kinetic/include/rosconsole/macros_generated.h:110:23: note: in expansion of macro ‘ROS_LOG’
 #define ROS_INFO(...) ROS_LOG(::ros::console::levels::Info, ROSCONSOLE_DEFAULT_NAME, __VA_ARGS__)
                       ^
/home/jack/catkin_ws/src/ROS-MSCL-master/ros_mscl/src/microstrain_3dm.cpp:551:7: note: in expansion of macro ‘ROS_INFO’
       ROS_INFO("Accel bias vector values are: %f %f %f",
       ^
/home/jack/catkin_ws/src/ROS-MSCL-master/ros_mscl/src/microstrain_3dm.cpp:552:45: error: expression cannot be used as a function
                biasVector.x(), biasVector.y(), biasVector.z());
                                             ^
/opt/ros/kinetic/include/ros/console.h:346:165: note: in definition of macro ‘ROSCONSOLE_PRINT_AT_LOCATION_WITH_FILTER’
     ::ros::console::print(filter, __rosconsole_define_location__loc.logger_, __rosconsole_define_location__loc.level_, __FILE__, __LINE__, __ROSCONSOLE_FUNCTION__, __VA_ARGS__)
                                                                                                                                                                     ^
/opt/ros/kinetic/include/ros/console.h:379:7: note: in expansion of macro ‘ROSCONSOLE_PRINT_AT_LOCATION’
       ROSCONSOLE_PRINT_AT_LOCATION(__VA_ARGS__); \
       ^
/opt/ros/kinetic/include/ros/console.h:561:35: note: in expansion of macro ‘ROS_LOG_COND’
 #define ROS_LOG(level, name, ...) ROS_LOG_COND(true, level, name, __VA_ARGS__)
                                   ^
/opt/ros/kinetic/include/rosconsole/macros_generated.h:110:23: note: in expansion of macro ‘ROS_LOG’
 #define ROS_INFO(...) ROS_LOG(::ros::console::levels::Info, ROSCONSOLE_DEFAULT_NAME, __VA_ARGS__)
                       ^
/home/jack/catkin_ws/src/ROS-MSCL-master/ros_mscl/src/microstrain_3dm.cpp:551:7: note: in expansion of macro ‘ROS_INFO’
       ROS_INFO("Accel bias vector values are: %f %f %f",
       ^
/home/jack/catkin_ws/src/ROS-MSCL-master/ros_mscl/src/microstrain_3dm.cpp:552:61: error: expression cannot be used as a function
                biasVector.x(), biasVector.y(), biasVector.z());
                                                             ^
/opt/ros/kinetic/include/ros/console.h:346:165: note: in definition of macro ‘ROSCONSOLE_PRINT_AT_LOCATION_WITH_FILTER’
     ::ros::console::print(filter, __rosconsole_define_location__loc.logger_, __rosconsole_define_location__loc.level_, __FILE__, __LINE__, __ROSCONSOLE_FUNCTION__, __VA_ARGS__)
                                                                                                                                                                     ^
/opt/ros/kinetic/include/ros/console.h:379:7: note: in expansion of macro ‘ROSCONSOLE_PRINT_AT_LOCATION’
       ROSCONSOLE_PRINT_AT_LOCATION(__VA_ARGS__); \
       ^
/opt/ros/kinetic/include/ros/console.h:561:35: note: in expansion of macro ‘ROS_LOG_COND’
 #define ROS_LOG(level, name, ...) ROS_LOG_COND(true, level, name, __VA_ARGS__)
                                   ^
/opt/ros/kinetic/include/rosconsole/macros_generated.h:110:23: note: in expansion of macro ‘ROS_LOG’
 #define ROS_INFO(...) ROS_LOG(::ros::console::levels::Info, ROSCONSOLE_DEFAULT_NAME, __VA_ARGS__)
                       ^
/home/jack/catkin_ws/src/ROS-MSCL-master/ros_mscl/src/microstrain_3dm.cpp:551:7: note: in expansion of macro ‘ROS_INFO’
       ROS_INFO("Accel bias vector values are: %f %f %f",
       ^
/home/jack/catkin_ws/src/ROS-MSCL-master/ros_mscl/src/microstrain_3dm.cpp:556:30: error: expression cannot be used as a function
       biasVector.x(req.bias.x);
                              ^
/home/jack/catkin_ws/src/ROS-MSCL-master/ros_mscl/src/microstrain_3dm.cpp:557:30: error: expression cannot be used as a function
       biasVector.y(req.bias.y);
                              ^
/home/jack/catkin_ws/src/ROS-MSCL-master/ros_mscl/src/microstrain_3dm.cpp:558:30: error: expression cannot be used as a function
[... cut ...]
                                     ^
ROS-MSCL-master/ros_mscl/CMakeFiles/ros_mscl.dir/build.make:62: recipe for target 'ROS-MSCL-master/ros_mscl/CMakeFiles/ros_mscl.dir/src/microstrain_3dm.cpp.o' failed
make[2]: *** [ROS-MSCL-master/ros_mscl/CMakeFiles/ros_mscl.dir/src/microstrain_3dm.cpp.o] Error 1
CMakeFiles/Makefile2:5796: recipe for target 'ROS-MSCL-master/ros_mscl/CMakeFiles/ros_mscl.dir/all' failed
make[1]: *** [ROS-MSCL-master/ros_mscl/CMakeFiles/ros_mscl.dir/all] Error 2
Makefile:138: recipe for target 'all' failed

rdslord, 2020-04-23:
It looks like you also need to update [MSCL](https://github.com/LORD-MicroStrain/MSCL).

isValid, 2020-04-24:
Before compiling the new program, I have installed MSCL version 55.01. Is there a newer version?

rdslord, 2020-04-24:
55.0.1 is the most up to date version. If you are unsure which version is running, remove your old version and verify the path and version number of your new install.

isValid, 2020-04-26:
After re-installing and uninstalling, I checked the version of MSCL
command: '' dpkg -l | grep mscl''
print: '' ii  c++-mscl  55.0.1  amd64  Installation of MSCL''
There are still problems compiling the driver again. The previous version of the driver is no problem.

rdslord, 2020-04-26:
Are they the same problems as before?

isValid, 2020-04-27:
Yes, the same problem as before

isValid, 2020-04-27:
And about the addition of switch parameters, I look forward to your update

rdslord, 2020-04-27:
There is now a parameter for heading source in the launch file so that you can choose whether or not to use the internal mag.

rdslord, 2020-04-27:
These issues:

> "microstrain_3dm.cpp: 756: 29: error: expression cannot be used as a function
> biasVector.x (), biasVector.y (), biasVector.z ()); "

and

> microstrain_3dm.cpp:2402:37: error: ‘const class mscl::MipNodeFeatures’ has no member named ‘supportedStatusSelectors’
> if(msclInertialNode->features().supportedStatusSelectors().size() > 1)

are the result of an outdated version of mscl. The function "supportedStatusSelectors" as well as the accessor functions called on biasVector were added in 55.0.1. 

Running the latest version of this driver with an outdated version of mscl will result in those errors.

isValid, 2020-04-28:
Thank you for updating the driver. I downloaded MSCL again and encountered new problems during compilation:

microstrain_3dm.cpp:1167:22: error: expected unqualified-id before ‘[’ token 
     for (const auto& [key, value] : status)  
microstrain_3dm.cpp:1167:22: error: expected ‘;’ before ‘[’ token
microstrain_3dm.cpp:1167:23: error: ‘key’ was not declared in this scope
     for (const auto& [key, value] : status)
microstrain_3dm.cpp:1167:28: error: ‘value’ was not declared in this scope
     for (const auto& [key, value] : status)
...
microstrain_3dm.cpp:1221:22: error: expected unqualified-id before ‘[’ token
     for (const auto& [key, value] : status)
microstrain_3dm.cpp:1221:22: error: expected ‘;’ before ‘[’ token
...
microstrain_3dm.cpp:2335:24: error: expected unqualified-id before ‘[’ token
       for (const auto& [key, value] : status)
...

This seems to be a problem with the driver?

rdslord, 2020-04-28:
Structured bindings are a c++17 feature. You may need to update gcc if you are not using a version that supports c++17.

rdslord, 2020-05-05:
I'm going to close this issue as the original two requests have been added and resolved. Please reach out if you have other problems and, of course, open another issue if you encounter more issues or have other requests.

closed 2020-05-05 by rdslord