LORD-MicroStrain/microstrain_inertial #7: The validity of the parameter heading_source

isValid, 2020-06-02:
When I set Heading Aiding to None in MIP Monitor, I will use ros_mscl to drive gx5-25. Of course,
the heading_source in the launch file is set to 1.When I went to check the imu data, I felt that the magnetometer was turned on, so I put gx5-25 in the MIP Monitor and found that Heading Aiding became Magnetometer. So I think the validity of Heading_source is suspect.
So can you check this question.
Thank you.

rdslord, 2020-06-02:
To use no heading aid, the heading_source value should be set to 0 in your launch file. Thank you for drawing my attention to the misleading comment in the code that said that 1 was none and 2 was mag. It has been updated now.

isValid, 2020-06-03:
ok,thanks.
I found a new problem in the test. The data before filtering (/imu/data) takes a few seconds to stabilize after the IMU rotates, so I chose the data after filtering (/filtered/imu/data), However, the data release frequency after filtering is only 10 Hz, and it seems that the release frequency cannot be modified. Can you increase the data release frequency after filtering?

closed 2020-06-04 by isValid

rdslord, 2020-06-04:
There is a parameter in the launch file called nav_rate. To change the rate of the filtered data, change that value in the launch file.