LORD-MicroStrain/microstrain_inertial #10: device connection issues

jwrl7, 2020-06-15:
I was able to get gx5-45 up and running using a laptop, MSCL c++ pre-built x64 library and ros_mscl. 
I am now trying to repeat this; connecting the unit to a nvidia xavier but ran into trouble with the ros_mscl failing to open the device. 

Essentially driver reports that the device is disconnected. The unit itself is rapidly blinking green indicating that it is streaming without GPS which is true since its inside.
I installed the MSCL library with the pe-built arm64 debian which I'm suspecting might be the problem?

I also tried using different USB ports and a powered USB hub. No luck. 

At first glance it seems this: https://github.com/LORD-MicroStrain/ROS-MSCL/blob/master/ros_mscl/src/microstrain_3dm.cpp#L132 might be where its failing. 

Left is the output from ros_mscl. Right is the output from `dmesg`
[image]

mglord, 2020-06-16:
I'm not familiar with the Nvidia Xavier specifically, but it looks like it has an ARM 64-bit processor and is most likely running a debian-based linux distro, so that should be the correct MSCL pre-built package.

It seems like MSCL is being included correctly as the error is not related to mscl::Connection being undefined, but the error while attempting to connect could be an issue with Boost. It's probably there, but just to double check can you confirm that this directory exists:
`/usr/share/c++-mscl/Boost/include`

Another common problem is lack of permissions on the port - please double check that your user is part of the dialout group or otherwise has permissions to access the port.

Just to rule it out could you try removing the realpath() conversion on the port name from the mscl::Connection line and/or including it in the ROS_INFO call directly above:

`ROS_INFO("Attempting to open serial port <%s> at <%d> \n", realpath(port.c_str(), NULL), baudrate);`
and/or
`mscl::Connection connection = mscl::Connection::Serial(port.c_str(), baudrate);`

It's a fairly new addition to support symlinks and it would be worth double checking that it's resolving correctly in your environment.

Let me know how these go and we can troubleshoot further from there!

jwrl7, 2020-06-19:
Resolved: added `dialout` to the user group
Thanks!

closed 2020-06-19 by jwrl7