LORD-MicroStrain/microstrain_inertial #11: Can't get it to work with 3DM-GX5-10

civerachb-cpr, 2020-06-22:
I'm sure there's a simple solution, but I can't get this ROS driver to work with my 3DM-GX5-10.  It works with the -15, -25, and -45 just fine though.

When I run `roslaunch ros_mscl microstrain.launch` I get the following output:

`$ roslaunch ros_mscl microstrain.launch 
... logging to /home/USERNAME/.ros/log/a47ad57c-b4a7-11ea-bccd-8c04ba888725/roslaunch-CPR01954L-4493.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://CPR01954L:45471/

SUMMARY
========

PARAMETERS
 * /gx5/ros_mscl_node/auto_init: True
 * /gx5/ros_mscl_node/baudrate: 115200
 * /gx5/ros_mscl_node/declination: 0.23
 * /gx5/ros_mscl_node/declination_source: 2
 * /gx5/ros_mscl_node/device_setup: True
 * /gx5/ros_mscl_node/dynamics_mode: 1
 * /gx5/ros_mscl_node/gps_frame_id: navsat_link
 * /gx5/ros_mscl_node/gps_rate: 4
 * /gx5/ros_mscl_node/heading_source: 1
 * /gx5/ros_mscl_node/imu_angular_cov: [0.01, 0, 0, 0, 0...
 * /gx5/ros_mscl_node/imu_frame_id: gx5_link
 * /gx5/ros_mscl_node/imu_linear_cov: [0.01, 0, 0, 0, 0...
 * /gx5/ros_mscl_node/imu_orientation_cov: [0.01, 0, 0, 0, 0...
 * /gx5/ros_mscl_node/imu_rate: 100
 * /gx5/ros_mscl_node/nav_rate: 10
 * /gx5/ros_mscl_node/odom_child_frame_id: base_link
 * /gx5/ros_mscl_node/odom_frame_id: wgs84_odom_link
 * /gx5/ros_mscl_node/port: /dev/ttyACM0
 * /gx5/ros_mscl_node/publish_imu: True
 * /gx5/ros_mscl_node/readback_settings: True
 * /gx5/ros_mscl_node/save_settings: True
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/expected: ['ros_mscl_node: ...
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/imu/expected: ['ros_mscl_node: ...
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/imu/path: imu
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/imu/remove_prefix: ros_mscl_node
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/imu/timeout: 5.0
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/imu/type: diagnostic_aggreg...
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/packet/expected: ['ros_mscl_node: ...
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/packet/path: packet
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/packet/remove_prefix: ros_mscl_node
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/packet/timeout: 5.0
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/packet/type: diagnostic_aggreg...
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/path: general
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/port/expected: ['ros_mscl_node: ...
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/port/path: port
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/port/remove_prefix: ros_mscl_node
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/port/timeout: 5.0
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/port/type: diagnostic_aggreg...
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/remove_prefix: ros_mscl_node
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/timeout: 5.0
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/type: diagnostic_aggreg...
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/path: ros_mscl_node
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/type: diagnostic_aggreg...
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    imu_diagnostic_aggregator (diagnostic_aggregator/aggregator_node)
  /gx5/
    ros_mscl_node (ros_mscl/ros_mscl_node)

auto-starting new master
process[master]: started with pid [4507]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to a47ad57c-b4a7-11ea-bccd-8c04ba888725
process[rosout-1]: started with pid [4521]
started core service [/rosout]
process[gx5/ros_mscl_node-2]: started with pid [4538]
process[imu_diagnostic_aggregator-3]: started with pid [4539]
[ INFO] [1592844270.702135696]: Attempting to open serial port </dev/ttyACM0> at <115200> 

terminate called after throwing an instance of 'std::logic_error'
  what():  basic_string::_M_construct null not valid
[gx5/ros_mscl_node-2] process has died [pid 4538, exit code -6, cmd /home/USERNAME/Code/microstrain_ws/devel/lib/ros_mscl/ros_mscl_node __name:=ros_mscl_node __log:=/home/USERNAME/.ros/log/a47ad57c-b4a7-11ea-bccd-8c04ba888725/gx5-ros_mscl_node-2.log].
log file: /home/USERNAME/.ros/log/a47ad57c-b4a7-11ea-bccd-8c04ba888725/gx5-ros_mscl_node-2*.log`

I get this same error on both Kinetic & Melodic.  The serial port it's attempting to open (/dev/ttyACM0) doesn't exist on my machine.  Plugging the device into the USB port and running dmesg yeilds:

`[ 2297.846737] usb 1-3: new full-speed USB device number 6 using xhci_hcd
[ 2297.974752] usb 1-3: device descriptor read/64, error -71
[ 2298.210750] usb 1-3: device descriptor read/64, error -71
[ 2298.446736] usb 1-3: new full-speed USB device number 7 using xhci_hcd
[ 2298.574756] usb 1-3: device descriptor read/64, error -71
[ 2298.810754] usb 1-3: device descriptor read/64, error -71
[ 2298.918799] usb usb1-port3: attempt power cycle
[ 2299.570744] usb 1-3: new full-speed USB device number 8 using xhci_hcd
[ 2299.570896] usb 1-3: Device not responding to setup address.
[ 2299.778897] usb 1-3: Device not responding to setup address.
[ 2299.986740] usb 1-3: device not accepting address 8, error -71
[ 2300.114742] usb 1-3: new full-speed USB device number 9 using xhci_hcd
[ 2300.114893] usb 1-3: Device not responding to setup address.
[ 2300.322888] usb 1-3: Device not responding to setup address.
[ 2300.530746] usb 1-3: device not accepting address 9, error -71
[ 2300.530826] usb usb1-port3: unable to enumerate USB device`

Is this a hardware problem with my sensor?  Or a problem with the udev rule included in the debian folder of the repository?  Or does this driver simply not work with the GX5-10?

mglord, 2020-06-22:
The GX5-10 doesn't support communication via USB - you will need to connect via serial port or use an RS232 to USB converter. Please feel free to reach out to Barry Trutor ([barry.trutor@parker.com](barry.trutor@parker.com)) in support for more specific information about setup configuration or where to get the right cables.

Sorry for the confusion!

civerachb-cpr, 2020-06-22:
Ah, good to know! Thanks for clearing that up.

civerachb-cpr, 2020-06-23:
I've found dsub to serial + serial to USB cables and everything appears to be working as-expected now.  Mildly irritating that the GX5-10 doesn't natively support USB, but I guess that's the trade-off for the price.

closed 2020-06-23 by civerachb-cpr