LORD-MicroStrain/microstrain_inertial #12: Hosting on buildfarm/deb server?

civerachb-cpr, 2020-06-24:
Are there any plans to host this driver on the ROS buildfarm and/or a custom deb server that can be added to sources.list to make redistribution easier?  Being able to specify this driver as a ROS dependency and being able to install it easily with `rosdep install [...]` (and have that pull in the MSCL SDK packages) would be great, rather than needing to build it from source all the time.

civerachb-cpr, 2020-08-04:
This issue is related to #14, as rosdep fails to find the libmscl dependency on Melodic.

civerachb-cpr, 2021-02-19:
In case anyone stumbles upon this request via a google search, Clearpath Robotics has released a forked version of this package & the libmscl dependency through their own build farm & package server.

The source for the fork is here: https://github.com/clearpathrobotics/ros_mscl

To install the packages via apt, follow the instructions on https://packages.clearpathrobotics.com to add Clearpath's apt and rosdep sources to your machine.  Currently supported for Ubuntu 18.04 and 16.04 (ROS Melodic & Kinetic).  Ubuntu 20.04 (ROS Noetic) is coming soon as of the time of this writing.

nathanmillerparker, 2021-05-05:
Hey Chris,

I started down the path of working on this, but got stuck at having to have our own package server for MSCL... that is harder for us to get setup than it should be, but that's the reality of our IT situation currently.  I got pulled off onto something else, but will eventually get back to this.  I do appreciate that you guys at Clearpath have provided a solution for people who need it!

civerachb-cpr, 2021-05-05:
Thanks for the update!  I've been doing my best to keep our released version in-sync with yours and submitting PRs as-needed when we run into problems.
If hosting your own package server isn't an option, have you looked into submitting a PR to the OSRF's build farm and having it hosted there?

nathanmillerparker, 2021-05-05:
I was going down the route of having the driver hosted on build.ros.org, and ran into needing the MSCL binary package on a package server... I guess we could have that built from source as well, but it has a few large dependencies (mainly boost), so I'd have to look at what it would take to build them together.  We already have the binaries available on Github, but I'm not sure that is helpful with bloom.  We have added a task to take a look at this.  As you can probably guess, this is not my area of expertise.  I appreciate the PRs you've submitted and your efforts to keep things sync'd!

civerachb-cpr, 2021-05-05:
Our hosted version of libmscl literally just grabs the latest version from Github and uses them as-is.
As-of today I think we're pretty much sync'd with the latest version.  Our version numbers are off because of how blooming works.  I typically start one minor version ahead of yours, keep the patch initially the same, and increment from there.  Next time I do a re-sync I'll probably increase the minor version again, just to make sure we don't have conflicts.  It's not the tidiest way, but it seems to be working so far.

nathanmillerparker, 2021-05-05:
Sounds good... I'll let you know when we are successful in getting this to work!

robbiefish, 2021-11-19:
@civerachb-cpr 

The packages in this repo are now being hosted on the ROS buildfarm for `melodic`, `noetic`, `foxy`, and `galactic` under the name `microstrain_inertial_driver`, `microstrain_inertial_msgs`, and `microstrain_inertial_examples`.

The rename was due to requirements laid out by the ROS maintainers [here](https://www.ros.org/reps/rep-0144.html), but aside from that, the functionality in the hosted packages should be the same.

closed 2021-11-22 by robbiefish