LORD-MicroStrain/microstrain_inertial #14: Rosdep Error when building on Melodic

civerachb-cpr, 2020-08-04:
The ROS package currently depends on `libmscl`, but when building the package for Melodic, running `rosdep install --from-paths src --ignore-src` results in this error:

`ros_mscl: Cannot locate rosdep definition for [libmscl]`

This occurs after downloading the pre-built .deb packages from https://github.com/LORD-MicroStrain/MSCL#downloads and installing with `dpkg -i *.deb`.

The only work-around I've found is to remove the `<build_depend>libmscl</build_depend>` line from package.xml.

Is there a missing step to add a new source to rosdep/sources.list.d to resolve the missing dependency?

kbhakt, 2021-02-18:
I also am getting a similar error. Any updates on this issue?

civerachb-cpr, 2021-02-19:
Since LORD doesn't appear especially inclined to maintain this package, we've created a fork here: https://github.com/clearpathrobotics/ros_mscl

This package, and the libmscl dependency, are installable as apt packages for Ubuntu 16.04 and 18.04 (20.04 coming soon!) if you follow the instructions on https://packages.clearpathrobotics.com to add Clearpath's apt and rosdep sources.

kbhakt, 2021-02-19:
Ok thanks! I was trying with Ubuntu 20.04 and ROS noetic, so this explains the dependency issue. I will try with an older version of Ubuntu and ROS. Appreciate the response!

When do you envision the 20.04 version coming out?

civerachb-cpr, 2021-02-19:
Hopefully within a couple of weeks.  We're upgrading our build-farm to support ROS2 and Noetic right now.  Once those changes are done we'll start publishing updated packages as we implement them.

The dependency issue was happening to me on Melodic; it's caused by a missing rosdep entry resolving libmscl to an apt package.  For example: https://github.com/clearpathrobotics/public-rosdistro/blob/11e27c9d69845c0f76df461a0e3234fd157ce6f4/rosdep/melodic.yaml#L30

If you add Clearpath's apt and rosdep sources rosdep should be able to resolve `libmscl` into the `c++-mscl` apt package.  There is a small problem with the current apt version of mscl_ros where libmscl.so isn't resolved by ldconfig correctly.  We have a fix for that staged for release, and a work-around is described in the README in the ros_mscl fork I linked to above.

nathanmillerparker, 2021-02-22:
Hi civerachb-cpr,

Saying that we are "not inclined to maintain this package" is a bit disingenuous.  We are actively supporting requests, fixing any bugs we or our customers find, and updating the driver code as features and sensors are added.  

I do understand that the 2 issues you have posted around the build process have been open for some time and I apologize for that.  We would like to fix these issues but a combination of a lack of bandwidth and expertise in these areas have prevented us from doing them as of yet.  They are on our radar.

That being said... you are, of course, welcome to fork the driver and host it for your own purposes.  I would ask that you periodically update your driver with changes from our driver if possible.  Having 2 drivers that are diverging in functionality will make it difficult for us to support customer issues.

Thank you,

Nathan

civerachb-cpr, 2021-02-22:
My apologies. That was a poor choice of words on my part.  When I wrote that I was feeling exasperated that those two requests had been open for so long and there hadn't even been a comment to say "thanks for the input" or "we're looking into that and will keep you updated."  I misinterpreted your silence as a lack of interest.  I'm sorry if my words came off as overly snarky.

I'm glad to hear you are planning on rectifying these problems though!  If you need help figuring out the exact process for submitting your packages to OSRF to get the packages hosted on their servers you can probably reach out directly to us; this is your driver and building it/hosting it ourselves was a last-resort after months of silence from you.  We _want_ to use the official driver, provided by you, we were just unable to.

As for our fork, we were definitely planning on keeping in sync'd with yours as much as we're able to.  Once your version is officially released (either through OSRF, or by setting up your own server and providing instructions for adding the appropriate rosdep & apt sources) we'll likely archive our fork and unpublish ours so as to avoid any confusion to the wider community.

nathanmillerparker, 2021-02-22:
Hey civerachb-cpr,

No problem, I just didn't want customers to see this in a google search and think we've abandoned the driver all-together!  Also, the responsibility for the driver went through a couple of hands since you posted those support tickets, so I apologize that we didn't respond promptly.

I'll get in touch with you for your assistance on this so we can get it fixed!

Aposhian, 2021-06-03:
Any updates on publishing packages to OSRF? This would make working with the package much more ergonomic.

nathanmillerparker, 2021-06-08:
We hear you and would like to add this, but we haven't been able to at this point.   It is on our radar.

robbiefish, 2021-11-19:
If building from source, this issue should be fixed in the `master`, `ros`, and `ros2` branches as we are now installing MSCL in the `CMakeLists.txt` manually instead of depending on it in the `package.xml`. For more documentation on this please see the following documentation for [ros](https://github.com/LORD-MicroStrain/microstrain_inertial/tree/ros#mscl) and [ros2](https://github.com/LORD-MicroStrain/microstrain_inertial/tree/ros2#mscl)

Regarding the buildfarm, this package is now published to OSRF please see the following documentation for [ros](https://github.com/LORD-MicroStrain/microstrain_inertial/tree/ros#buildfarm) and [ros2](https://github.com/LORD-MicroStrain/microstrain_inertial/tree/ros2#buildfarm)

civerachb-cpr, 2021-11-19:
That's fantastic news! Thanks for the update.  I'll see about pulling down the fork that we've been publishing to avoid situations where there are multiple versions available.

robbiefish, 2021-11-22:
Sounds good, I am going to close this and #12, please reopen if you feel either are not fixed

closed 2021-11-22 by robbiefish