LORD-MicroStrain/microstrain_inertial #17: Changes to service not carried into CMakeLists.txt

cocharland, 2020-08-12:
Recent commit to master breaks library, unable to build without modifications to CMakeLists.txt to reflect the new naming convention. 
` fatal error: ros_mscl/SetSensor2VehicleRotation.h: No such file or directory
 #include "ros_mscl/SetSensor2VehicleRotation.h" `
After the recent change the library will not build.

nathanmillerparker, 2020-08-12:
This should be fixed in the latest commit.  Not sure why the file wasn't updated in the previous push, sorry about that!

closed 2020-08-12 by nathanmillerparker

cocharland, 2020-08-13:
After a clean build, there are still missing service definitions in the CMakeLists.txt
` ros_mscl/src/set_sensor2vehicle_offset_client.cpp:2:10: fatal error: ros_mscl/SetSensorToVehicleTranslation.h: No such file or directory
 #include "ros_mscl/SetSensorToVehicleTranslation.h" `

nathanmillerparker, 2020-08-13:
This should build now... let me know if you have any trouble!