LORD-MicroStrain/microstrain_inertial #19: gnss1PpsPulseInfo error

skyrobos, 2020-09-03:
Hi,
when running the launch file I get the Error: The gnss1PpsPulseInfo option has not been set
I'm using the 3DM-GX5-25 sensor with the same launch configuration as in the master branch

civerachb-cpr, 2020-09-03:
I'm getting the exact same error. Just build the latest master:

roslaunch ros_mscl microstrain.launch port:=/dev/ttyACM1
... logging to /home/administrator/.ros/log/4932db58-ee11-11ea-b4a3-0090fb648fca/roslaunch-cpr-pnb01-30537.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://cpr-pnb01:43229/

SUMMARY
========

PARAMETERS
 * /gx5/ros_mscl_node/angular_zupt: True
 * /gx5/ros_mscl_node/angular_zupt_topic: /moving_ang
 * /gx5/ros_mscl_node/auto_init: True
 * /gx5/ros_mscl_node/baudrate: 115200
 * /gx5/ros_mscl_node/declination: 0.23
 * /gx5/ros_mscl_node/declination_source: 2
 * /gx5/ros_mscl_node/device_setup: True
 * /gx5/ros_mscl_node/dynamics_mode: 1
 * /gx5/ros_mscl_node/gps_frame_id: gps_antenna_wgs84
 * /gx5/ros_mscl_node/gps_rate: 4
 * /gx5/ros_mscl_node/heading_source: 1
 * /gx5/ros_mscl_node/imu_angular_cov: [0.01, 0, 0, 0, 0...
 * /gx5/ros_mscl_node/imu_frame_id: gx5_link
 * /gx5/ros_mscl_node/imu_linear_cov: [0.01, 0, 0, 0, 0...
 * /gx5/ros_mscl_node/imu_orientation_cov: [0.01, 0, 0, 0, 0...
 * /gx5/ros_mscl_node/imu_rate: 100
 * /gx5/ros_mscl_node/nav_rate: 10
 * /gx5/ros_mscl_node/odom_child_frame_id: sensor_ned
 * /gx5/ros_mscl_node/odom_frame_id: sensor_wgs84
 * /gx5/ros_mscl_node/port: /dev/ttyACM1
 * /gx5/ros_mscl_node/publish_imu: True
 * /gx5/ros_mscl_node/readback_settings: True
 * /gx5/ros_mscl_node/save_settings: True
 * /gx5/ros_mscl_node/velocity_zupt: True
 * /gx5/ros_mscl_node/velocity_zupt_topic: /moving_vel
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/expected: ['ros_mscl_node: ...
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/imu/expected: ['ros_mscl_node: ...
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/imu/path: imu
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/imu/remove_prefix: ros_mscl_node
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/imu/timeout: 5.0
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/imu/type: diagnostic_aggreg...
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/packet/expected: ['ros_mscl_node: ...
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/packet/path: packet
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/packet/remove_prefix: ros_mscl_node
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/packet/timeout: 5.0
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/packet/type: diagnostic_aggreg...
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/path: general
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/port/expected: ['ros_mscl_node: ...
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/port/path: port
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/port/remove_prefix: ros_mscl_node
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/port/timeout: 5.0
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/port/type: diagnostic_aggreg...
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/remove_prefix: ros_mscl_node
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/timeout: 5.0
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/type: diagnostic_aggreg...
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/path: ros_mscl_node
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/type: diagnostic_aggreg...
 * /rosdistro: kinetic
 * /rosversion: 1.12.16

NODES
  /
    imu_diagnostic_aggregator (diagnostic_aggregator/aggregator_node)
  /gx5/
    ros_mscl_node (ros_mscl/ros_mscl_node)

auto-starting new master
process[master]: started with pid [30554]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 4932db58-ee11-11ea-b4a3-0090fb648fca
process[rosout-1]: started with pid [30581]
started core service [/rosout]
process[gx5/ros_mscl_node-2]: started with pid [30613]
process[imu_diagnostic_aggregator-3]: started with pid [30614]
[ INFO] [1599156861.704546035]: Attempting to open serial port </dev/ttyACM1> at <115200> 

[ INFO] [1599156862.071826660]: Model Name:    3DM-GX5-25

[ INFO] [1599156862.072024378]: Serial Number: 6253.105125

[ INFO] [1599156862.128460314]: Setting to Idle: Stopping data streams and/or waking from sleep
[ INFO] [1599156862.131526412]: Setting AHRS active channel fields
[ INFO] [1599156862.143620264]: Setting Declination Source
[ INFO] [1599156862.149538817]: Setting Auto Initialization Control
[ INFO] [1599156862.173718669]: Setting Estimation Filter active channel fields
[ INFO] [1599156862.185647802]: Setting heading source to 0X01
[ INFO] [1599156862.335577821]: Setting spin rate to <200>
[ INFO] [1599156862.431548378]: Starting Data Parsing
[FATAL] [1599156863.714124559]: Error: The gnss1PpsPulseInfo option has not been set
[gx5/ros_mscl_node-2] process has finished cleanly
log file: /home/administrator/.ros/log/4932db58-ee11-11ea-b4a3-0090fb648fca/gx5-ros_mscl_node-2*.log

The log file specified on the last line does not exist; if it did I would also attach it.

Nova555, 2020-09-04:
I did, and the mistake remained.

rdslord, 2020-09-04:
Hey! Sorry about this! If you update to MSCL v58.0.7 and get the newest version, this should be resolved. Thank you!

skyrobos, 2020-09-06:
updating both the MSCL and the package solved the problem, closing the issue.

closed 2020-09-06 by skyrobos