LORD-MicroStrain/microstrain_inertial #20: [Question] what is the functionality of the device setup?

skyrobos, 2020-09-06:
In the launch file, there is the device setup param.
Does it change the device configuration?
I used the SensorConnect to config the device if the value of the device setup is true, will the configuration will be changed?

rdslord, 2020-09-08:
If the device setup param is true, then the device will be configured according to the other parameters in the launch file. If the device is already configured, device setup should be set to false.

skyrobos, 2020-09-08:
Can you please give an example of the other parameters?

rdslord, 2020-09-08:
In the [example launch file](https://github.com/LORD-MicroStrain/ROS-MSCL/blob/master/ros_mscl/launch/microstrain.launch), parameters like declination_source, declination, or dynamics_mode are only set if device_setup is true. If you have already configured your device, you should set device_setup to false so that your configuration isn't overridden.

skyrobos, 2020-09-08:
So if I configured the device via SensoeConnet or MIPS then the value should be false?

rdslord, 2020-09-08:
Yes.

closed 2020-09-08 by skyrobos