LORD-MicroStrain/microstrain_inertial #21: No IMU data published on the ROS node

Vineet-Pandey, 2020-09-08:
I am using the ROS package to listen to the IMU data published. But upon launching the node, I get zero value o/p  as shown below:

```
header:
   seq: 440
   stamp:
     secs: 0
     nsecs:         0
   frame_id: ''
orientation:
   x: 0.0
   y: 0.0
   z: 0.0
   w: 0.0
orientation_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity:
   x: 0.0
   y: 0.0
   z: 0.0
angular_velocity_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
linear_acceleration:
   x: 0.0
   y: 0.0
   z: 0.0
linear_acceleration_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]

```
Upon investigating further, I noticed that there are following errors too:

```
[FATAL] [1599582241.557977093]: Error: The gnss1PpsPulseInfo option has not been set
```
and 
```
WARNING These nodes have died: 
 * gx5/ros_mscl_node-2
```
Could you please help me in figuring out the possible solution? TIA.

rdslord, 2020-09-08:
Hey! Could you confirm that you are using MSCL v58.0.7 and that you are using the most current version of ROS-MSCL?

Vineet-Pandey, 2020-09-08:
Hi. I was using version 57.2.2. Now, I installed the latest version but I am still receiving the same error.

rdslord, 2020-09-08:
Did you also update to the latest version of this package?

Vineet-Pandey, 2020-09-08:
Yes, and it is working now! Thank you.

closed 2020-09-08 by Vineet-Pandey