LORD-MicroStrain/microstrain_inertial #23: About topic '/gx5/nav/filtered_imu/data'

isValid, 2020-09-23:
I updated the dependent libraries(v58.0.7) and packages. 
Then I modified some parameters in the launch file:
filter_sensor2vehicle_frame_selector: 0->3
filter_heading_source: 1->0
After I started GX5-25.
The topic '/gx5/nav/filtered_imu/data' show:
orientation: x:0 y:0 z:0 w:0
linear_acceleration: x:0 y:0 z:0
I think there is a problem, can you please check.

nathanmillerparker, 2020-09-23:
If your filter heading source is "0", the kalman filter will not initialize without an initial heading.  You will need to use the InitFilterHeading service or the InitFilterEuler to initialize the filter.

isValid, 2020-09-24:
OK,thank you, I change the heading source is "1", I can get the orientation and linear_acceleration data, but the z value of linear_acceleration is close to 0, it seems it should be -9.8, like the data before filtering?

isValid, 2020-09-24:
[](url)

nathanmillerparker, 2020-09-24:
The linear acceleration excludes gravity

isValid, 2020-09-24:
There is gravitational acceleration before filtering but not after filtering?

nathanmillerparker, 2020-09-24:
For that data quantity, yes.  If we didn't remove gravity, it would have just been a copy of what comes from IMU.. this is a convenience quantity that some of our customers have asked for.

isValid, 2020-09-24:
Okay, I understand, so if I need, I can use the linear acceleration of IMU

nathanmillerparker, 2020-09-24:
Yep

isValid, 2020-09-24:
ok,thank you~

closed 2020-09-24 by isValid