LORD-MicroStrain/microstrain_inertial #24: What's the difference between /gx5/nav/odom and /gx5/nav/odom

xhzhu-robotic, 2020-09-23:
[image]

Hi, I just use 3DM-GX5-45 GNSS with activated antenna. I want to get IMU current position using GPS. But I'm not sure what position mean in /gx5/nav/odom and /gx5/nav/odom ROS published topic. 

Is it the global postion in ECEF frame?

Are these information calculated for antenna current positon or IMU current position if IMU and antenna are placed in different location ?

And could you tell me what's the difference between /gx5/nav/odom and /gx5/nav/odom please?

I would appreciate it if you can help me.

xhzhu-robotic, 2020-09-23:
Sorry about the spelling mistakes. I want to ask the difference between /gx5/nav/odom and /gx5/gps/odom

nathanmillerparker, 2020-09-24:
The "/gx5/nav/odom" is the solution from the sensor-fusion kalman filter, the "/gx5/gps/odom" comes from the GPS module itself.

nathanmillerparker, 2020-09-24:
_Is it the global postion in ECEF frame?_

LLH frame for position and NED frame for velocity

_Are these information calculated for antenna current positon or IMU current position if IMU and antenna are placed in different location ?_

The GPS data is at the antenna position, the NAV data is at the device, assuming the antenna offset has been defined

xhzhu-robotic, 2020-09-24:
Hi Parker. Thanks for replying me. Does the position data in /gx5/gps/odom is the same as the data in /gx5/gps/fix? Because I find these two values of data are all most same.

nathanmillerparker, 2020-09-24:
Yes, those are the same values coming from the same data source

closed 2020-09-24 by xhzhu-robotic