LORD-MicroStrain/microstrain_inertial #25: Question on the reference frame of the /nav/odom/pose

zzangjinsun, 2020-10-04:
Hello!

I am using a 3DM-GX5-45 module.

I have a quick question on the reference frame of the /nav/odom/pose which is defined here:

https://github.com/LORD-MicroStrain/ROS-MSCL/blob/ac34e8cdc6f709b8429fc4a9b2e17dcd9440cd2e/ros_mscl/src/microstrain_3dm.cpp#L1395-L1405

The user manual says that the reference of the **orientation** is the **ECEF**, but I wonder whether it's referential is the **NED**.
(However, it looks clear that the position is LLH.)

Thx!!

nathanmillerparker, 2020-10-05:
Hi,

Yeah, the manual says the Earth-fixed frame, which is a bit ambiguous.  The correct frame is the NED frame, so the transformation is q_ned2sensor.  Sorry for the confusion.

zzangjinsun, 2020-10-05:
Thank you very much for the quick reply!

closed 2020-10-05 by zzangjinsun