LORD-MicroStrain/microstrain_inertial #27: Call to publish() on an invalid Publisher

JohannesByle, 2020-11-05:
When running the `roslaunch ros_mscl microstrain.launch` command I get the following error. I am not sure if this is due to me not configuring the catkin_ws properly or if there is some setup step I missed.
```
... logging to /home/johannes/.ros/log/2799b704-1f0e-11eb-b44e-6bfd0c90fc44/roslaunch-johannes-ThinkPad-T520-9721.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://johannes-ThinkPad-T520:40185/

SUMMARY
========

PARAMETERS
 * /gx5/ros_mscl_node/baudrate: 115200
 * /gx5/ros_mscl_node/device_setup: True
 * /gx5/ros_mscl_node/filter_adaptive_level: 2
 * /gx5/ros_mscl_node/filter_adaptive_time_limit_ms: 15000
 * /gx5/ros_mscl_node/filter_angular_zupt: True
 * /gx5/ros_mscl_node/filter_angular_zupt_topic: /moving_ang
 * /gx5/ros_mscl_node/filter_auto_heading_alignment_selector: 1
 * /gx5/ros_mscl_node/filter_auto_init: True
 * /gx5/ros_mscl_node/filter_data_rate: 10
 * /gx5/ros_mscl_node/filter_declination: 0.23
 * /gx5/ros_mscl_node/filter_declination_source: 2
 * /gx5/ros_mscl_node/filter_dynamics_mode: 1
 * /gx5/ros_mscl_node/filter_enable_acceleration_constraint: 0
 * /gx5/ros_mscl_node/filter_enable_altimeter_aiding: False
 * /gx5/ros_mscl_node/filter_enable_angular_constraint: 0
 * /gx5/ros_mscl_node/filter_enable_external_heading_aiding: False
 * /gx5/ros_mscl_node/filter_enable_gnss_antenna_cal: False
 * /gx5/ros_mscl_node/filter_enable_gnss_heading_aiding: True
 * /gx5/ros_mscl_node/filter_enable_gnss_pos_vel_aiding: True
 * /gx5/ros_mscl_node/filter_enable_magnetometer_aiding: False
 * /gx5/ros_mscl_node/filter_enable_odometer_aiding: False
 * /gx5/ros_mscl_node/filter_enable_velocity_constraint: 0
 * /gx5/ros_mscl_node/filter_enable_vertical_gyro_constraint: False
 * /gx5/ros_mscl_node/filter_enable_wheeled_vehicle_constraint: False
 * /gx5/ros_mscl_node/filter_gnss_antenna_cal_max_offset: 0.1
 * /gx5/ros_mscl_node/filter_heading_source: 1
 * /gx5/ros_mscl_node/filter_init_attitude: [0.0, 0.0, 0.0]
 * /gx5/ros_mscl_node/filter_init_condition_src: 0
 * /gx5/ros_mscl_node/filter_init_position: [0.0, 0.0, 0.0]
 * /gx5/ros_mscl_node/filter_init_reference_frame: 2
 * /gx5/ros_mscl_node/filter_init_velocity: [0.0, 0.0, 0.0]
 * /gx5/ros_mscl_node/filter_odom_child_frame_id: sensor_ned
 * /gx5/ros_mscl_node/filter_odom_frame_id: sensor_wgs84
 * /gx5/ros_mscl_node/filter_relative_position_frame: 2
 * /gx5/ros_mscl_node/filter_relative_position_ref: [0.0, 0.0, 0.0]
 * /gx5/ros_mscl_node/filter_reset_after_config: True
 * /gx5/ros_mscl_node/filter_sensor2vehicle_frame_selector: 0
 * /gx5/ros_mscl_node/filter_sensor2vehicle_frame_transformation_euler: [0.0, 0.0, 0.0]
 * /gx5/ros_mscl_node/filter_sensor2vehicle_frame_transformation_matrix: [1.0, 0.0, 0.0, 0...
 * /gx5/ros_mscl_node/filter_sensor2vehicle_frame_transformation_quaternion: [0.0, 0.0, 0.0, 1.0]
 * /gx5/ros_mscl_node/filter_speed_lever_arm: [0.0, 0.0, 0.0]
 * /gx5/ros_mscl_node/filter_velocity_zupt: True
 * /gx5/ros_mscl_node/filter_velocity_zupt_topic: /moving_vel
 * /gx5/ros_mscl_node/gnss1_antenna_offset: [0.0, -0.7, -1.0]
 * /gx5/ros_mscl_node/gnss1_data_rate: 2
 * /gx5/ros_mscl_node/gnss1_frame_id: gnss1_antenna_wgs84
 * /gx5/ros_mscl_node/gnss2_antenna_offset: [0.0, 0.7, -1.0]
 * /gx5/ros_mscl_node/gnss2_data_rate: 2
 * /gx5/ros_mscl_node/gnss2_frame_id: gnss2_antenna_wgs84
 * /gx5/ros_mscl_node/imu_angular_cov: [0.01, 0, 0, 0, 0...
 * /gx5/ros_mscl_node/imu_data_rate: 100
 * /gx5/ros_mscl_node/imu_frame_id: gx5_link
 * /gx5/ros_mscl_node/imu_linear_cov: [0.01, 0, 0, 0, 0...
 * /gx5/ros_mscl_node/imu_orientation_cov: [0.01, 0, 0, 0, 0...
 * /gx5/ros_mscl_node/port: /dev/ttyACM0
 * /gx5/ros_mscl_node/publish_filter: True
 * /gx5/ros_mscl_node/publish_gnss1: True
 * /gx5/ros_mscl_node/publish_gnss2: True
 * /gx5/ros_mscl_node/publish_imu: True
 * /gx5/ros_mscl_node/rtk_dongle_enable: True
 * /gx5/ros_mscl_node/save_settings: True
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/expected: ['ros_mscl_node: ...
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/imu/expected: ['ros_mscl_node: ...
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/imu/path: imu
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/imu/remove_prefix: ros_mscl_node
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/imu/timeout: 5.0
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/imu/type: diagnostic_aggreg...
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/packet/expected: ['ros_mscl_node: ...
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/packet/path: packet
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/packet/remove_prefix: ros_mscl_node
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/packet/timeout: 5.0
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/packet/type: diagnostic_aggreg...
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/path: general
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/port/expected: ['ros_mscl_node: ...
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/port/path: port
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/port/remove_prefix: ros_mscl_node
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/port/timeout: 5.0
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/port/type: diagnostic_aggreg...
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/remove_prefix: ros_mscl_node
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/timeout: 5.0
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/type: diagnostic_aggreg...
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/path: ros_mscl_node
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/type: diagnostic_aggreg...
 * /rosdistro: noetic
 * /rosversion: 1.15.8

NODES
  /
    imu_diagnostic_aggregator (diagnostic_aggregator/aggregator_node)
  /gx5/
    ros_mscl_node (ros_mscl/ros_mscl_node)

ROS_MASTER_URI=http://localhost:11311

process[gx5/ros_mscl_node-1]: started with pid [9735]
process[imu_diagnostic_aggregator-2]: started with pid [9736]
[ INFO] [1604543258.955646433]: Attempting to open serial port </dev/ttyACM0> at <115200> 

[ INFO] [1604543258.970028549]: Model Name:    3DM-GX5-15

[ INFO] [1604543258.970132341]: Serial Number: 6254.97990

[ INFO] [1604543258.987964403]: Setting to Idle: Stopping data streams and/or waking from sleep
[ INFO] [1604543258.993921694]: Setting IMU data to stream at 100 hz
[ INFO] [1604543259.011824163]: Setting Filter data to stream at 10 hz
[ INFO] [1604543259.023906510]: Setting autoinitialization to 1
[ INFO] [1604543259.036073166]: Saving the launch file configuration settings to the device
[ INFO] [1604543259.036197466]: Resetting the filter after the configuration is complete.
[ INFO] [1604543259.056128453]: Publishing IMU data.
[ INFO] [1604543259.057982941]: Publishing Filter data.
[ INFO] [1604543259.086002628]: Setting spin rate to <200>
[ INFO] [1604543259.093780897]: Starting Data Parsing
[FATAL] [1604543259.094339768]: ASSERTION FAILED
	file = /opt/ros/noetic/include/ros/publisher.h
	line = 104
	cond = false
	message = 
[FATAL] [1604543259.094417686]: Call to publish() on an invalid Publisher
[FATAL] [1604543259.094481036]: 

[gx5/ros_mscl_node-1] process has died [pid 9735, exit code -5, cmd /home/johannes/catkin_ws/devel/lib/ros_mscl/ros_mscl_node __name:=ros_mscl_node __log:=/home/johannes/.ros/log/2799b704-1f0e-11eb-b44e-6bfd0c90fc44/gx5-ros_mscl_node-1.log].
log file: /home/johannes/.ros/log/2799b704-1f0e-11eb-b44e-6bfd0c90fc44/gx5-ros_mscl_node-1*.log
```

JohannesByle, 2020-11-05:
I was able to find the issue. In line 1281 of microstrain_3dm.cpp the call to publish failed `m_mag_pub.publish(m_mag_msg);`. Commenting out this line fixed the problem

nathanmillerparker, 2020-11-05:
Hello Johannes,

Thank you for alerting us to this issue.  I just committed a fix for it.

Have a good day,

Nathan

JohannesByle, 2020-11-05:
Just tested it and it works, thank you!

closed 2020-11-05 by JohannesByle