LORD-MicroStrain/microstrain_inertial #29: build issue in jetson xavier nx

linzha0, 2021-02-10:
hi, I am trying to build this driver in my jetson xavier nx, which installed ubuntu 18.04. I've played microstrain_mips successfully. 

I followed the instruction, installed MSCL first and catkin_make the ros_mscl. Then I got wired error like
>CMake Error at ros_mscl/CMakeLists.txt:453 (target_link_libraries):
  Attempt to add link library "/opt/ros/melodic/lib/librosconsole.so" to
  target "set_tare_orientation_client" which is not built in this directory.

> CMake Error at ros_mscl/CMakeLists.txt:453 (target_link_libraries):
  Attempt to add link library "/opt/ros/melodic/lib/librosconsole_log4cxx.so"
  to target "set_tare_orientation_client" which is not built in this
  directory.

>CMake Error at ros_mscl/CMakeLists.txt:453 (target_link_libraries):
  Attempt to add link library
  "/opt/ros/melodic/lib/librosconsole_backend_interface.so" to target
  "set_tare_orientation_client" which is not built in this directory.

I think all the executable has this kind of problem.  Thanks for any suggestions. 
Also, can anyone explain a little the difference between the microstrain_mips and ROS_MSCL?

nathanmillerparker, 2021-02-10:
Hi linzha0,

If you could attach the full output of your build, it might help us determine what might be going on.

ROS_MSCL is officially supported by Lord Microstrain (the product manufacturer) and is updated as new devices are brought into the market.  microstrain_mips was started by an individual developer.  We branched the original project and added to it.

linzha0, 2021-02-10:
sorry, I can't attach all the output, because it's more than the terminal can display. What can see all the executable in the CMakeLists.txt have the link problem, they are **not built in this directory.**

nathanmillerparker, 2021-02-10:
Can you run (in your root project directory):

catkin_make clean
catkin_make 

Does that fix it?

The only references to this error that I can find are a problem with the setup.

linzha0, 2021-02-13:
Hi, sorry for the late reply. It's not working. I also tried in my laptop installed with ubuntu 18.04. Got same issue.

rdslord, 2021-02-14:
Hey there, it sounds like you might be missing some environment variables. Have you called "source devel/setup.bash" from your catkin workspace? If so, does it succeed and what is the output?

linzha0, 2021-02-17:
when I comment all the service executables,  the actual issue is ros_mscl is trying to build **microstrain_diagnostic_updater**, which has the same name from _microstrain_mips_ ros package. I suggest to change the name of **microstrain_diagnostic_updater**. People may test both packages.

closed 2021-02-17 by linzha0