LORD-MicroStrain/microstrain_inertial #32: ROS Message Conventions

CaptKrasno, 2021-02-17:
I am super excited to see official  support from LORD/Microstrain for ROS!   I have been looking through the code and have a few comments.

Ros convention specifies that message files should be camel cased.
http://wiki.ros.org/ROS/Patterns/Conventions#Messages

I find that it is often best to have the messages split into a separate ROS package ( example:  https://bitbucket.org/croman_and_the_barbarians/mscl_ros/src/master/ ).   This allows systems with multiple hosts to parse messages without needing to install the entirety of MSCL and the ros driver.  Often you want a computer to run the sensors and one to display a GUI.   In that case, the GUI computer would only need to compile mscl_msgs to read them and display statuses.  

I also made a few pull requests for some modifications and corrections. 

Loving the package so far,
Kris

nathanmillerparker, 2021-02-17:
Hey CaptKrasno,

Thanks for these... let us process this along with the other pull request you made over the next few days and provide an update.
We are glad you are enjoying the driver and we are always looking forward to making it better!

Thanks again!

Nathan

CaptKrasno, 2021-02-18:
Nathan,

I am happy to make the tweaks to the ROS messages and submit them as a pull request if/when you want to implement them.  I didn't bother yet since it would change the naming of the message types and potentially break compatibility with other packages (and to limit the number of simultaneous pull requests lol).      

-Kris

nathanmillerparker, 2021-02-18:
Hey Kris,

I am also worried about breaking compatibility with our customer's code, although I do also want to conform to the standard.  I am training up a new employee this week, let me take a look at this issue next week and get back to you.  I certainly appreciate the help!

Thanks,

Nathan

nathanmillerparker, 2021-02-22:
Hey Kris,

If you want to add the pull request for the message format, we will work to get it integrated.  We've decided to conform to the standard and help any customers through the transition.

Thanks for the help!

Nathan

CaptKrasno, 2021-02-22:
I will update the message formats.   Before I start though:   do you plan on merging the [gps_correlation_timestamp](https://github.com/LORD-MicroStrain/ROS-MSCL/pull/30) pull request?  If so, can you resolve the pull request before before I update the message format so we can avoid needless merge conflicts later?

CaptKrasno, 2021-02-24:
resolved with [https://github.com/LORD-MicroStrain/ROS-MSCL/pull/34](https://github.com/LORD-MicroStrain/ROS-MSCL/pull/34)

closed 2021-02-24 by CaptKrasno