LORD-MicroStrain/microstrain_inertial #33: Attempting to open serial port </dev/ttyACM0> at <115200>   [ERROR] [1613787499.782086336]: Device Disconnected

JiaminGuo, 2021-02-20:
I have already install the drivers and download the package and run catkin_make.
However,when I run roslaunch ros_mscl microstrain.launch, I can not connect the 3dm-gx5-45
My 3dm-gx5-45 works well when I use it in windows, but it can't work in ubuntu 18.04

robot@robot:~/imu_ws$ roslaunch ros_mscl microstrain.launch
... logging to /home/robot/.ros/log/e5d5113e-7321-11eb-8e50-e470b8ffae33/roslaunch-robot-9372.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://robot:34765/

SUMMARY
========

PARAMETERS
 * /gx5/ros_mscl_node/baudrate: 115200
 * /gx5/ros_mscl_node/device_setup: True
 * /gx5/ros_mscl_node/filter_adaptive_level: 2
 * /gx5/ros_mscl_node/filter_adaptive_time_limit_ms: 15000
 * /gx5/ros_mscl_node/filter_angular_zupt: True
 * /gx5/ros_mscl_node/filter_angular_zupt_topic: /moving_ang
 * /gx5/ros_mscl_node/filter_auto_heading_alignment_selector: 1
 * /gx5/ros_mscl_node/filter_auto_init: True
 * /gx5/ros_mscl_node/filter_data_rate: 10
 * /gx5/ros_mscl_node/filter_declination: 0.23
 * /gx5/ros_mscl_node/filter_declination_source: 2
 * /gx5/ros_mscl_node/filter_dynamics_mode: 1
 * /gx5/ros_mscl_node/filter_enable_acceleration_constraint: 0
 * /gx5/ros_mscl_node/filter_enable_altimeter_aiding: False
 * /gx5/ros_mscl_node/filter_enable_angular_constraint: 0
 * /gx5/ros_mscl_node/filter_enable_external_gps_time_update: False
 * /gx5/ros_mscl_node/filter_enable_external_heading_aiding: False
 * /gx5/ros_mscl_node/filter_enable_gnss_antenna_cal: False
 * /gx5/ros_mscl_node/filter_enable_gnss_heading_aiding: True
 * /gx5/ros_mscl_node/filter_enable_gnss_pos_vel_aiding: True
 * /gx5/ros_mscl_node/filter_enable_magnetometer_aiding: False
 * /gx5/ros_mscl_node/filter_enable_odometer_aiding: False
 * /gx5/ros_mscl_node/filter_enable_velocity_constraint: 0
 * /gx5/ros_mscl_node/filter_enable_vertical_gyro_constraint: False
 * /gx5/ros_mscl_node/filter_enable_wheeled_vehicle_constraint: False
 * /gx5/ros_mscl_node/filter_external_gps_time_topic: /external_gps_time
 * /gx5/ros_mscl_node/filter_gnss_antenna_cal_max_offset: 0.1
 * /gx5/ros_mscl_node/filter_heading_source: 1
 * /gx5/ros_mscl_node/filter_init_attitude: [0.0, 0.0, 0.0]
 * /gx5/ros_mscl_node/filter_init_condition_src: 0
 * /gx5/ros_mscl_node/filter_init_position: [0.0, 0.0, 0.0]
 * /gx5/ros_mscl_node/filter_init_reference_frame: 2
 * /gx5/ros_mscl_node/filter_init_velocity: [0.0, 0.0, 0.0]
 * /gx5/ros_mscl_node/filter_odom_child_frame_id: sensor_ned
 * /gx5/ros_mscl_node/filter_odom_frame_id: sensor_wgs84
 * /gx5/ros_mscl_node/filter_pps_source: 1
 * /gx5/ros_mscl_node/filter_relative_position_frame: 2
 * /gx5/ros_mscl_node/filter_relative_position_ref: [0.0, 0.0, 0.0]
 * /gx5/ros_mscl_node/filter_reset_after_config: True
 * /gx5/ros_mscl_node/filter_sensor2vehicle_frame_selector: 0
 * /gx5/ros_mscl_node/filter_sensor2vehicle_frame_transformation_euler: [0.0, 0.0, 0.0]
 * /gx5/ros_mscl_node/filter_sensor2vehicle_frame_transformation_matrix: [1.0, 0.0, 0.0, 0...
 * /gx5/ros_mscl_node/filter_sensor2vehicle_frame_transformation_quaternion: [0.0, 0.0, 0.0, 1.0]
 * /gx5/ros_mscl_node/filter_speed_lever_arm: [0.0, 0.0, 0.0]
 * /gx5/ros_mscl_node/filter_velocity_zupt: True
 * /gx5/ros_mscl_node/filter_velocity_zupt_topic: /moving_vel
 * /gx5/ros_mscl_node/gnss1_antenna_offset: [0.0, -0.7, -1.0]
 * /gx5/ros_mscl_node/gnss1_data_rate: 2
 * /gx5/ros_mscl_node/gnss1_frame_id: gnss1_antenna_wgs84
 * /gx5/ros_mscl_node/gnss2_antenna_offset: [0.0, 0.7, -1.0]
 * /gx5/ros_mscl_node/gnss2_data_rate: 2
 * /gx5/ros_mscl_node/gnss2_frame_id: gnss2_antenna_wgs84
 * /gx5/ros_mscl_node/gpio1_behavior: 0
 * /gx5/ros_mscl_node/gpio1_feature: 0
 * /gx5/ros_mscl_node/gpio1_pin_mode: 0
 * /gx5/ros_mscl_node/gpio2_behavior: 0
 * /gx5/ros_mscl_node/gpio2_feature: 0
 * /gx5/ros_mscl_node/gpio2_pin_mode: 0
 * /gx5/ros_mscl_node/gpio3_behavior: 0
 * /gx5/ros_mscl_node/gpio3_feature: 0
 * /gx5/ros_mscl_node/gpio3_pin_mode: 0
 * /gx5/ros_mscl_node/gpio4_behavior: 0
 * /gx5/ros_mscl_node/gpio4_feature: 0
 * /gx5/ros_mscl_node/gpio4_pin_mode: 0
 * /gx5/ros_mscl_node/gpio_config: False
 * /gx5/ros_mscl_node/gps_leap_seconds: 18
 * /gx5/ros_mscl_node/imu_angular_cov: [0.01, 0, 0, 0, 0...
 * /gx5/ros_mscl_node/imu_data_rate: 100
 * /gx5/ros_mscl_node/imu_frame_id: gx5_link
 * /gx5/ros_mscl_node/imu_linear_cov: [0.01, 0, 0, 0, 0...
 * /gx5/ros_mscl_node/imu_orientation_cov: [0.01, 0, 0, 0, 0...
 * /gx5/ros_mscl_node/port: /dev/ttyACM0
 * /gx5/ros_mscl_node/publish_filter: True
 * /gx5/ros_mscl_node/publish_gnss1: True
 * /gx5/ros_mscl_node/publish_gnss2: True
 * /gx5/ros_mscl_node/publish_imu: True
 * /gx5/ros_mscl_node/raw_file_directory: /home/your_name
 * /gx5/ros_mscl_node/raw_file_enable: False
 * /gx5/ros_mscl_node/raw_file_include_support_data: False
 * /gx5/ros_mscl_node/rtk_dongle_enable: True
 * /gx5/ros_mscl_node/save_settings: False
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/expected: ['ros_mscl_node: ...
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/imu/expected: ['ros_mscl_node: ...
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/imu/path: imu
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/imu/remove_prefix: ros_mscl_node
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/imu/timeout: 5.0
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/imu/type: diagnostic_aggreg...
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/packet/expected: ['ros_mscl_node: ...
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/packet/path: packet
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/packet/remove_prefix: ros_mscl_node
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/packet/timeout: 5.0
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/packet/type: diagnostic_aggreg...
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/path: general
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/port/expected: ['ros_mscl_node: ...
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/port/path: port
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/port/remove_prefix: ros_mscl_node
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/port/timeout: 5.0
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/port/type: diagnostic_aggreg...
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/remove_prefix: ros_mscl_node
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/timeout: 5.0
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/type: diagnostic_aggreg...
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/path: ros_mscl_node
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/type: diagnostic_aggreg...
 * /rosdistro: melodic
 * /rosversion: 1.14.10

NODES
  /
    imu_diagnostic_aggregator (diagnostic_aggregator/aggregator_node)
  /gx5/
    ros_mscl_node (ros_mscl/ros_mscl_node)

auto-starting new master
process[master]: started with pid [9382]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to e5d5113e-7321-11eb-8e50-e470b8ffae33
process[rosout-1]: started with pid [9393]
started core service [/rosout]
process[gx5/ros_mscl_node-2]: started with pid [9400]
process[imu_diagnostic_aggregator-3]: started with pid [9401]
[ INFO] [1613787499.770348583]: Attempting to open serial port </dev/ttyACM0> at <115200> 

[ERROR] [1613787499.782086336]: Device Disconnected
[gx5/ros_mscl_node-2] process has finished cleanly
log file: /home/robot/.ros/log/e5d5113e-7321-11eb-8e50-e470b8ffae33/gx5-ros_mscl_node-2*.log
^C[imu_diagnostic_aggregator-3] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

kbhakt, 2021-02-21:
I'm not sure if this will solve your issue, but I had to give permissions to the serial port using the chmod command

JiaminGuo, 2021-02-22:
> I'm not sure if this will solve your issue, but I had to give permissions to the serial port using the chmod command

Thank you so much, my 3dm-gx5-45 is working

closed 2021-02-22 by nathanmillerparker