LORD-MicroStrain/microstrain_inertial #35: Crash when publish_filter param is set to false

CaptKrasno, 2021-02-23:
Whenever publish_filter is set to false the ros_mscl_node crashes with the following error:

```
[FATAL] [1614105301.072624001]: ASSERTION FAILED
	file = /opt/ros/noetic/include/ros/publisher.h
	line = 104
	cond = false
	message = 
[FATAL] [1614105301.072672212]: Call to publish() on an invalid Publisher
[FATAL] [1614105301.072693563]: 
```

After a brief look through the source code it looks like when publish_filter is set to false the m_filter_pub* publishers are not initialized.   However they are still called in Microstrain::parse_filter_packet regardless of the state of publish_filter.   

It seems like Microstrain::parse_filter_packet() should never be called though if the microstrain is being configured properly not to send filter messages.     I haven't done any further digging though.

Here is my launch config that caused the error if you want to replicate it.

**aura/launch/microstrain.launch : **
```
<?xml version="1.1"?>
<launch>
  <!-- Declare arguments with default values -->
  <arg name="vessel_ns"        default = "aura" />
  <arg name="port"        default = "/dev/ttyAP3" />
  <arg name="baudrate"    default = "115200" />

  <!-- ****************************************************************** -->
  <!-- Microstrain sensor node -->
  <!-- ****************************************************************** -->

  <node name="ros_mscl_node"
        pkg="ros_mscl"
        type="ros_mscl_node"
        output="screen"
        ns="$(arg vessel_ns)/nav/primary_imu">
    <param name="port"     value="$(arg port)"     type="str" />
    <param name="baudrate" value="$(arg baudrate)" type="int" />
    <rosparam command="load" file="$(find aura)/config/microstrain.yaml"/>
  </node>

</launch>
```

**aura/launch/microstrain.launch : **
```
##
## microstrain config
##

device_setup: true
save_settings: false
raw_file_enable: false

publish_imu: true
imu_data_rate: 100 #hz
imu_frame_id: "primary_imu"
imu_orientation_cov:  [0.01, 0, 0, 0, 0.01, 0, 0, 0, 0.01]
imu_linear_cov:       [0.01, 0, 0, 0, 0.01, 0, 0, 0, 0.01]
imu_angular_cov:      [0.01, 0, 0, 0, 0.01, 0, 0, 0, 0.01]

filter_enable_external_gps_time_update: True
filter_external_gps_time_topic: "/aura/pps_time"
gps_leap_seconds: 18

publish_filter: false  # does not crash if set to true
publish_gnss1: false
publish_gnss2: false

publish_gps_corr: true
```

nathanmillerparker, 2021-02-23:
Noted... this should be fixed in our next commit.  Thanks!

nathanmillerparker, 2021-02-24:
Hey Kris,

The code that was just committed should fix this issue.  Let me me know if you are still seeing it.

CaptKrasno, 2021-02-24:
Yup that fixed it!

Thanks,
-Kris

closed 2021-02-24 by CaptKrasno