LORD-MicroStrain/microstrain_inertial #37: Please check the time stamp of this new version ROS-MSCL

nonlinear1, 2021-02-28:
I can run lio-sam with the 3DM-GX5-25 under the old version ROS-MSCL .But this newer version ROS-MSCL cann't work! I found the time stamp of the data of imu isn't correct! I think you have changed the time stamp format of your ROS-MSCL? if not, please check!
 Thank you very much!

nathanmillerparker, 2021-03-05:
Hi nonlinear, did this fix your problem?

nonlinear1, 2021-03-06:
Thank you very much! I have tested your new version of ROS-MSCL. This fix my problem!

ycl-web, 2022-12-02:
May I ask which version of ros_mscl is it? I also encountered the same problem. thank you very much @nonlinear1 @nathanmillerparker

robbiefish, 2022-12-02:
@ycl-web May I ask what version of the driver you are using? The current [`ros`](https://github.com/LORD-MicroStrain/microstrain_inertial/tree/ros) and [`ros2`](https://github.com/LORD-MicroStrain/microstrain_inertial/tree/ros2) branches have undergone a change that may have changed how the data is timestamped.

Also please note that the way the data is timestamped is dependent on configuration. The default behavior of the launch file is to timestamp the data when it is collected. If the data is transmitted over USB at a high rate, this can result in data being batched up and the timestamps will reflect this.

If you configure the driver with `use_device_timestamp: True`, the data will be timestamped with the device timestamp.

If you configure the driver with `use_ros_time: True` the data will be timestamped with ROS time right before it is published, which would display similar behavior to the default, but could behave differently if running in simulation

ycl-web, 2022-12-02:
First of all thank you very much for answering my question. I am using ros. When using the previous version, the function package is still called ros_mscl, and the data collected at that time can run lio-sam. But I downloaded the latest version of the driver, but only modified the baud rate and imu_data_rate, which made lio-sam unable to run. Is the current driver default setting the same as before? Thanks again for your answer

ycl-web, 2022-12-02:
@robbiefish

ycl-web, 2022-12-09:
thanks for your help. First I built it from source, the driver output is in the image below.
The part I modified in the source code is to put in the "param.yml" file:
port : "/dev/ttyUSB0"
baudrate: 460800
set_baud : True
imu_data_rate : 500
Other settings are default settings.
I have a few problems: 
1. I convert the quaternion of the imu/data message to roll, pitch, yaw. When I turn the imu, the yaw value will continuously increase or decrease to a value after the imu is turned to a fixed position, while the roll and pitch are displayed in real time. 2.device_setup: True
use_device_timestamp : False
use_ros_time : False
These three settings are the default settings, do they need to be modified?
3. When I run lio-sam with this version of the driver, the image built in rviz will float around, but it will not float around with the previous ros_mscl.
My operating system is ubuntu18.04
The ros version is melodic
The imu version is 3dm-gx5-25
Thanks again for your help. @robbiefish 
[image]

closed 2023-01-08 by github-actions[bot]