LORD-MicroStrain/microstrain_inertial #40: Raw Bin file to CSV

KrillesHomies, 2021-03-11:
Hello, is there currently any available program that can convert the Raw bin file, that the program creates, to a readable format like CSV?

nathanmillerparker, 2021-03-11:
Hi KrillesHomies,

We currently do not have a publicly available and supported tool that works for all our devices (I might have something that will work for you if you are not using a GQ7, you can email me at nathan.miller (at) parker.com)  Out of curiosity, what would you be looking for: A python script? An executable?  Also, does it matter if it is Windows or Linux?  Finally, what device are you working with?

KrillesHomies, 2021-03-15:
Thank you for your help. The solution you sent worked well!

closed 2021-03-15 by KrillesHomies