LORD-MicroStrain/microstrain_inertial #41: Error: Failed to communicate with the device.

JohannesByle, 2021-03-15:
When running `roslaunch ros_mscl microstrain.launch` I randomly get the following error. The IMU had been working fine for several months before that, and it just randomly stopped working. The error appeared after I used Sensor Connect on Windows, although that may have just been a coincidence. I tried switching USB ports and it would sometimes randomly work, and it will also sometimes start working after leaving it connected for a while. It works pefectly on Windows with Sensor Connect and MIP monitor, and I have tried to update the firmware but it's on the latest version.
```
... logging to /home/lidar/.ros/log/76cfd982-85e0-11eb-b2e7-ed3bd350afa3/roslaunch-Shimron7-4217.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://Shimron7:36309/

SUMMARY
========

PARAMETERS
 * /gx5/ros_mscl_node/baudrate: 115200
 * /gx5/ros_mscl_node/device_setup: True
 * /gx5/ros_mscl_node/filter_adaptive_level: 2
 * /gx5/ros_mscl_node/filter_adaptive_time_limit_ms: 15000
 * /gx5/ros_mscl_node/filter_angular_zupt: True
 * /gx5/ros_mscl_node/filter_angular_zupt_topic: /moving_ang
 * /gx5/ros_mscl_node/filter_auto_heading_alignment_selector: 1
 * /gx5/ros_mscl_node/filter_auto_init: True
 * /gx5/ros_mscl_node/filter_data_rate: 10
 * /gx5/ros_mscl_node/filter_declination: 0.23
 * /gx5/ros_mscl_node/filter_declination_source: 2
 * /gx5/ros_mscl_node/filter_dynamics_mode: 1
 * /gx5/ros_mscl_node/filter_enable_acceleration_constraint: 0
 * /gx5/ros_mscl_node/filter_enable_altimeter_aiding: False
 * /gx5/ros_mscl_node/filter_enable_angular_constraint: 0
 * /gx5/ros_mscl_node/filter_enable_external_gps_time_update: False
 * /gx5/ros_mscl_node/filter_enable_external_heading_aiding: False
 * /gx5/ros_mscl_node/filter_enable_gnss_antenna_cal: False
 * /gx5/ros_mscl_node/filter_enable_gnss_heading_aiding: True
 * /gx5/ros_mscl_node/filter_enable_gnss_pos_vel_aiding: True
 * /gx5/ros_mscl_node/filter_enable_magnetometer_aiding: False
 * /gx5/ros_mscl_node/filter_enable_odometer_aiding: False
 * /gx5/ros_mscl_node/filter_enable_velocity_constraint: 0
 * /gx5/ros_mscl_node/filter_enable_vertical_gyro_constraint: False
 * /gx5/ros_mscl_node/filter_enable_wheeled_vehicle_constraint: False
 * /gx5/ros_mscl_node/filter_external_gps_time_topic: /external_gps_time
 * /gx5/ros_mscl_node/filter_gnss_antenna_cal_max_offset: 0.1
 * /gx5/ros_mscl_node/filter_heading_source: 1
 * /gx5/ros_mscl_node/filter_init_attitude: [0.0, 0.0, 0.0]
 * /gx5/ros_mscl_node/filter_init_condition_src: 0
 * /gx5/ros_mscl_node/filter_init_position: [0.0, 0.0, 0.0]
 * /gx5/ros_mscl_node/filter_init_reference_frame: 2
 * /gx5/ros_mscl_node/filter_init_velocity: [0.0, 0.0, 0.0]
 * /gx5/ros_mscl_node/filter_odom_child_frame_id: sensor_ned
 * /gx5/ros_mscl_node/filter_odom_frame_id: sensor_wgs84
 * /gx5/ros_mscl_node/filter_pps_source: 1
 * /gx5/ros_mscl_node/filter_relative_position_frame: 2
 * /gx5/ros_mscl_node/filter_relative_position_ref: [0.0, 0.0, 0.0]
 * /gx5/ros_mscl_node/filter_reset_after_config: True
 * /gx5/ros_mscl_node/filter_sensor2vehicle_frame_selector: 0
 * /gx5/ros_mscl_node/filter_sensor2vehicle_frame_transformation_euler: [0.0, 0.0, 0.0]
 * /gx5/ros_mscl_node/filter_sensor2vehicle_frame_transformation_matrix: [1.0, 0.0, 0.0, 0...
 * /gx5/ros_mscl_node/filter_sensor2vehicle_frame_transformation_quaternion: [0.0, 0.0, 0.0, 1.0]
 * /gx5/ros_mscl_node/filter_speed_lever_arm: [0.0, 0.0, 0.0]
 * /gx5/ros_mscl_node/filter_velocity_zupt: True
 * /gx5/ros_mscl_node/filter_velocity_zupt_topic: /moving_vel
 * /gx5/ros_mscl_node/gnss1_antenna_offset: [0.0, -0.7, -1.0]
 * /gx5/ros_mscl_node/gnss1_data_rate: 2
 * /gx5/ros_mscl_node/gnss1_frame_id: gnss1_antenna_wgs84
 * /gx5/ros_mscl_node/gnss2_antenna_offset: [0.0, 0.7, -1.0]
 * /gx5/ros_mscl_node/gnss2_data_rate: 2
 * /gx5/ros_mscl_node/gnss2_frame_id: gnss2_antenna_wgs84
 * /gx5/ros_mscl_node/gpio1_behavior: 0
 * /gx5/ros_mscl_node/gpio1_feature: 0
 * /gx5/ros_mscl_node/gpio1_pin_mode: 0
 * /gx5/ros_mscl_node/gpio2_behavior: 0
 * /gx5/ros_mscl_node/gpio2_feature: 0
 * /gx5/ros_mscl_node/gpio2_pin_mode: 0
 * /gx5/ros_mscl_node/gpio3_behavior: 0
 * /gx5/ros_mscl_node/gpio3_feature: 0
 * /gx5/ros_mscl_node/gpio3_pin_mode: 0
 * /gx5/ros_mscl_node/gpio4_behavior: 0
 * /gx5/ros_mscl_node/gpio4_feature: 0
 * /gx5/ros_mscl_node/gpio4_pin_mode: 0
 * /gx5/ros_mscl_node/gpio_config: False
 * /gx5/ros_mscl_node/gps_leap_seconds: 18
 * /gx5/ros_mscl_node/imu_angular_cov: [0.01, 0, 0, 0, 0...
 * /gx5/ros_mscl_node/imu_data_rate: 100
 * /gx5/ros_mscl_node/imu_frame_id: gx5_link
 * /gx5/ros_mscl_node/imu_linear_cov: [0.01, 0, 0, 0, 0...
 * /gx5/ros_mscl_node/imu_orientation_cov: [0.01, 0, 0, 0, 0...
 * /gx5/ros_mscl_node/port: /dev/ttyACM0
 * /gx5/ros_mscl_node/publish_filter: True
 * /gx5/ros_mscl_node/publish_gnss1: True
 * /gx5/ros_mscl_node/publish_gnss2: True
 * /gx5/ros_mscl_node/publish_imu: True
 * /gx5/ros_mscl_node/raw_file_directory: /home/your_name
 * /gx5/ros_mscl_node/raw_file_enable: False
 * /gx5/ros_mscl_node/raw_file_include_support_data: False
 * /gx5/ros_mscl_node/rtk_dongle_enable: True
 * /gx5/ros_mscl_node/save_settings: False
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/expected: ['ros_mscl_node: ...
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/imu/expected: ['ros_mscl_node: ...
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/imu/path: imu
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/imu/remove_prefix: ros_mscl_node
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/imu/timeout: 5.0
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/imu/type: diagnostic_aggreg...
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/packet/expected: ['ros_mscl_node: ...
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/packet/path: packet
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/packet/remove_prefix: ros_mscl_node
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/packet/timeout: 5.0
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/packet/type: diagnostic_aggreg...
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/path: general
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/port/expected: ['ros_mscl_node: ...
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/port/path: port
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/port/remove_prefix: ros_mscl_node
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/port/timeout: 5.0
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/port/type: diagnostic_aggreg...
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/remove_prefix: ros_mscl_node
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/timeout: 5.0
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/type: diagnostic_aggreg...
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/path: ros_mscl_node
 * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/type: diagnostic_aggreg...
 * /rosdistro: noetic
 * /rosversion: 1.15.9

NODES
  /
    imu_diagnostic_aggregator (diagnostic_aggregator/aggregator_node)
  /gx5/
    ros_mscl_node (ros_mscl/ros_mscl_node)

ROS_MASTER_URI=http://localhost:11311

process[gx5/ros_mscl_node-1]: started with pid [4233]
process[imu_diagnostic_aggregator-2]: started with pid [4234]
[ INFO] [1615848514.696709117]: Attempting to open serial port </dev/ttyACM0> at <115200> 

[FATAL] [1615848514.949320403]: Error: Failed to communicate with the device.
terminate called after throwing an instance of 'mscl::Error_Communication'
  what():  Failed to communicate with the device.
[gx5/ros_mscl_node-1] process has died [pid 4233, exit code -6, cmd /home/lidar/catkin_ws/install_isolated/lib/ros_mscl/ros_mscl_node __name:=ros_mscl_node __log:=/home/lidar/.ros/log/76cfd982-85e0-11eb-b2e7-ed3bd350afa3/gx5-ros_mscl_node-1.log].
log file: /home/lidar/.ros/log/76cfd982-85e0-11eb-b2e7-ed3bd350afa3/gx5-ros_mscl_node-1*.log

```

nathanmillerparker, 2021-03-16:
Hi Johannes,

Are you running in native Linux or a virtual machine?  What device and version of the driver are you using?

Nathan

JohannesByle, 2021-03-16:
The device is a 3dm-gx5-15 and I tested it on two different laptops running native Ubuntu 20.04, and both of them had previously been working with the device. I believe it's firmware version is 1.1.76 because when I ran the installer it said it could not update the firmware because it was already on the latest version.

mglord, 2021-03-16:
It looks like it's successfully opening the port, but not receiving a response when attempting to read information from the device. The most common cause of this is an incorrect baud rate but that wouldn't explain the intermittent success. Just to be sure, can you double check the baud rate configured on the device in SensorConnect - this will be displayed in the device info sidebar on the left side of the screen towards the bottom under Connection. If the specified value is not 115200 you can launch the ROS node with a different baud rate by including the baudrate argument on launch, ex:
`roslaunch ros_mscl microstrain.launch baudrate:=921600`

The second most common problem is bandwidth issues due to streaming. Is the device streaming data (LED blinking rapidly) or idle (LED pulsing slowly) when the ROS node is launched? If streaming can you try one of the following?
- Use SensorConnect to configure the device not to stream on startup (Save/Load Settings -> Save as Startup Settings -> Uncheck "Stream data on startup" -> Save)
- Add the below code to microstrain_3dm.cpp on line 300 before `//Print the device info` to set to idle immediately after establishing the connection.
```
for(int i = 0; i < 3; i++)
{
  try
  {
    m_inertial_device->setToIdle();
    break;
  }
  catch(const mscl::Error& e){/*ignore*/}
}
```

If neither of these resolves the issue the following info may be helpful:
- How is the device connected (USB, serial-to-USB converter) - are you using the same cable to connect to all the systems?
- The device configuration - this can be exported via SensorConnect (Save/Load Settings -> Export) and may help us recreate the issue.

Thanks!

JohannesByle, 2021-03-17:
Thank you! Just changing the setting in SensorConnect without the extra code in microstrain_3dm.cpp fixed the error, or at least it seems to be working at the moment. I hadn't paid attention to it before this error occurred, but while trying to fix the problem the LED was definitely blinking as though it was streaming before I launched the ros node, so it was probably the "stream data on startup" setting.

closed 2021-03-17 by JohannesByle