LORD-MicroStrain/microstrain_inertial #43: Taring GX5-25 via ros service

kbhakt, 2021-03-26:
Hello,

I am trying to tare all of the axis after I have mounted the sensor on my device in a certain orientation. I have tried to use the "set_tare_orientation" service call but it seems to only take 1 axis as input. I would like to tare all the axes simultaneously. Can you provide me a solution that would allow to do this easily with a command from terminal.

Currently I am using:
rosservice call /gx5/set_tare_orientation "axis: 0".

Ideally I would change 0 to all or something similar. Is this functionality available within this API? Or do I just need to call the command 3 times?

Thanks in advance!

mglord, 2021-03-26:
The tare parameter is a bitfield value where the axes are defined:
```
Reset all axes to zero: 0
Roll:  1  (0001)
Pitch: 2  (0010)
Yaw:   4  (0100)
```
In order to tare all axes at the same time you can pass in 7 (0111). By passing in 0 the tare configuration values are cleared.

Sorry for the confusion! Let us know if using "axis: 7" does not produce the expected results.

kbhakt, 2021-03-31:
Thank you for response! This worked!

I had some follow up questions:

1. Is there a method for saving the taring offsets when the imu is powered off/on in Sensor Connect?

2. In the ROS_MSCL lib, is there a way to access the taring offsets in order to save them for the next run of experiments to ensure that we are matching things as closely as possible. I would like to ideally set the offsets to a prescribed number but without having to connect to Sensor Connect as it it resets it every time the device is powered off/on.

nathanmillerparker, 2021-03-31:
Hi kbhakt,

1. If you call the DeviceSettings service with a "function_selector" value of "3", it will save all the device settings for the next power on.

2. When you tare the device, it stores the transformation internally in the sensor2vehicle frame rotation quaternion.  For non-GQ7 devices (GX5,GX4, etc.) you would use the GetSensor2VehicleRotation service to get the tared rotations as Euler angles.  You would use the corresponding SetSensor2VehicleRotation service to set them next time (also an option in the launch file exists to do this with the caveat that the launch file is defined as the transformation and not the rotation... they are inverses of each other.)  I know that last part can be confusing, but we are transitioning to the more generally accepted "transformation" representation.  Practically, what it means for you is if you read values using GetSensor2VehicleRotation and then put them in the launch file, you need to negate them - i.e. if you read (0.1, 0.2, 0.3), you would put (-0.1, -0.2, -0.3) in the launch file.  You do not need to negate them if you use the "SetSensor2VehicleRotation" service.

Hope that helps!

closed 2021-04-20 by kbhakt