LORD-MicroStrain/microstrain_inertial #46: It looks like it's successfully opening the port, but not receiving a response when attempting to read information from the device. The most common cause of this is an incorrect baud rate but that wouldn't explain the intermittent success. Just to be sure, can you double check the baud rate configured on the device in SensorConnect - this will be displayed in the device info sidebar on the left side of the screen towards the bottom under Connection. If the specified value is not 115200 you can launch the ROS node with a different baud rate by including the baudrate argument on launch, ex:

chwbo, 2021-04-08:
It looks like it's successfully opening the port, but not receiving a response when attempting to read information from the device. The most common cause of this is an incorrect baud rate but that wouldn't explain the intermittent success. Just to be sure, can you double check the baud rate configured on the device in SensorConnect - this will be displayed in the device info sidebar on the left side of the screen towards the bottom under Connection. If the specified value is not 115200 you can launch the ROS node with a different baud rate by including the baudrate argument on launch, ex:
`roslaunch ros_mscl microstrain.launch baudrate:=921600`

The second most common problem is bandwidth issues due to streaming. Is the device streaming data (LED blinking rapidly) or idle (LED pulsing slowly) when the ROS node is launched? If streaming can you try one of the following?
- Use SensorConnect to configure the device not to stream on startup (Save/Load Settings -> Save as Startup Settings -> Uncheck "Stream data on startup" -> Save)
- Add the below code to microstrain_3dm.cpp on line 300 before `//Print the device info` to set to idle immediately after establishing the connection.
```
for(int i = 0; i < 3; i++)
{
  try
  {
    m_inertial_device->setToIdle();
    break;
  }
  catch(const mscl::Error& e){/*ignore*/}
}
```

If neither of these resolves the issue the following info may be helpful:
- How is the device connected (USB, serial-to-USB converter) - are you using the same cable to connect to all the systems?
- The device configuration - this can be exported via SensorConnect (Save/Load Settings -> Export) and may help us recreate the issue.

Thanks!

_Originally posted by @mglord in https://github.com/LORD-MicroStrain/ROS-MSCL/issues/41#issuecomment-800377407_

nathanmillerparker, 2021-04-08:
Hi chwbo,

What would you like us to do with this post?

mglord, 2021-04-08:
I think something may have gone wrong when linking to this comment (originally from ROS issue #41) in [MSCL issue 153](https://github.com/LORD-MicroStrain/MSCL/issues/153). I'm going to go ahead and close this, but feel free to re-open if I've misunderstood!

closed 2021-04-08 by mglord