LORD-MicroStrain/microstrain_inertial #47: 3DM-GQ7 Data Rate Questions

Justin-Ribbit, 2021-04-13:
Note: I reached out to the customer support staff on microstrain.com and they referred me here.

I have recently started using the 3DM-GQ7 navigation system for a navigation application and have encountered some issues that I was hoping to receive some assitance with. Specifically, I am using this sensor along with the MSCL library (version 61.0.16) and the associated ROS drivers (1.1.2) and have been encountering some issues with the data rates. I was hoping you might be able to provide some assistance and information related to the following questions:

1. What are the limits for the data rates of the IMU, GNSS, RTK, and filter? (e.g., I have noticed that I cannot increase the GNSS rate past 2 Hz, but I have not been able to find any documentation outlining why this is.)

2. Is there any documentation that outlines the allowable data rates for this device? Do these rates change depending on whether the device is operated with "sensor connect" vs. the ROS driver?

3. I cannot increase the frequency of the filter past ~35 Hz without the ROS driver crashing:
```
setting /run_id to 53fc2f06-9c64-11eb-8725-78d004293634
process[rosout-1]: started with pid [4713]
started core service [/rosout]
process[gq7/ros_mscl_node-2]: started with pid [4720]
process[imu_diagnostic_aggregator-3]: started with pid [4721]
[ INFO] [1618324029.269280639]: Attempting to open serial port </dev/ttyS0> at <115200> 
[FATAL] [1618324029.521824890]: Error: Failed to communicate with the device.
terminate called after throwing an instance of 'mscl::Error_Communication'
  what():  Failed to communicate with the device.
```
 Can you provide any insight into why that may be? I am hoping to have the filter output at 100 Hz.

4. The actual frequency of the IMU, filter, and GNSS publishers are all lower than the values that we set them to. Can you provide any insight here as well? Please see below for the ROS info stream outlining this issue:

```
[ INFO] [1618337094.343093775]: Attempting to open serial port </dev/ttyS0> at <115200> 
[ INFO] [1618337094.372635445]: Model Name:    3DM-GQ7
[ INFO] [1618337094.372677056]: Serial Number: 6284.114135
[ INFO] [1618337094.404865449]: Setting to Idle: Stopping data streams and/or waking from sleep
[ INFO] [1618337094.407846635]: Setting IMU data to stream at 100 hz
[ INFO] [1618337094.414994665]: Setting Declination Source
[ INFO] [1618337094.423942439]: Setting GNSS1 data to stream at 2 hz
[ INFO] [1618337094.430925752]: Setting GNSS1 antenna offset to [0.340000, -0.750000, -0.175000]
[ INFO] [1618337094.441619450]: Setting GNSS2 data to stream at 2 hz
[ INFO] [1618337094.453710334]: Setting GNSS2 antenna offset to [0.340000, 0.750000, -0.173000]
[ INFO] [1618337094.469539928]: Setting RTK dongle enable to 1
[ INFO] [1618337094.474337034]: Setting Filter data to stream at 35 hz
```

I then investigated the ROS outputs and found the following:
```
nav_filter publishing rate:
average rate: 27.487 min: 0.000s max: 0.065s std dev: 0.01542s window: 247

imu/data publishing rate:
average rate: 78.779 min: 0.004s max: 0.039s std dev: 0.00944s window: 391
```

Any insight/advice/tips would be greatly appreciated. I am happy to provide any other additional details that may be necessary or helpful.

nathanmillerparker, 2021-04-13:
Hi Justin,

It looks like you are having issues with the available bandwidth over the serial interface.  From what I'm able to determine from the info that you have provided, you are connecting to the GQ7 via a standard serial port at 115200 baud.  Unfortunately, that baudrate is likely too low to support all of the data you want to get out of the device.  Can you try changing your device's baudrate to 921600 using SensorConnect, save the change, and make the corresponding change to your ROS launch file?

On the MSCL error, if the device is bandwidth-saturated, it is difficult for the MSCL library to communicate with it and command responses may not come over as data is prioritized.  

On the max data rates: it is 2Hz for both GNSS modules and 1 kHz for IMU and Filter data.  Our datasheet does enumerate the 1khz maximum rate, but not the 2 hz GNSS datarate.  I will make sure this gets updated in the datasheet and into our online manual as it appears there was an oversight.  

Let me know if this doesn't fix your issue or if you need any additional assistance!

Nathan

Justin-Ribbit, 2021-04-14:
Hi Nathan,

Thank you for the speedy response. I have tried your suggestions and am still encountering the same issue as before. As instructed, I:

1. Increased the baud rate to 921600 in sensor connect and saved the change.
2. Updated the launch file to reflect this baud rate.

The filter rate still seems to max out at 35 Hz, with even a small increase above that causing the driver to throw a `mscl::Error_Communication`. I've attached the ROS infostream output for both 35 Hz and 40 Hz below:

**35 Hz:**
```
[INFO] [1618417593.622093289]: Attempting to open serial port </dev/ttyS0> at <921600> 
[ INFO] [1618417593.652411360]: Model Name:    3DM-GQ7
[ INFO] [1618417593.652450365]: Serial Number: 6284.114135
[ INFO] [1618417593.685409110]: Setting to Idle: Stopping data streams and/or waking from sleep
[ INFO] [1618417593.687990939]: Setting IMU data to stream at 100 hz
[ INFO] [1618417593.695261531]: Setting Declination Source
[ INFO] [1618417593.706009915]: Setting GNSS1 data to stream at 2 hz
[ INFO] [1618417593.712951936]: Setting GNSS1 antenna offset to [0.340000, -0.750000, -0.175000]
[ INFO] [1618417593.724805184]: Setting GNSS2 data to stream at 2 hz
[ INFO] [1618417593.742842359]: Setting GNSS2 antenna offset to [0.340000, 0.750000, -0.173000]
[ INFO] [1618417593.763747163]: Setting RTK dongle enable to 1
[ INFO] [1618417593.773202127]: Setting Filter data to stream at 35 hz
[ INFO] [1618417593.784526285]: Setting PPS source to 0X01
[ INFO] [1618417593.791658381]: Setting autoadaptive options to: level = 2, time_limit = 15000
[ INFO] [1618417593.801641892]: Filter aiding set to: pos/vel = 1, gnss heading = 1, altimeter = 1, odometer = 1, magnetometer = 1, external heading = 0
[ INFO] [1618417593.841294278]: Setting wheeled vehicle contraint enable to 0
[ INFO] [1618417593.861664971]: Setting vertical gyro contraint enable to 0
[ INFO] [1618417593.878356443]: Setting GNSS antenna calibration to: enable = 0, max_offset = 0.100000
[ INFO] [1618417594.038443232]: Resetting the filter after the configuration is complete.
[ INFO] [1618417594.187186920]: ERROR opening raw binary datafile at ~/ws_ribbit/gps_data/3DM-GQ7_6284.114135_21_04_14_16_26_34.bin
[ INFO] [1618417594.189437512]: Publishing IMU data.
[ INFO] [1618417594.191433479]: Publishing Magnetometer data.
[ INFO] [1618417594.193319737]: Publishing GNSS1 data.
[ INFO] [1618417594.198335391]: Publishing GNSS2 data.
[ INFO] [1618417594.203186614]: Publishing Filter data.
[ INFO] [1618417594.240425636]: Setting spin rate to <200>
[ INFO] [1618417594.247479204]: Starting Data Parsing
```

**and 40 Hz:**
```
[ INFO] [1618417609.590700520]: Attempting to open serial port </dev/ttyS0> at <921600> 
[ INFO] [1618417609.620977320]: Model Name:    3DM-GQ7
[ INFO] [1618417609.621096291]: Serial Number: 6284.114135
[ INFO] [1618417609.654487810]: Setting to Idle: Stopping data streams and/or waking from sleep
[ INFO] [1618417609.657259221]: Setting IMU data to stream at 100 hz
[ INFO] [1618417609.664567765]: Setting Declination Source
[ INFO] [1618417609.674047985]: Setting GNSS1 data to stream at 2 hz
[ INFO] [1618417609.681710057]: Setting GNSS1 antenna offset to [0.340000, -0.750000, -0.175000]
[ INFO] [1618417609.692932326]: Setting GNSS2 data to stream at 2 hz
[ INFO] [1618417609.703065000]: Setting GNSS2 antenna offset to [0.340000, 0.750000, -0.173000]
[ INFO] [1618417609.718940945]: Setting RTK dongle enable to 1
[ INFO] [1618417609.723527183]: Setting Filter data to stream at 40 hz
[ INFO] [1618417609.755399026]: Setting PPS source to 0X01
[ INFO] [1618417609.767147845]: Setting autoadaptive options to: level = 2, time_limit = 15000
[ INFO] [1618417609.773158177]: Filter aiding set to: pos/vel = 1, gnss heading = 1, altimeter = 1, odometer = 1, magnetometer = 1, external heading = 0
[ INFO] [1618417609.917712683]: Setting wheeled vehicle contraint enable to 0
[ INFO] [1618417609.978959733]: Setting vertical gyro contraint enable to 0
[ INFO] [1618417610.050824891]: Setting GNSS antenna calibration to: enable = 0, max_offset = 0.100000
[ INFO] [1618417610.708676333]: Resetting the filter after the configuration is complete.
[FATAL] [1618417610.959210401]: Error: Failed to communicate with the device.
terminate called after throwing an instance of 'mscl::Error_Communication'
  what():  Failed to communicate with the device.
[gq7/ros_mscl_node-2] process has died [pid 1442, exit code -6, cmd /root/ws_ribbit/devel/lib/ros_mscl/ros_mscl_node __name:=ros_mscl_node __log:=/root/.ros/log/36388c14-9d3e-11eb-af55-78d004293634/gq7-ros_mscl_node-2.log].
log file: /root/.ros/log/36388c14-9d3e-11eb-af55-78d004293634/gq7-ros_mscl_node-2*.log
```

This issue is also reflected in sensor connect - increasing the baud rate results in a communication error.

Any thoughts?

nathanmillerparker, 2021-04-14:
Hey Justin,

Could you describe your setup (target computer, using/not using virtual machine, physical connection USB/UART, etc.) and provide your launch file?  Calculating the data requirements, at 921600, I wouldn't expect you to have issues at 100 Hz IMU, 2 Hz GNSS, and 40 Hz Filter data.

Thanks,

Nathan

Justin-Ribbit, 2021-04-14:
- Target computer: Nuvo-8208GC
- The unit is connected to the computer via a serial connection provided with the device.
- The ROS driver is running inside a docker container set up (--privileged and access to /dev/ttyS0)
- The ROS driver is version 1.1.2 and the MSCL version is 61.0.16

The launch file that I am using:
```
<?xml version="1.1"?>
<launch>

  <!-- Standalone example launch file for GX3, GX4, GX/CX5, RQ1 and GQ7 series devices-->
  <!-- Note: Feature support is device-dependent and some of the following settings may have no affect on your device. --> 
     <!--  Please consult your device's documentation for supported features -->

  <!-- Declare arguments with default values -->
  <arg name="name"        default = "gq7" />
  <arg name="port"        default = "/dev/ttyS0" />
  <arg name="baudrate"    default = "912600" />
  <arg name="debug"       default = "false" />
  <arg name="diagnostics" default = "true" />

  <!-- ****************************************************************** -->
  <!-- Microstrain sensor node -->
  <!-- ****************************************************************** -->
    
  <node name="ros_mscl_node" pkg="ros_mscl" type="ros_mscl_node" output="screen" ns="$(arg name)">

    <!-- ****************************************************************** -->
    <!-- General Settings -->
    <!-- ****************************************************************** -->
    
    <param name="port"     value="$(arg port)"     type="str" />
    <param name="baudrate" value="$(arg baudrate)" type="int" />
    
    <!-- Controls if the driver-defined setup is sent to the device
          false - The driver will ignore the settings below and use the device's current settings
          true  - Overwrite the current device settings with those listed below
     -->
     
    <param name="device_setup" value="true" type="bool" />
    

    <!-- Controls if the driver-defined settings are saved 
          false - Do not save the settings
          true  - Save the settings in the device's non-volatile memory
     -->
     
    <param name="save_settings" value="false" type="bool" />
    

    <!-- Controls if the driver creates a raw binary file
          false - Do not create the file
          true  - Create the file
          Notes: 1) The filename will have the following format - 
                    model_number "_" serial_number "_" datetime (year_month_day_hour_minute_sec) ".bin"
                    example: "3DM-GX5-45_6251.00001_20_12_01_01_01_01.bin"
                 2) This file is useful for getting support from the manufacturer
     -->
     
    <param name="raw_file_enable" value="true" type="bool" />

    <!-- (GQ7 only) Controls if the driver requests additional factory support data to be included in the raw data file
          false - Do not request the additional data
          true  - Request the additional channels
          Notes: Including this feature increases communication bandwidth requirements... for serial data connections,
                 please ensure the baudrate is sufficient for the added data channels.
     -->

    <param name="raw_file_include_support_data" value="false" type="bool" />

[... 261 lines cut ...]
    <param name="filter_gnss_antenna_cal_max_offset" value="0.1"   type="double" />
  </node>

  <!-- ****************************************************************** -->
  <!-- Diagnostics -->
  <!-- ****************************************************************** -->
    
  <group if="$(arg diagnostics)">

    <!-- Diagnostic Aggregator for robot monitor usage -->
    <node pkg="diagnostic_aggregator" type="aggregator_node" name="imu_diagnostic_aggregator">
      <rosparam command="load" file="$(find ros_mscl)/config/diagnostic_analyzers.yaml" />
    </node>
  </group>

  <!-- ****************************************************************** -->
  <!-- Set the debug level to debug -->
  <!-- ****************************************************************** -->
 
  <group if="$(arg debug)">
    <env name="ROSCONSOLE_CONFIG_FILE"
       value="$(find ros_mscl)/config/custom_rosconsole.conf"/>
  </group>
</launch>
```

nathanmillerparker, 2021-04-14:
Hi Justin,

So I got ahold of one of our Linux laptops to test this out and everything worked fine with your launch file on my device.  The argument "<arg name="baudrate"    default = "**912**600" />" in the launch file you provided looks like it has a typo in the baudrate, but that wouldn't have caused the issue that you were seeing above.

It looks like your Nuvo-8208 may not support baudrates above 115200:

**
[image]
**

If you need the data out of the driver at the rates you've specified, you could use a serial to USB converter or one of our native USB cables.  

Let me know if there is anything else I can help with!

Nathan

Justin-Ribbit, 2021-04-15:
Hi Nathan,

I have confirmed with the manufacturer that the device can indeed support a baudrate of 921600 over its COM1 and COM2 RS-232 ports - please see attached image:
[image]

When using this setting, I cannot connect to the device at all:
```
[ INFO] [1618438356.647400093]: Attempting to open serial port </dev/ttyS0> at <921600> 
[FATAL] [1618438356.903710738]: Error: Failed to communicate with the device.
terminate called after throwing an instance of 'mscl::Error_Communication'
  what():  Failed to communicate with the device.
[gq7/ros_mscl_node-2] process has died [pid 137, exit code -6, cmd /root/ws_ribbit/devel/lib/ros_mscl/ros_mscl_node __name:=ros_mscl_node __log:=/root/.ros/log/8459e084-9d6e-11eb-9d38-78d004293634/gq7-ros_mscl_node-2.log].
log file: /root/.ros/log/8459e084-9d6e-11eb-9d38-78d004293634/gq7-ros_mscl_node-2*.log
```

Any ideas?

nathanmillerparker, 2021-04-15:
Did you try a baudrate of "115200" in the ROS launch file?  As I read the setup utility, that should result in a real baudrate of 921600, since it is multiplied by 8 either the driver or hardware.  If that doesn't work, I'm not sure quite what to do other than try a USB<->Serial converter, USB direct to the device, or adjust your data rate requirements to fit within 115200.

Justin-Ribbit, 2021-04-15:
I tried baud rates ranging from 115200 to 921600 and the same error occurred when operating with the baudrate multiplier. Unfortunately this application requires the need of a very accurate PPS signal so a USB interface is not an option.

I have a few follow up questions so that I can get a better picture of what may be causing these errors:

1. Can you describe what may be happening when I receive a "bad connection" right after the: 
`[ INFO] [1618438356.647400093]: Attempting to open serial port </dev/ttyS0> at <115200>` line?
2. What does it mean when I get the same error after the:
`[ INFO] [1618417610.708676333]: Resetting the filter after the configuration is complete.` line?
3. Initially (before enabling the bios setting) the computer limited the RS-232 COM1 port baudrate to 115200, why could we set the launchfile to 921600 and still communicate with the device without a communication error?

I am just trying to get a sense of the requirements and causes of these errors.

nathanmillerparker, 2021-04-15:
Hi Justin,

Answers inline below:

**Can you describe what may be happening when I receive a "bad connection" right after the:
[ INFO] [1618438356.647400093]: Attempting to open serial port </dev/ttyS0> at <115200> line?**

That means communication with the device could not be established; either the port number or baudrate do not match the physical system.  In your case, as you have a unique UART setup that will multiply the clock, it complicates the issue.  It is important to note that this isn't an issue with the ROS driver or the underlaying MSCL code, if you were to use a simple terminal, it would also have issues communicating.  

**What does it mean when I get the same error after the:
[ INFO] [1618417610.708676333]: Resetting the filter after the configuration is complete. line?**

It means that there isn't an issue connecting to the device (i.e. the port and baudrate are correct), but there is too much data coming over for the baudrate used (the device is commanded to start streaming data right after this line.)  What is happening is the MSCL library defaults to a 0.25 sec timeout on a command and either the device output communication buffer has been filled with enough data to delay the command response beyond that limit or the command response cannot be written to the device buffer due to it being full.  This obviously isn't a desirable effect, but is one of the costs of having command/data over a single UART interface, and requires a bit more care on the user's part to manage data rates with-respect-to the baudrate.

**Initially (before enabling the bios setting) the computer limited the RS-232 COM1 port baudrate to 115200, why could we set the launchfile to 921600 and still communicate with the device without a communication error?**

I don't know... sorry.  The UART on that particular computer operates differently than what I am used to.  You'd need to ask the manufacturer why it would seem to work, then not work as the bandwidth utilization exceeds 115200.

nathanmillerparker, 2021-04-28:
Hey Justin.... just checking in to see if you were able to resolve this issue?

Justin-Ribbit, 2021-05-04:
Hello,
Apologies for the delay. Yes, this issues was resolved, but it is not clear why this works. Here are the steps taken to ensure proper operation with my particular setup:

On the device (using SensorConnect on Windows):
- Set the baud rate to 921600
- Remove any published topics
- Restart the device after setup

On the computer:
- In bios, set the COM port of the device to it's "high speed" mode
- Set the baud rate to 115200 (the high speed mode multiplies the clock by a factor of 8, not sure why this works but it does)

ROS driver launch file:
- Ensure that the "save_settings" flag in the launch file is set to "false".
- Set the baudrate in the launch file to 115200

For some reason, this combination of steps works. If you have any insight into this, please let me know. Otherwise this issue can be closed.

nathanmillerparker, 2021-05-04:
Interesting.... my guess is something to do with the underlaying serial port driver using 115200, even though the real connection is at 921600, but I don't know for sure.  I'm glad you found a fix!

closed 2021-05-05 by Justin-Ribbit