LORD-MicroStrain/microstrain_inertial #51: USB Supported for ROS / Linux?

sbillin, 2021-04-23:
Hello, I am trying to use the USB interface with ROS-MSCL on Linux. 

**System Setup**
IMU:  3DM-GX5-25 with MicroStrain's USB cable
Embedded Processor:  NVIDIA Jetson AGX Xavier (ARM64 device)
OS:  Ubuntu 18.04
ROS:  Melodic

I installed the MSCL libraries and compiled ROS-MSCL in my ROS workspace without any problem.  I have the IMU connected to the Jetson via the USB interface and the green status light on the IMU is blinking rapidly.

When I try to run the ROS-MSCL launch script (roslaunch ros_mscl microstrain.launch) to connect to the IMU I encounter a "Device Disconnected" problem:

[image]

I believe this happens because the launch script is trying to connect to the IMU via serial port, and I don't see any input option to connect via USB.  So my questions are:

**Question 1)**  Is the USB interface supported for ROS-MSCL on Linux?  If so, how do I configure the launch script to use USB?

**Question 2)**  If the answer to Question 1 is no, then is the USB interface supported by MSCL on Linux?  (i.e., is USB supported on Linux but it just hasn't been integrated into the ROS interface?)  If so, do you have any advice for how to go about using the USB over ROS?

Any guidance you can offer is greatly appreciated!

kbhakt, 2021-04-23:
When I've used ROS and Linux I have had to make sure I have done:

chmod +x /dev/ttyACM0 to give it permissions to read over the port

Also sometimes I just have to launch the microstrain.launch file multiple times before it recognizes, especially when using multiple IMUs at once.

Hope this helps.

mglord, 2021-04-23:
Thanks @kbhakt!

Double checking port permissions tends to resolve the issue for most, but definitely let us know if that doesn't work for you!

Regarding USB vs serial: MSCL handles USB and serial connection the same way, so the same class is used for both (`SerialConnection` class, `Connection::Serial()` to instantiate). The main difference when connected via serial port (not USB) is that the specified baud rate must match what's configured on the device. We have seen some cases where Linux is picky about the specified baud rate over USB, but that seems to be a trait of the environment rather than the device. If setting up permissions doesn't help, try specifying a baud rate of 921600.

Another customer was using an RS-232 to USB adaptor with the same device you have (NVIDIA Jetson AGX Xavier) and ran into some issues with the adapter driver ([MSCL issue 153](https://github.com/LORD-MicroStrain/MSCL/issues/153) | [comment](https://github.com/LORD-MicroStrain/MSCL/issues/153#issuecomment-816473999)). I'm unfamiliar with this specific device so am unsure if similar issues can occur with a USB (not adaptor) connection.

**Other things to try:**

From your screenshot it is unclear to me whether the error occurred when attempting to open the port (Connection Error - wrong permissions, incorrect port name) or when first attempting to communicate with the device (Communication Error - no response received from device, wrong baud rate).

**If it's a connection error** you should try adding a add a try/catch around the Connection initialization and read out the associated error code. This code should correspond to a system error code and can be looked up for your operating system for additional info. Ex:
```
try
{
  mscl::Connection c(PORT_NAME, BAUD_RATE);
}
catch (mscl::Error_Connection& e)
{
    cout << "Error: " << e.what() << " (" << e.code() << ")" << endl;
}
```

**If it's a communication error** and the device is streaming data on startup it's possible there is a connection bandwidth issue. If the device is streaming data and running into bandwidth issues command responses are often dropped, even when just trying to read basic device info or send a ping. Setting the node to idle is the best way to re-establish consistent communication. The below can be added directly after instantiating the InertialNode object to ensure successful communication.

You can also use SensorConnect to configure the device not to stream data on startup (Save/Load Settings -> Save as Startup Settings -> Uncheck "Stream data on startup" -> Save). After initial configuration the ROS driver will automatically re-enable streaming regardless of this setting.

```
// attempt set to idle multiple times in case responses are dropped due to data streaming
for(int i = 0; i < 3; i++)
{
  try
  {
    m_inertial_device->setToIdle();
    break; // stop trying if successful
  }
  catch(const mscl::Error& e){/*ignore*/}
}
```

sbillin, 2021-04-23:
Thank you for the awesome support!

chmod +x /dev/ttyACM0 alone didn't resolve it for me, but that and the other permission descriptions put me on the right track.  My problem is also similar to Issue #10.

Here is what worked for me:
1. Add my user to the dialout group, which has ownership of /dev/ttyACM0
You can see dialout is the ownership group by:  ls -ls /dev/ttyACM0

2. Grant the dialout group permission to read/write to device
sudo chmod ugo+rw /dev/ttyACM0

3. Run launch file, and all is working

And if that doesn't work, can also do (I did this but I think only the group permissions would be sufficient):
sudo chmod +rwx /dev/ttyACM0

My linux knowledge is limited, so forgive me if I'm using any wrong terminology to describe anything here.

Thanks again for your help!

closed 2021-04-23 by sbillin