LORD-MicroStrain/microstrain_inertial #52: Acceleration due to gravity components in X-Y-Z in filtered output

Vineet-Pandey, 2021-04-26:
I am using 3DM-GX5-25 IMU and the Auto-Adaptive Extended Kalman Filter works really well in removing the noise from the raw data. However, it removes the gravity component in acceleration while filtering which is important in  3D SLAM algorithms with LiDARs. Is there a way to get the filtered data but with the gravity components intact? TIA.

nathanmillerparker, 2021-04-27:
Hi Vineet,

I wanted to let you know that I'm working on this.  I have implemented and tested it, but unfortunately we need a firmware update in one of our devices (not the one you are using) to make this work across all devices supported by the driver.  Our firmware team is working on that now and it will hopefully be released in 2 weeks.  If you need a solution earlier than that, I can tell you how to edit the driver locally to get what you want, since it already works for your device.  Just let me know your preferred approach.

Thanks,

Nathan

Vineet-Pandey, 2021-04-28:
Hi Nathan,

Thanks! Could you please let me know what changes should I make to the driver so that I can test it out?

nathanmillerparker, 2021-04-28:
Sure.... here is the diff (essentially, replaces linear accel data quantity out of the device with the compensated accel data quantity):

@@ -591,7 +591,7 @@ void Microstrain::run()
             mscl::MipTypes::ChannelField::CH_FIELD_ESTFILTER_ESTIMATED_ATT_UNCERT_QUAT,
             mscl::MipTypes::ChannelField::CH_FIELD_ESTFILTER_ESTIMATED_ANGULAR_RATE,
             mscl::MipTypes::ChannelField::CH_FIELD_ESTFILTER_ESTIMATED_ATT_UNCERT_EULER,
**-            mscl::MipTypes::ChannelField::CH_FIELD_ESTFILTER_ESTIMATED_LINEAR_ACCEL,
+            mscl::MipTypes::ChannelField::CH_FIELD_ESTFILTER_COMPENSATED_ACCEL,**
             mscl::MipTypes::ChannelField::CH_FIELD_ESTFILTER_ESTIMATED_ORIENT_EULER,
             mscl::MipTypes::ChannelField::CH_FIELD_ESTFILTER_HEADING_UPDATE_SOURCE,
             mscl::MipTypes::ChannelField::CH_FIELD_ESTFILTER_NED_RELATIVE_POS};

@@ -1915,7 +1915,7 @@ void Microstrain::parse_filter_packet(const mscl::MipDataPacket &packet)
       }
     }break;
     
**-    case mscl::MipTypes::CH_FIELD_ESTFILTER_ESTIMATED_LINEAR_ACCEL:
+    case mscl::MipTypes::CH_FIELD_ESTFILTER_COMPENSATED_ACCEL:**

Vineet-Pandey, 2021-04-28:
Hi, 
The change seems to be working, Thanks! However, the filter seems to be overshooting more than the actual data many times. Please find the Filtered output vs the raw data for the x component of linear acceleration in the plot below. Is there a way to tune it to reduce the overshoots?
[image]

nathanmillerparker, 2021-04-28:
Hi Vineet,

So, there are a few details to unpack here....

The compensated quantities that are used to generate the "filtered_imu" message (which we should probably rename to "compensated_imu") are generated by the kalman filter by subtracting off estimated bias and scale-factor (if those states are enabled.)  These data quantities are then supplied at the full rate without any additional filtering in the GX5 series devices.

The data quantities that you see in the "imu" message do not have the bias and scale-factors removed, but are passed through a low-pass filter that is set by the requested data rate (or manually overridden for a particular cut-off frequency.)  This additional filtering is why they appear to overshoot less than the quantities above.  The drawback to these quantities is they still have bias and scale-factor baked-in.  

When using the GX5-25, you should be fine to use the "imu" quantities, as the kalman filter does not have sufficient observability to estimate accelerometer bias and scale factor, which is something that can be done on a GNSS/INS, like the GX5-45.  Essentially, the "imu" quantities are probably going to be better (in a noise sense) than the "filtered_imu" quantities on the GX5 AHRS.  Both are valid, just one is filtered more than the other.  Our driver works across the majority of our devices, and the different devices' firmware can have different approaches to filtering the data, so it can be hard to spell all of this out for each device.  

I hope that helps,

Nathan

Vineet-Pandey, 2021-04-28:
So, if my understanding is correct, would it be possible to use the low-pass filter before feeding it to the Kalman filter (Although, this might increase the lag in the compensated o/p)?

nathanmillerparker, 2021-04-28:
Unfortunately, no.  The kalman filter runs at the maximum rate for the device (500 Hz for a GX5-25) and the user can select any decimation of that rate for the output data.  The low-pass filter sits in-between the data generation and output.  One solution would be to put an additional low-pass filter on this particular output quantity, but the GX5 processor is running at a pretty high utilization and (see below) there isn't much added benefit.

For the GX5-25, there would essentially be no benefit to using the filtered quantity for acceleration (although there would be a benefit for angular rate, since the bias is estimated.)  If you are only concerned with the acceleration, I would use the value out of the imu message.   On a different device, this recommendation may change.

closed 2021-04-28 by Vineet-Pandey