LORD-MicroStrain/microstrain_inertial #53: GQ7. nav/relative_pos/odom/pos/pos/position doesn't update.

R-Liebert, 2021-05-04:
Hi! 

We're using the rostopic echo command in melodic trying to read the position values. But they do not seem to update. Orientation updates as expected.

We're using gq7 on a Xavier NX and are trying to get the NED pose (x, y, theta(yaw)). We couldn't find any topic publishing that information. So we figured combining position and heading would do the trick, but we're not sure if this is the most practical way.

nathanmillerparker, 2021-05-04:
The nav/relative_pos/odom topic will provide a relative NED position, but you must set the following launch file parameters:

<param name="publish_relative_position"         value="**true**" type="bool" />
<param name="filter_relative_position_frame"    value="2"  type="int" />
<rosparam param="filter_relative_position_ref"> [**Initial_Lat, Initial_Lon, Initial_height**]</rosparam>

NED position only makes sense relative to some initial starting point, so that is why we have this requirement.

For any of the position values, be it from /nav/odom or nav/relative_pos/odom, they will only update when valid GNSS data is available.  Please make sure you have the antennas securely connected with a clear view of the sky.  It is also important to have the correct heading alignment method selected in the launch file.  The default for the GQ7 is to use dual antenna, which requires your antenna offsets to be correct.  Depending on your application, you could use kinematic alignment (for ground vehicles) or the magnetometer (assuming you have done a calibration after installation in your vehicle.)  These options are all set in the launch file.

You can check the /nav/status message, which will tell you if you've entered full navigation mode "value=4" in which you should see the position updating.  Any value less than 4 on the GQ7 means you do not have all the info required to initialize (likely either heading or no GNSS.)

Finally, you can check the status of the individual GNSS modules by checking either gnss1/fix or gnss2/fix.

Let me know if that gets you what you need!

Nathan

R-Liebert, 2021-05-04:
Thanks for the quick reply! 

That do actually explain everything, considering we were located in a university basement. We'll perform the next test outside or atleast above ground level.  

I consider this issue solved and will close the thread. 

Robin

closed 2021-05-04 by R-Liebert