LORD-MicroStrain/microstrain_inertial #56: How to capture and set acceleromter and gyro bias on start up?

JTomasi, 2021-05-13:
Hello,

I was hoping you could help me understand the process of acquiring and setting the IMU biases when using the ROS driver. I have looked over the source code and noticed that there are several services available for getting and setting the accelerometer and gyroscope biases. I also noticed a service that captures the gyroscope bias automatically. My ideal use case is that every time I am using the device, I would like to capture and initialize the biases on startup before the fitler begins estimating. I was wondering if you could please clarify the following:

1. Is there any documentation or an example for how to do this?
2. What are the steps required in order to capture the IMU biases and apply them before filtering when using the ROS driver?
3. I was planning on modifying the microstrain_3dm.cpp file with the appropriate service calls before the main data parsing loop begins. Is there an alternative method of acquiring and setting these values on start up?
4. Does the `gyro_bias_capture` service also set the bias values or does it simply save them in a local variable and print them to the screen?

Please let me know if there is any additional information that you require.

nathanmillerparker, 2021-05-14:
Hello... answers inlined:

**Is there any documentation or an example for how to do this?**

I'm not sure what device you are using, but here is the user manual for the GX5-25: https://www.microstrain.com/sites/default/files/applications/files/3dm-gx5-25_user_manual_8500-0012_rev_d.pdf

This manual assumes you are using the old MIP monitor software and not the ROS driver, but some of the text may be helpful. 
You shouldn't really need to set the accel bias and it would probably be difficult for you to do that properly (requires a machined block and flipping the device to get gravity aligned with each of the device axis.)

If you are using an AHRS, the filter will estimate the gyro bias for you and remove it.  The resulting angular rate can be found in the nav/filtered_imu/data topic.  If you're using a GX5-10, you would need to capture/set the gyro bias.

**What are the steps required in order to capture the IMU biases and apply them before filtering when using the ROS driver?**

The device must be still to capture the bias.  The service assumes it is still for 10 seconds (note: you could edit that if you wanted, the value that is passed into m_inertial_device->captureGyroBias() is in milliseconds and is 10,000 by default.)  

**I was planning on modifying the microstrain_3dm.cpp file with the appropriate service calls before the main data parsing loop begins. Is there an alternative method of acquiring and setting these values on start up?**

You could use the underlaying MSCL calls that you see in the services if you want.

**Does the gyro_bias_capture service also set the bias values or does it simply save them in a local variable and print them to the screen?**

The gyro bias capture writes the bias values to the device RAM, so they are valid until power cycled.  

Hope that helps,

Nathan

JTomasi, 2021-05-14:
Hi Nathan,

Thank you for the detailed response. I forgot to mention that I am using a 3DM-GQ7. Based on your reply, my understanding is that the built in filter will estimate the bias and remove it, so there is no need to capture it manually, correct?

nathanmillerparker, 2021-05-14:
That's correct!  No need to capture the biases with the GQ7 as the filter will do that for you.  Just note, the imu/data topic will still have the bias present, but the nav/filtered_imu/data topic will have it removed.  If you don't need IMU data, just the filter pos/vel/attitude output, this doesn't matter, but I wanted to be clear.

JTomasi, 2021-05-14:
Okay, thank you for the clarificaiton. I do need to use IMU data, but I require IMU data that has not been corrected with any external measurement sources so I am trying to determine if I should use the imu/data topic or the nav/filtered_imu/data topic.

How is the filtered IMU data created? Is it simply the IMU data with the bias terms and gravity vector removed, or is it also corrected with, for example, GNSS measurements? Am I missing any other considerations?

nathanmillerparker, 2021-05-14:
Filtered IMU data has the Kalman filter's estimate of bias removed.  Indirectly, GNSS measurements would affect this as they affect all of the filter states.  You can use the imu/data as it is not affected by the filter at all.

JTomasi, 2021-05-14:
A few more clarifying questions:

1. Is it possible to extract the estimated IMU biases from the filter using the ROS driver so that I can manually remove them from the "raw" imu/data?
2. Are there "discontinuities" introduced in the nav/filtered_imu/data due to the corrections by the GNSS? By discontinuities, I am referring to drastic changes in the output IMU values (linear acceleration and angular velocity) when GNSS corrections are applied that do NOT appear in the imu/data measurements?
3. Can you please elaborate on how the calibration of the system is performed? Does calibration ever need to be performed manually? For example, how does the filter estimate the biases - do I need to ensure that the unit is stationary during start up?
4. What config parameters need to set in order to ensure that the device enters "Full navigation" mode (filter state = 4) while running? I have been collecting data with it using the launch file provided in the repo and it only ever enters "vertical gyro" mode (filter state = 2). On the device documentation, it mentions that:
> The mode that the Navigation Filter operates in is determined by the initial conditions provided to the Navigation Filter through the Navigation Filter Initialization (0x0D,0x52) command.

What settings need to be modified so that it enters full navigation mode?

5. Can you confirm that the 3DM-GQ7 does not have a "filter dynamics mode" setting?

nathanmillerparker, 2021-05-14:
**Is it possible to extract the estimated IMU biases from the filter using the ROS driver so that I can manually remove them from the "raw" imu/data?**

You can't read them directly with the current ROS driver.

**Are there "discontinuities" introduced in the nav/filtered_imu/data due to the corrections by the GNSS? By discontinuities, I am referring to drastic changes in the output IMU values (linear acceleration and angular velocity) when GNSS corrections are applied that do NOT appear in the imu/data measurements?**

As the filter converges, there will be "jumps" in the bias estimates, that will translate to "jumps" in the IMU data.  Whether these are considered drastic is really a function of how you use the data.  The corrections tend to be small in magnitude.

**Can you please elaborate on how the calibration of the system is performed? Does calibration ever need to be performed manually? For example, how does the filter estimate the biases - do I need to ensure that the unit is stationary during start up?**

The filter continually estimates the bias as it receives measurements, it's just part of the calculations in the Kalman filter.  You do not need to stay still and it can be beneficial to be moving around.  These biases are typically pretty small, so they do not have a dramatic effect on the filter performance.

**What config parameters need to set in order to ensure that the device enters "Full navigation" mode (filter state = 4) while running? I have been collecting data with it using the launch file provided in the repo and it only ever enters "vertical gyro" mode (filter state = 2). On the device documentation, it mentions that:
The mode that the Navigation Filter operates in is determined by the initial conditions provided to the Navigation Filter through the Navigation Filter Initialization (0x0D,0x52) command.**

You need GNSS data coming into the unit (so a clear view of the sky) and you need to have one of the heading alignment methods selected in the launch file (shown below.)  You will almost always want auto position, vel, and attitude initialization.  The alignment selector is a bitfield, the default is dual antenna.  For dual antenna to work, you need to setup your antenna offsets correctly in the launch file.  For kinematic heading alignment your application needs to move like a car (velocity in the direction of travel) and your sensor2vehicle frame transformation needs to be setup correctly in the launch file.  For magnetometer alignment, you need to magnetically calibrate the device as it is attached to your vehicle, but using our SensorConnect software (which runs the calibration.)  The easiest is dual antenna and that is why it is the default.

    **<param name="filter_init_condition_src"              value="0"     type="int" />
    <param name="filter_auto_heading_alignment_selector" value="1"     type="int" />**
    <param name="filter_init_reference_frame"            value="2"     type="int" />

**What settings need to be modified so that it enters full navigation mode?**

Described above

**Can you confirm that the 3DM-GQ7 does not have a "filter dynamics mode" setting?**

That is correct, it does not have this setting.

JTomasi, 2021-05-14:
Thank you again for the detailed response!

A further setting that I am interested in is the RTK status message. I have udpated my launch file so that this message is now being published. I am hoping to use RTK whenever possible while the filter is running and I am particularly interested in knowing when the device is using RTK and when it is not using RTK. Can you provide some clarity on how to interpret the RTK status message to find this out? Is there a particular field that indicates this? There are several fields which are outlined in the status message and I have been trying to decipher some of them using the documentation for the GQ7 here: [GQ7 documentation](https://s3.amazonaws.com/files.microstrain.com/GQ7+User+Manual/user_manual_content/RTK/Status%20Flags.htm) but it is not entirely clear if there is a single field that I should pay attention to, or if there are multiple fields that I need to interpret. 

I have run a few tests and found that instances where the /gq7/gnssX/fix position covariance increases dramatically are likely instances when the RTK signal is no longer available. I can't really determine which status flag this corresponds to in the RTK status message however. 

Any insight you could provide would be appreciated. Thanks for all of your help!

nathanmillerparker, 2021-05-17:
I'm going to add a bit better info to the driver to satisfy this need.  In theory, you could look at the position covariance and if you are getting RTK-level accuracy, you can assume RTK is being used.  The RTK status message just tells you if you are receiving RTK corrections, not if they are being used by the filter.  Let me add and commit that change for you.

JTomasi, 2021-05-18:
Ok perfect. Please let me know when this change has been implemented.

nathanmillerparker, 2021-05-19:
Just an update on this... I have the change made to the ROS driver, but am waiting on the MSCL library to be released prior to my commit.  I'll let you know when it's live.

nathanmillerparker, 2021-05-19:
Ok.. this has been updated.  You want to look for the following Boolean values from each "/gq7/GNSS#/aiding_status" topic to determine RTK fix (done independently for each receiver):
bool    has_position_fix = TRUE
bool    differential_corrections = TRUE
bool    integer_fix = TRUE 

Hope that helps!

Nathan

JTomasi, 2021-05-26:
Great, thank you. Do you have any indication when the release version of the ROS driver may have these updates?

And my final question for this issue: is the heading reported on the topic "gq7/nav/heading" a measure of true heading or magnetic heading? I can't seem to find any information in the documenation related to this value. Can you also inform me of the difference between the "gq7/nav/heading" and "gq7/nav/heading_state" topics?

nathanmillerparker, 2021-05-26:
The updates should be available now... let me know if you don't see them.

The heading for the GQ7 is reported as true heading.  You will want to pull from gq7/nav/heading... the other is a measurement into the Kalman filter, but it is for the GX5.  I'll add it to the next release to remove it for the GQ7.

JTomasi, 2021-06-11:
Great, thanks for the all the assitance.

closed 2021-06-11 by JTomasi