LORD-MicroStrain/microstrain_inertial #57: Time sync between 3DM-GX5-45 and Velodyne HDL32E

lincob, 2021-05-17:
Hello,
I'm trying to PPS sync both devices, GPS timestamps from the IMU messages appear to be working fine, PPS output from IMU is detected on the Lidar input, but in order to write correct timestamps the velodyne unit also need to be fed a NMEA GPRMC message at 1Hz through TCP. Is it possible with the current version of the driver ?
Thanks.

nathanmillerparker, 2021-05-18:
The driver itself does not output NMEA, but you could write a small converter program to do this given the topics available.

lincob, 2021-05-18:
Thanks Nathan... I've tried that... with a limited amount of success, almost no success at all tbh 😉
Any help would be appreciated. I've seen a lot of packages converting NMEA strings to NavSatFix messages but not the other way around. Thanks.

nathanmillerparker, 2021-05-18:
I was able to get ahold of some Python code that one of our engineers uses to interface to the Velodyne for his testing... hopefully it helps:

def gps_time_to_utc(gps_week: int, gps_seconds: int, leap_seconds: int = 18):
    datetimeformat = "%Y-%m-%d %H:%M:%S"
    epoch = datetime.datetime.strptime("1980-01-06 00:00:00", datetimeformat)
    elapsed = datetime.timedelta(days=(gps_week * 7), seconds=(gps_seconds - leap_seconds))
    return epoch + elapsed

def build_nmea_packet_from_gps_time(gps_week: int, gps_seconds: int):
    utc_time = gps_time_to_utc(gps_week, gps_seconds)
    fix_time_string = utc_time.strftime("%H%M%S")
    fix_day_string = utc_time.strftime("%d%m%y")
    receiver_status = "A"
    lat = "0000.000"
    lat_dir = 'N'
    lon = "0000.000"
    lon_dir = 'E'
    speed_over_ground = '000.0'
    track_direction = '000.0'
    magnetic_variation = "000.0"
    magnetic_variation_dir = 'W'
    nmea_packet = pynmea2.RMC("GP", "RMC", (fix_time_string, receiver_status, lat, lat_dir, lon, lon_dir, speed_over_ground, track_direction, fix_day_string, magnetic_variation, magnetic_variation_dir))
    packet_string = str(nmea_packet) + "\n\r"
    return packet_string

closed 2022-01-30 by github-actions[bot]

lincob, 2024-03-06:
> I was able to get ahold of some Python code that one of our engineers uses to interface to the Velodyne for his testing... hopefully it helps:

Hey Nathan,

It's been a while, caught up in plain old life in general, sorry for the late reply, I'm only able to get my hand back on the project now.

I've adapted the code to my needs, I paste it below for reference.

I'm connected to the lidar though ethernet, I'm subscribing to the GPS time reference, building the nmea sentence and sending it to ethernet socket.
I've dropped the leap seconds part since I'm building the sentence directly from GPS time instead of UNIX ROS time. Thanks for your help. Cheers.

#!/usr/bin/env python

import rospy
from std_msgs.msg import String
from sensor_msgs.msg import TimeReference
from datetime import datetime, timedelta
import pynmea2
import socket

s = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
s.connect(("192.168.0.201",10110))

def build_nmea_packet_from_time_ref(ts):
    utc_time = datetime.utcfromtimestamp(ts)# + timedelta(seconds=1)
    fix_time_string = utc_time.strftime("%H%M%S")
    fix_day_string = utc_time.strftime("%d%m%y")
    receiver_status = "A"
    lat = "0000.000"
    lat_dir = 'N'
    lon = "0000.000"
    lon_dir = 'E'
    speed_over_ground = '000.0'
    track_direction = '000.0'
    magnetic_variation = "000.0"
    magnetic_variation_dir = 'W'
    nmea_packet = pynmea2.RMC("GP", "RMC", (fix_time_string, receiver_status, lat, lat_dir, lon, lon_dir, speed_over_ground, track_direction, fix_day_string, magnetic_variation, magnetic_variation_dir))
    packet_string = str(nmea_packet) + "\n\r"
    return packet_string

def callback(data):
    rospy.loginfo("PASS")
    rospy.loginfo(data.time_ref.secs)
    data = build_nmea_packet_from_time_ref(data.time_ref.secs)
    print(data)
    s.send(data.encode())

def send_gprmc():

    # In ROS, nodes are uniquely named. If two nodes with the same
    # name are launched, the previous one is kicked off. The
    # anonymous=True flag means that rospy will choose a unique
    # name for our 'listener' node so that multiple listeners can
    # run simultaneously.
    rospy.init_node('send_gprmc', anonymous=True)

    rospy.Subscriber("gx5/gnss1/time_ref", TimeReference, callback)
    print("SUBSCRIBED")
    # spin() simply keeps python from exiting until this node is stopped
    rospy.spin()

if __name__ == '__main__':
    send_gprmc()

lincob, 2024-03-06:
@jaeseok4104 sorry I don't quite understand your question ?