LORD-MicroStrain/microstrain_inertial #58: 3DM-GQ7 ROS Driver Data Recieved in Bursts

JTomasi, 2021-05-18:
Hello,

I am inquiring about how the data that the 3DM-GQ7 generates is processed by the ROS driver. It seems that the data is processed/received in ROS in bursts. I wanted to confirm if this is intended behaviour or if there is perhaps something incorrect with my configuration. My exact issue is outlined below:

I have collected a large rosbag containing several topics including:
- /gq7/imu/data
- /gq7/nav/odom

Both of these topics were set to a data rate of 100 Hz. I am using the rosbag for testing a simple node that is running at 100Hz and noticed that the messages on the /gq7/nav/odom/ topic are quite "bursty". To verify this, I wrote a very simple node that simply loops at 100Hz and prints to the terminal during each loop. The node also has two callback functions for each topic. During each callback, I simply print the message header seq to the terminal. The output of this node is below:

[image]

As you can see, the IMU data is read in at the expected 100Hz, but the filtered nav data seems to be processed by the ROS driver in bursts, i.e., there is no new nav message during some node loops, while there are a large number during others.

As an additional check, I plotted the difference in time between the /gq7/nav/odom message timestamps that were collected at 100 Hz. The nominal difference should be around 0.01s:

[image]

Seen here, the average time difference between messages is around 0.01s but there is quite a bit of variation, which may be causing the above timing issues.

Please let me know if you have any insights or can confirm if the messages that the ROS driver sends are intended to be sent in bursts. Thanks for your help.

nathanmillerparker, 2021-05-18:
Hello,

A couple of things are probably going on here:

1) It is likely that you are using the "collected timestamp" and not the "device-generated timestamp."  You can switch to the device timestamp using the following launch file setting: 

&lt;param name="use_device_timestamp" value=**"true"** type="bool" /&gt;

One issue with doing this though, is the device time is referenced to GPS time, so any syncing you need to do with other systems would be affected, assuming they need local PC time.

2) Are you using USB?  If you are, the host computer is in charge of polling the data from the USB device (GQ7.)  This typically causes "bursty" data and is affected by processor load.  There isn't much we can do about this one as the host is in control.  You do get all the data and bandwidth isn't much of a concern, but the data can come on chunks.

3) Are you using the UART?  If so, you might be overrunning the available bandwidth.  I would recommend checking the requested data rates / data quantities against the baudrate.  This is the better option for control systems that need data to arrive with minimum latency (and bursting), but bandwidth can become a concern.

Let me know if you need any additional info.

Nathan

JTomasi, 2021-05-18:
Hey Nathan,

Thanks again for the detailed reponse.

I will try using the "device-generated timestamp" setting for future data collection runs. In regards to your other points, I am using UART with the maximum allowable baudrate of 921600 so I don't believe bandwidth is a concern.

Another plot that I can share is the plot of the time difference between sequential ROSbag message timestamps (this is different from the header timestamp that I shared in the message above). This plot shows the difference in time between sequential /gq7/nav/odom/ messages that the rosbag recorded from the driver. To me, this suggests that the driver bunches the data together before outputting it. Can you provide any insight here?

[image]

nathanmillerparker, 2021-05-18:
Could you provide your launch file so I can try to recreate this issue?

Thank you

JTomasi, 2021-05-18:
Sure, please see below. Note that the baudrate is set to 115200 as per #47 

```
<?xml version="1.1"?>
<launch>

  <!-- Standalone example launch file for GX3, GX4, GX/CX5, RQ1 and GQ7 series devices-->
  <!-- Note: Feature support is device-dependent and some of the following settings may have no affect on your device. --> 
     <!--  Please consult your device's documentation for supported features -->

  <!-- Declare arguments with default values -->
  <arg name="name"        default = "gq7" />
  <arg name="port"        default = "/dev/ttyS0" />
  <arg name="baudrate"    default = "115200" />
  <arg name="debug"       default = "false" />
  <arg name="diagnostics" default = "true" />

  <!-- ****************************************************************** -->
  <!-- Microstrain sensor node -->
  <!-- ****************************************************************** -->
    
  <node name="ros_mscl_node" pkg="ros_mscl" type="ros_mscl_node" output="screen" ns="$(arg name)">

    <!-- ****************************************************************** -->
    <!-- General Settings -->
    <!-- ****************************************************************** -->
    
    <param name="port"     value="$(arg port)"     type="str" />
    <param name="baudrate" value="$(arg baudrate)" type="int" />
    
    <!-- Controls if the driver-defined setup is sent to the device
          false - The driver will ignore the settings below and use the device's current settings
          true  - Overwrite the current device settings with those listed below
     -->
     
    <param name="device_setup" value="true" type="bool" />
    

    <!-- Controls if the driver-defined settings are saved 
          false - Do not save the settings
          true  - Save the settings in the device's non-volatile memory
     -->
     
    <param name="save_settings" value="false" type="bool" />
    

    <!-- Controls if the driver creates a raw binary file
          false - Do not create the file
          true  - Create the file

          Notes: 1) The filename will have the following format - 
                    model_number "_" serial_number "_" datetime (year_month_day_hour_minute_sec) ".bin"
                    example: "3DM-GX5-45_6251.00001_20_12_01_01_01_01.bin"
                 2) This file is useful for getting support from the manufacturer
     -->
     
    <param name="raw_file_enable" value="true" type="bool" />

    <!-- (GQ7 only) Controls if the driver requests additional factory support data to be included in the raw data file
          false - Do not request the additional data
          true  - Request the additional channels

          Notes: Including this feature increases communication bandwidth requirements... for serial data connections,
                 please ensure the baudrate is sufficient for the added data channels.
     -->

    <param name="raw_file_include_support_data" value="true" type="bool" />

    <!-- The directory to store the raw data file (no trailing '/')-->

[... 265 lines cut ...]
    <param name="filter_gnss_antenna_cal_max_offset" value="0.1"   type="double" />
  </node>

  <!-- ****************************************************************** -->
  <!-- Diagnostics -->
  <!-- ****************************************************************** -->
    
  <group if="$(arg diagnostics)">

    <!-- Diagnostic Aggregator for robot monitor usage -->
    <node pkg="diagnostic_aggregator" type="aggregator_node" name="imu_diagnostic_aggregator">
      <rosparam command="load" file="$(find ros_mscl)/config/diagnostic_analyzers.yaml" />
    </node>
  </group>

  <!-- ****************************************************************** -->
  <!-- Set the debug level to debug -->
  <!-- ****************************************************************** -->
 
  <group if="$(arg debug)">
    <env name="ROSCONSOLE_CONFIG_FILE"
       value="$(find ros_mscl)/config/custom_rosconsole.conf"/>
  </group>
</launch>
```

nathanmillerparker, 2021-05-18:
Ok... I thought that the following setting might be the case:

<param name="raw_file_include_support_data" value=**"true"** type="bool" />

So, this has a couple of effects on the datastream that need some discussion:

1) More data is added to the filter (nav) data packets and the underlaying MIP binary protocol has a size limitation of 261 bytes per packet.  The additional data elements cause the filter data to be split across 2 MIP packets.  That means, the driver will process 2 MIP packets per 10 mS interval.  The effect is you will get back-to-back sequence numbers (and minimal dt in both plots that you show above.)  This doesn't totally explain what you are seeing above, but part of it.

2) There is a known bug in the GQ7 firmware that requires: the device to be connected via UART, a lot of satellites in view, and the support data enabled.  In this configuration, you may see dropped/missing/delayed data due to the GQ7 UART output buffer size being insufficient for the amount of data each GNSS sends.  There is a fix for this issue and we are currently testing it with a release target date of 6/1/21.  This would explain the band of packets above 0.04s dt in the rosbag image above.

I would suggest not including the support data unless you need to to get simulation/data analysis support from us..... especially when doing real-time control with the device.  The reason is, there is a significant amount of additional data (all of the raw satellite ranging and ephemeris info) that comes over when enabled.  This comes out all at once at 2 hz for each of the GNSS modules (plus the RTK, if receiving corrections)... and it is enough data to push around the reception time of the other data (IMU and NAV.)

I also noticed that in your first image "Node Timer Callback" seems to be called at 100 Hz.  The spin rate of the node should be 200 Hz. This is something I'll need to check on tomorrow (I'm away from the office today.)

My guess is that if you disable "raw_file_include_support_data", you will see your timing issues mostly clear up, if not clear up completely.  I'll take a deeper look at it tomorrow as I will be finishing up the addition to the driver to support your other request wrt RTK status.

Let me know if that helps,

Nathan

JTomasi, 2021-05-18:
I should also mention that I am currently using

JTomasi, 2021-05-18:
Hi Nathan,

Apologies but I doubled checked and I had the incorrect version of the launch file. The setting you are referring to was actually set to false when I collected this data. The above information is very useful however.

I have posted the correct launch file below. One issue that I discovered was that I had incorrectly set the following parameter:
` <param name="filter_auto_heading_alignment_selector" value="0"     type="int" />`

I missed the comment indicating that this was a bitfield and not an enum - I have since update this but perhaps this also contributed. Please let me know if there is anything else that you notice.

```
<?xml version="1.1"?>
<launch>

  <!-- Standalone example launch file for GX3, GX4, GX/CX5, RQ1 and GQ7 series devices-->
  <!-- Note: Feature support is device-dependent and some of the following settings may have no affect on your device. --> 
     <!--  Please consult your device's documentation for supported features -->

  <!-- Declare arguments with default values -->
  <arg name="name"        default = "gq7" />
  <arg name="port"        default = "/dev/ttyS0" />
  <arg name="baudrate"    default = "115200" />
  <arg name="debug"       default = "false" />
  <arg name="diagnostics" default = "true" />

  <!-- ****************************************************************** -->
  <!-- Microstrain sensor node -->
  <!-- ****************************************************************** -->
    
  <node name="ros_mscl_node" pkg="ros_mscl" type="ros_mscl_node" output="screen" ns="$(arg name)">

    <!-- ****************************************************************** -->
    <!-- General Settings -->
    <!-- ****************************************************************** -->
    
    <param name="port"     value="$(arg port)"     type="str" />
    <param name="baudrate" value="$(arg baudrate)" type="int" />
    
    <!-- Controls if the driver-defined setup is sent to the device
          false - The driver will ignore the settings below and use the device's current settings
          true  - Overwrite the current device settings with those listed below
     -->
     
    <param name="device_setup" value="true" type="bool" />
    

    <!-- Controls if the driver-defined settings are saved 
          false - Do not save the settings
          true  - Save the settings in the device's non-volatile memory
     -->
     
    <param name="save_settings" value="false" type="bool" />
    

    <!-- Controls if the driver creates a raw binary file
          false - Do not create the file
          true  - Create the file

          Notes: 1) The filename will have the following format - 
                    model_number "_" serial_number "_" datetime (year_month_day_hour_minute_sec) ".bin"
                    example: "3DM-GX5-45_6251.00001_20_12_01_01_01_01.bin"
                 2) This file is useful for getting support from the manufacturer
     -->
     
    <param name="raw_file_enable" value="false" type="bool" />

    <!-- (GQ7 only) Controls if the driver requests additional factory support data to be included in the raw data file
          false - Do not request the additional data
          true  - Request the additional channels

          Notes: Including this feature increases communication bandwidth requirements... for serial data connections,
[... 272 lines cut ...]
    <param name="filter_gnss_antenna_cal_max_offset" value="0.1"   type="double" />
  </node>

  <!-- ****************************************************************** -->
  <!-- Diagnostics -->
  <!-- ****************************************************************** -->
    
  <group if="$(arg diagnostics)">

    <!-- Diagnostic Aggregator for robot monitor usage -->
    <node pkg="diagnostic_aggregator" type="aggregator_node" name="imu_diagnostic_aggregator">
      <rosparam command="load" file="$(find ros_mscl)/config/diagnostic_analyzers.yaml" />
    </node>
  </group>

  <!-- ****************************************************************** -->
  <!-- Set the debug level to debug -->
  <!-- ****************************************************************** -->
 
  <group if="$(arg debug)">
    <env name="ROSCONSOLE_CONFIG_FILE"
       value="$(find ros_mscl)/config/custom_rosconsole.conf"/>
  </group>
</launch>
```

nathanmillerparker, 2021-05-18:
Ok... well, let me take a look at this issue when I'm in the office tomorrow.  The filter_auto_heading_alignment_selector shouldn't affect this issue, but yes, that should be a "1" if you are wanting to initialize with dual antenna.

nathanmillerparker, 2021-05-19:
Ok, so I did a somewhat similar test with the exception that my nav data is at 10 hz (my baudrate is limited to 115200 in the setup that I'm using.)  I get what I expect would be decent results (shown below.)  The time that I am plotting is the arrival time of the packet.  The computer (actually a virtual machine) isn't doing any other calculations.  In your test was there something that could have been interrupting the processing of the GQ7 data?

[image]

JTomasi, 2021-05-20:
Hi Nathan,

I performed an additional test and am still getting the same result. Here, I collected both the IMU and nav/odom data at 200 Hz. I am running my node loop at 100Hz to emphasize the bursts in the nav messages. Note that the dt value is calculated based on the message header timestamp, which seems to be correct for both the IMU data and the nav/odom data. The same issue with how the nav data is passed through the driver can been - the nav data seems to be read in bursts. 

As an aside, I am currently using the vrsion of the ROS MSCL driver found [here](https://github.com/LORD-MicroStrain/ROS-MSCL/tree/618a61bcfbbe19502e139bf74b1d27a4f5421f9c). Is there any chance that this could be the cause of the issues? I plan on updating to the newest driver once the version with RTK that you are working on is completed.

[image]

nathanmillerparker, 2021-05-20:
The newest version of the driver was committed last night and supports the RTK info that you need (commented on the other support thread.) 

I don't believe anything changed in parsing of the MIP packets between versions.  Our devices interleave the IMU/NAV/GNSS MIP packets and the MSCL library just queues them up after verifying the checksum.  The driver picks up these queued results and distributes them as you've seen in the driver code you've edited. 

Can you share the code that you edited to generate the "Filtered nav message ####, ...." info message?  
Also, your launch file says your running at 115200, but given your setup (if I remember correctly), you are really running at 921600, right?

Justin-Ribbit, 2021-05-21:
Below is the beginning of the nav message callback function. I simply take the message, a subtract the timestamp from the previous timestamp.

You are correct, we are running at 921600 but our particular setup requires the config to be set to 115200.
```
void StateEstimationNodelet::navFilteredCB(
    const nav_msgs::OdometryConstPtr &msg){
  ROS_INFO_STREAM("\tFiltered nav message " << msg->header.seq << ", dt: " << (msg->header.stamp-temp_prev_time_nav).toSec());
  temp_prev_time_nav_ = msg->header.stamp;
```

JTomasi, 2021-05-21:
Below is the beginning of the nav message callback function. I simply take the message, a subtract the timestamp from the previous timestamp.

You are correct, we are running at 921600 but our particular setup requires the config to be set to 115200.
```
void StateEstimationNodelet::navFilteredCB(
    const nav_msgs::OdometryConstPtr &msg){
  ROS_INFO_STREAM("\tFiltered nav message " << msg->header.seq << ", dt: " << (msg->header.stamp-temp_prev_time_nav).toSec());
  temp_prev_time_nav_ = msg->header.stamp;
```

nathanmillerparker, 2021-05-21:
Is it possible this is a consumer-side issue?  I thought you were printing the info from our driver, not a different node listening to the topic.  Have you tried checking our driver side, something like the following:

Starting at line 1722 in the latest driver:

//Handle time 
uint64_t time = packet.collectedTimestamp().nanoseconds();    

//Check if the user wants to use the device timestamp instead of PC collected time  

if(packet.hasDeviceTime() &amp;&amp; m_use_device_timestamp)   
{     
     time = packet.deviceTimestamp().nanoseconds();  
}

//TEST CODE START
static uint64_t prev_time = 0;

ROS_INFO("NAV Packet id = %d, dt = %f", m_filter_valid_packet_count, ((double)time-(double)prevt_time)/1.0e9);
prev_time = time;

//TEST CODE END

//Filtered IMU timestamp and frame 
m_filtered_imu_msg.header.seq        = m_filter_valid_packet_count; 
m_filtered_imu_msg.header.stamp    = ros::Time().fromNSec(time);  
m_filtered_imu_msg.header.frame_id = m_filter_frame_id;

JTomasi, 2021-06-11:
Closing this issue as the recent driver updates have rectified this issue.

closed 2021-06-11 by JTomasi

edwardribbit, 2021-07-07:
Adding to the knowledge base here, this issue was related to ROS1 tcp message transport. Adding [transport hint](http://docs.ros.org/en/noetic/api/roscpp/html/classros_1_1TransportHints.html#a03191a9987162fca0ae2c81fa79fcde9) tcpNoDelay when initialising a [subscriber](http://docs.ros.org/en/noetic/api/roscpp/html/classros_1_1NodeHandle.html#a317fe4c05919e0bf3fb5162ccb2f7c28) causes the subscriber callback to be called at a constant frequency.

Aussie25, 2021-09-22:
> Hello,
> 
> I am inquiring about how the data that the 3DM-GQ7 generates is processed by the ROS driver. It seems that the data is processed/received in ROS in bursts. I wanted to confirm if this is intended behaviour or if there is perhaps something incorrect with my configuration. My exact issue is outlined below:
> 
> I have collected a large rosbag containing several topics including:
> 
>     * /gq7/imu/data
> 
>     * /gq7/nav/odom
> 
> 
> Both of these topics were set to a data rate of 100 Hz. I am using the rosbag for testing a simple node that is running at 100Hz and noticed that the messages on the /gq7/nav/odom/ topic are quite "bursty". To verify this, I wrote a very simple node that simply loops at 100Hz and prints to the terminal during each loop. The node also has two callback functions for each topic. During each callback, I simply print the message header seq to the terminal. The output of this node is below:
> 
> [image]
> 
> As you can see, the IMU data is read in at the expected 100Hz, but the filtered nav data seems to be processed by the ROS driver in bursts, i.e., there is no new nav message during some node loops, while there are a large number during others.
> 
> As an additional check, I plotted the difference in time between the /gq7/nav/odom header timestamps that were collected at 100 Hz. The nominal difference should be around 0.01s:
> 
> [image]
> 
> Seen here, the average time difference between messages is around 0.01s but there is quite a bit of variation, which may be causing the above timing issues.
> 
> Please let me know if you have any insights or can confirm if the messages that the ROS driver sends are intended to be sent in bursts. Thanks for your help.

Hi @JTomasi ,
Were you able to resolve the burst issue and did you try higher imu data frequency. And may I know how to plot the difference in time between the message header timestamps. 
Thanks