LORD-MicroStrain/microstrain_inertial #62: Microstrain 3DM-GX5-25 delay and latency

Jaetriel, 2021-06-02:
Hey, I've gone through the documentation of the 3DM-GX5-25 IMU but I couldn't find any info about delay/latency/timestamp resolution of any of the sensors or the filtered output either. I would like this info, if it's available somewhere, so I could achieve better timestamp syncing between my IMU and a Lidar sensor running PTP sync with my host machine.

ianmooreparker, 2021-06-02:
Hi @Jaetriel , thanks for your question. Using the internal timestamp is your best option. The collected timestamp will be based on the system time which is variable dependent on your system. We have not completed testing to measure the latency of the internal timestamp relative to the physical input of the internal sensor.

Jaetriel, 2021-06-07:
Hi @ianmooreparker thanks for the answer. Using the IMU's internal timestamp( so param `use_device_timestamp` set to true ) I get timestamps on my messages which look like this: 

> header: 
  seq: 13104
  stamp: 
    secs: 315970346
    nsecs:  56000000

and when not using the device timestamp, I get the current time, so something like:

> header: 
  seq: 2854
  stamp: 
    secs: 1623066569
    nsecs: 205010688

which should be what I need for my project, since the Lidar sensor I am using is synced to the host PC's clock as well. Is there any particular disadvantage to not using the device timestamp?

ianmooreparker, 2021-06-07:
Hi @Jaetriel the main disadvantage to not using the IMU received timestamp would be the latency in transmission. 
- If transmitting the Received Timestamp over USB, the dT between successive packets for received packets is going to be erratic 
- Using serial improves upon this challenge
- If using the GX5-25 device generated timestamp, the time of arrival at the PC will be used as the timestamp
Suggestions:
- Use a serial connection (not serial to USB) 
- If your PC has a native timestamp, use that
Does this answer your question?

Jaetriel, 2021-06-11:
thanks @ianmooreparker for the clarifications, I'm closing the issue

closed 2021-06-11 by Jaetriel