LORD-MicroStrain/microstrain_inertial #64: RTK Connection Issues

milievski-flyribbit, 2021-06-07:
Hello,

I am seeing an issue when attempting to use RTK with the 3DM-GQ7 unit on version 1.1.4 of the ROS drivers along with MSCL v62.0.0. 

The first very minor issue is that the newly updated `rtk/status` message was not getting published. This is a minor problem that I fix by changing `m_publish_rtk` on [line 536](https://github.com/LORD-MicroStrain/ROS-MSCL/blob/3d7372e8b2341c6b9cb8c898bd125e5afd491195/ros_mscl/src/microstrain_3dm.cpp#L536) to `rtk_dongle_enable` which is set in the microstrain.launch file.

However I am a bit stumped by the second issue. When I launch the ros drivers RTK is unable to connect the image attach is the status of the rtk/status update message as viewed by rqt.
[image]
Along with this error the RTK unit itself  goes from flashing green to rapidly flash blue quickly followed by red. The video attach depicts this flashing. 
https://user-images.githubusercontent.com/77297158/121091786-8e492400-c7b8-11eb-82e2-21dad9a1070f.mp4

The only other part of the code base which was modified was the launch file which I have attached here as a txt file.
[microstrain.txt](https://github.com/LORD-MicroStrain/ROS-MSCL/files/6612204/microstrain.txt)

nathanmillerparker, 2021-06-08:
Ok..  I will fix the driver given your bug report above.

Have you set up your RTK dongle with the SensorCloudRTK service?  The LED states for the dongle are outlined here: https://s3.amazonaws.com/files.microstrain.com/GQ7+User+Manual/user_manual_content/RTK/LED%20States.htm

It seems like your dongle is either not setup or there is an issue with the cell connection.  Please report back and we can proceed from there.

milievski-flyribbit, 2021-06-08:
Thanks for your quick responds.

Yes I can confirm I am getting RTK connection when using Sensor Connect, confirming that both the sensor has been set up as well as there is no cell connection issues.

nathanmillerparker, 2021-06-08:
When you say you have an RTK connection with SensorConnect, does that mean you are using the NTRIP client feature without the dongle or you have successfully connected the dongle to SensorCloudRTK?

milievski-flyribbit, 2021-06-08:
We have been able to successfully connect to sensorcloudRTK and see the correct RTK position via the web portal.

milievski-flyribbit, 2021-06-08:
We believe this to be a more recently introduced bug as we have been able to connect the RTK unit to SensorCloudRTK in the past with MSCL v 61.0.16

nathanmillerparker, 2021-06-08:
Ok, thank you.  That helps me frame the suggested debugging steps.  Let's get the ROS driver out of the loop and use SensorConnect to connect to the device. 

1. Please make sure you have the latest version of SensorConnect installed: https://www.microstrain.com/software/sensorconnect
2. Connect your device and make sure the Aux Port RTCM Handshaking is enabled:

[image]

3. Use the status quick view to get information about your device:

[image]

4. Make sure your sensor has initialized and please screenshot the status quick view for me.  I'm primarily interested in this section, but other info would help as well:

[image]

I'll take a look at this result and get back to you!

milievski-flyribbit, 2021-06-08:
Unfortunately, due to me being physically located away from the GPS unit for the day I will be unable to do this today. I will post an update tomorrow.

Based on my debugging attempts yesterday I have confirmed that the RTK unit status was providing updated to 3DMGQ7-GNSS/INS via the rtk/status. I could also see that the time since last updated was getting updated. I will provide screen shots tomorrow.

My question is have the current ros drivers been tested with the 3DMGQ7 + RTK unit combination?

nathanmillerparker, 2021-06-08:
Ok.

Yes, the combination was tested, but it would appear that the rtk dongle enable was already stored in the test unit EEPROM.  I am conducting additional testing as we speak.

milievski-flyribbit, 2021-06-08:
Thank you for all your help @nathanmillerparker!

nathanmillerparker, 2021-06-08:
No problem

milievski-flyribbit, 2021-06-08:
Okay, would it be possible to walk me though how everything was set up on your side on both the 3DMGQ7 plus the RTK unit on SensorConnect side, I must be setting something up wrong.

nathanmillerparker, 2021-06-08:
Using SensorConnect is only a debugging step that I suggested if you are having issues with the ROS driver.  In general, you shouldn't really need to do much except plug everything in and set the following launch file parameter:

    
    <param name="rtk_dongle_enable" value="**true**"    type="bool" />

I fixed the bug that you found and committed it, which wasn't discovered because my unit had the RTK dongle settings stored in EEPROM during testing.  I recommend pulling the latest driver.

With that in place, you should get something like the following:

[image]

This of course assumes that the dongle is able to connect to a cell tower, negotiate a connection, and validate with the RTK service.  Our coverage map is here: https://rtk.sensorcloud.com/coverage/  

Let me know if you run into any other problems.  If we need to make this more interactive, I can set up a Teams meeting sometime.

milievski-flyribbit, 2021-06-08:
Setting up a teams meeting would be awesome! This will allow me to confirm everything is correct with our set up in a very fast and effective method. Please let me know what time works best for you as I can make any time work in the following few weeks.

milievski-flyribbit, 2021-06-11:
Thank you for all your help at this point I think its this issue was found to be a problem with our set up rather then the drivers so I will be closing this issue.

closed 2021-06-11 by milievski-flyribbit