LORD-MicroStrain/microstrain_inertial #65: Sensor setup question

Jaetriel, 2021-06-11:
Hi, I've got a question regarding setting up the device to use with the ROS package. I use the 3DM-GX5-25 and was wondering, what kind of setup would I need to do from SensorConnect on Windows to be able to just plug the device into a Linux machine later on and run the ROS node without going through the device setup on startup (so device_setup set to false in the launch file) but still get pretty much the same info in ROS ( orientation, ang velocity and accel )?

ianmooreparker, 2021-06-11:
Hi @Jaetriel , 
You are correct to set device_setup to false in the launch file. You would need to setup data type and rate for applicable messages for the GX5-25. To make that process easier, I have attached a .json settings [file](https://parkercorp-my.sharepoint.com/:u:/g/personal/ian_moore_parker_com/EbfAVTbMnp1Bi3GhNqLTXwcBpfS1ooUtosDUtZ6a4Fks-Q?e=Uoumvr). Simply download, open SensorConnect, plug in your GX5-25, click Save/Load Settings tile > click the green folder > select this file > click Import. Once imported successfully, you can customize your configurations under the Configure tile or change your data rate under the Sampling tile. 
Let me know if this makes sense or if you have other questions!

Jaetriel, 2021-06-14:
@ianmooreparker thanks a lot for the file! I will give it a try the first chance I get and get back to you with any feedback

closed 2022-01-30 by github-actions[bot]