LORD-MicroStrain/microstrain_inertial #68: MSCL_LIB_PATH Error

ArunbAk, 2021-07-06:
Hi all,
I cloned the package to my workspace but when I ran `catkin_make` I am getting this error. 
catkin_make not compiled completely . 

```
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
MSCL_LIB_PATH
    linked by target "ros_mscl" in directory /home//home/arun/catkin_ws/src/ROS-MSCL/ros_mscl

-- Configuring incomplete, errors occurred!
See also "/home/arun/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/arun/catkin_ws/build/CMakeFiles/CMakeError.log".
Makefile:2378: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed
```
How to solve this issue?
Anyone please Help me out.

Thanks in advance.

nathanmillerparker, 2021-07-06:
Were you able to install MSCL per the build instructions? (located here: https://github.com/LORD-MicroStrain/ROS-MSCL)

[image]

ArunbAk, 2021-07-08:
Thanks for your response,

Nope , I cloned the  ROS-MSCL package from here(https://github.com/LORD-MicroStrain/ROS-MSCL) but when I do catkin_make to my workspace where I cloned the package it ended up with Errors. 

`Cmake Error : MSCL_LIB_PATH not included `
(something related to that)

At the last line of the instruction you have attached , How to update my install path in CMakeLists.txt ?
For example,
I have cloned package at **/home/arun/catkin_ws/src**  so how to update this path to the CMakeLists.txt?

nathanmillerparker, 2021-07-08:
Hopefully this helps:

There are 2 components that need to be installed on your system to run the ROS driver - our precompiled MSCL library and the ROS-MSCL package that you cloned.  Did you install the precompiled MSCL library on your system?  That will need to be installed to compile the ROS driver.

robbiefish, 2021-11-19:
Closing this as it appears to be solved. Please reopen this issue if it is still not working

closed 2021-11-19 by robbiefish