LORD-MicroStrain/microstrain_inertial #69: use 3dm-gx5-25 connect ubuntu,no ttyACM device

jingwenyi, 2021-07-19:
I use 3dm-gx5-25 connect ubuntu ok, but I can't find ttyACM* device

jingwenyi, 2021-07-19:
I use computer USB port

nathanmillerparker, 2021-07-21:
The device should enumerate as a standard VCP serial port.  Have you tried a different computer?

jingwenyi, 2021-07-22:
my ubuntu run in VMware, I have tried two computers. my ubuntu version is 16.4.  how to slove this problem?

jingwenyi, 2021-07-22:
I test px4 board, my computer can show the ttyACM0 device.

jingwenyi, 2021-07-22:
I test usb->ttl mode, my computer can show ttyUSB0 device.

nathanmillerparker, 2021-07-22:
Would you consider this issue resolved now?

jingwenyi, 2021-07-23:
no, I do not resolve this issue.  I  only prove my computer's  envirnment is ok.  I will try to use RS232 connect computer and 3dm-gx5-25.Have you never encountered such a problem before？

nathanmillerparker, 2021-07-23:
If you are running a virtual machine inside windows, we have seen this issue with USB... which is currently unresolved.

jingwenyi, 2021-07-27:
I test RS-232  to connect my computer. I change ttyACM0 to ttyUSB0:

roslaunch ros_mscl microstrain.launch

process[imu_diagnostic_aggregator-3]: started with pid [5648]
[ INFO] [1627371411.777577531]: Using MSCL Version: 62.0.0
[ INFO] [1627371411.777712173]: Attempting to open serial port </dev/ttyUSB0> at <115200> 

[ INFO] [1627371414.700256269]: Model Name:    3DM-GX5-25

[ INFO] [1627371414.700421816]: Serial Number: 6253.115711

[ INFO] [1627371414.729334271]: Setting to Idle: Stopping data streams and/or waking from sleep
[ INFO] [1627371414.734326368]: Setting IMU data to stream at 100 hz
[ INFO] [1627371414.796431570]: Setting Declination Source
[ INFO] [1627371414.809637191]: Setting Filter data to stream at 10 hz
[ INFO] [1627371414.829525226]: Note: The device does not support the vehicle dynamics mode command.
[ INFO] [1627371414.829560992]: Note: The device does not support the PPS source command.
[ INFO] [1627371414.829599963]: Setting heading source to 0X01
[ INFO] [1627371414.837701695]: Setting autoinitialization to 1
[ INFO] [1627371414.849772483]: Note: The device does not support the filte adaptive settings command.
[ INFO] [1627371414.849819980]: Note: The device does not support the filter aiding command.
[ INFO] [1627371414.849843783]: Note: The device does not support the relative position command.
[ INFO] [1627371414.849864731]: Note: The device does not support the filter speed lever arm command.
[ INFO] [1627371414.849884785]: Note: The device does not support the wheeled vehicle constraint command.
[ INFO] [1627371414.849919366]: Note: The device does not support the vertical gyro constraint command.
[ INFO] [1627371414.849945299]: Note: The device does not support the GNSS antenna calibration command.
[ INFO] [1627371414.849964376]: Note: The device does not support the next-gen filter initialization command.
[ INFO] [1627371414.858532262]: Setting sensor2vehicle frame rotation with euler angles [-0.000000, -0.000000, -0.000000]
[ INFO] [1627371414.867672254]: Resetting the filter after the configuration is complete.
[ INFO] [1627371414.900984757]: Publishing IMU data.
[ INFO] [1627371414.903051174]: Publishing Magnetometer data.
[ INFO] [1627371414.907565751]: Publishing Filter data.
[ INFO] [1627371415.054158878]: Setting spin rate to <200>
[ INFO] [1627371415.088762500]: Starting Data Parsing

rostopic list
/diagnostics
/diagnostics_agg
/diagnostics_toplevel_state
/gx5/device/status
/gx5/imu/data
/gx5/mag
/gx5/nav/filtered_imu/data
/gx5/nav/heading
/gx5/nav/heading_state
/gx5/nav/odom
/gx5/nav/status
/moving_ang
/rosout
/rosout_agg

roslaunch ros_mscl_cpp_example listener.launch

UMMARY
========

PARAMETERS
 * /listener_cpp/listener_cpp/device: gx5
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /listener_cpp/
    listener_cpp (ros_mscl_cpp_example/listener)

ROS_MASTER_URI=http://localhost:11311

WARNING: Package name "3dmodel" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
**ERROR: cannot launch node of type [ros_mscl_cpp_example/listener]: can't locate node [listener] in package [ros_mscl_cpp_example]**
No processes to monitor
shutting down processing monitor...
... shutting down processing monitor complete

why did this operation (roslaunch ros_mscl_cpp_example listener.launch) fail?

robbiefish, 2021-07-29:
@jingwenyi The launch file problem should be fixed in the latest commit to master. Can you pull master and try again?

robbiefish, 2021-11-19:
Closing this as it appears to be solved. Please reopen this issue if it is still not working

closed 2021-11-19 by robbiefish