LORD-MicroStrain/microstrain_inertial #72: RTK Statues

Aussie25, 2021-09-28:
I am using 3DMGQ7 with the ROS driver.  Is it possible to check the statues of the RTK in the ROS driver if the sensor is RTK float or fix. The sensor is receiving the RTCM messages through ntrip client. 

Thank you.

nathanmillerparker, 2021-09-28:
Yes.... you are looking for the  "differential_corrections" and "integer_fix" flags of the GNSSAidingStatus message to be set.  These are published for both GNSS modules in the "gnss#/aiding_status" topic, where "#" is 1 or 2.  An important note, only one of the receivers needs to be reporting an integer fix at any given time to know you have an RTK fix.

Hope that helps!

Aussie25, 2021-09-29:
Thank you very much, that helps.

closed 2021-09-29 by Aussie25