LORD-MicroStrain/microstrain_inertial #73: Change Statues Messages Data Rates

Aussie25, 2021-10-15:
Hi, when I try run the gq7 at 500hz for both imu and filter data rate. The messages received are not consistent. The baudrate is set to 921600. 
[image]
 
Is It possible to reduce the data of the topics like gq7/gnss1/aiding_statues, gq7/gnss2/aiding_statues , gq7/mag , gq7/nav/dual_antenna statues and gq7/nav/statues.

Aussie25, 2021-10-19:
Thank for the response. 
To make the the filter rate dependent to the channel name is it done like this for example "supportedChannels.push_back(mscl::MipChannel(CH_FIELD_ESTFILTER_FILTER_STATUS, 10));"

Aussie25, 2021-10-20:
Thanks you very much, I will try it out.

Aussie25, 2021-11-01:
Hey, 
Tried out the above code snippet but the topics are not publishing at 10hz but instead of 1000hz. Whereby the filter data rate is set to 500hz.

robbiefish, 2021-11-01:
Hi @Aussie25 

I am not sure I understand what you are saying. Are you saying that you are seeing filter data at 1000hz despite setting the `filter_data_rate` argument in your launch file to 500?

How are you measuring the data rates? I measured the rates with the command `rostopic hz /gq7/nav/status /gq7/nav/dual_antenna_status /gq7/nav/filtered_imu/data /gq7/nav/heading /gq7/nav/heading_state /gq7/nav/odom`

I have implemented the changes we spoke about in the following [fork](https://github.com/robbiefish/microstrain_inertial/tree/example/different_rates), does this looks similar to your setup?

Aussie25, 2021-11-02:
Hi Robbie,

Yup, my set up is identical as your fork version. I even tried running your fork set up, leads to the same issues. 

Despite setting the `filter_data_rate` as 500 in the launch file, the filter data rate is being publish at 1000hz. I am measuring the topic frequency with `rostopic hz` too. 

My terminal output after launch 

```
PARAMETERS
 * /gx5/ros_mscl_node/baudrate: 921600
 * /gx5/ros_mscl_node/device_setup: True
 * /gx5/ros_mscl_node/filter_adaptive_level: 2
 * /gx5/ros_mscl_node/filter_adaptive_time_limit_ms: 15000
 * /gx5/ros_mscl_node/filter_angular_zupt: True
 * /gx5/ros_mscl_node/filter_angular_zupt_topic: /moving_ang
 * /gx5/ros_mscl_node/filter_auto_heading_alignment_selector: 1
 * /gx5/ros_mscl_node/filter_auto_init: True
 * /gx5/ros_mscl_node/filter_data_rate: 500
 * /gx5/ros_mscl_node/filter_declination: 0.23
 * /gx5/ros_mscl_node/filter_declination_source: 2
 * /gx5/ros_mscl_node/filter_dynamics_mode: 1
 * /gx5/ros_mscl_node/filter_enable_altimeter_aiding: False
 * /gx5/ros_mscl_node/filter_enable_external_gps_time_update: False
 * /gx5/ros_mscl_node/filter_enable_external_heading_aiding: False
 * /gx5/ros_mscl_node/filter_enable_gnss_antenna_cal: False
 * /gx5/ros_mscl_node/filter_enable_gnss_heading_aiding: True
 * /gx5/ros_mscl_node/filter_enable_gnss_pos_vel_aiding: True
 * /gx5/ros_mscl_node/filter_enable_magnetometer_aiding: False
 * /gx5/ros_mscl_node/filter_enable_odometer_aiding: False
 * /gx5/ros_mscl_node/filter_enable_vertical_gyro_constraint: False
 * /gx5/ros_mscl_node/filter_enable_wheeled_vehicle_constraint: False
 * /gx5/ros_mscl_node/filter_external_gps_time_topic: /external_gps_time
 * /gx5/ros_mscl_node/filter_frame_id: sensor_wgs84
 * /gx5/ros_mscl_node/filter_gnss_antenna_cal_max_offset: 0.1
 * /gx5/ros_mscl_node/filter_heading_source: 1
 * /gx5/ros_mscl_node/filter_init_attitude: [0.0, 0.0, 0.0]
 * /gx5/ros_mscl_node/filter_init_condition_src: 0
 * /gx5/ros_mscl_node/filter_init_position: [0.0, 0.0, 0.0]
 * /gx5/ros_mscl_node/filter_init_reference_frame: 2
 * /gx5/ros_mscl_node/filter_init_velocity: [0.0, 0.0, 0.0]
 * /gx5/ros_mscl_node/filter_pps_source: 1
 * /gx5/ros_mscl_node/filter_relative_position_frame: 2
 * /gx5/ros_mscl_node/filter_relative_position_ref: [0, 0, 0.01]
 * /gx5/ros_mscl_node/filter_reset_after_config: True
 * /gx5/ros_mscl_node/filter_sensor2vehicle_frame_selector: 1
 * /gx5/ros_mscl_node/filter_sensor2vehicle_frame_transformation_euler: [0.0, 0.0, 0.0]
 * /gx5/ros_mscl_node/filter_sensor2vehicle_frame_transformation_matrix: [1.0, 0.0, 0.0, 0...
 * /gx5/ros_mscl_node/filter_sensor2vehicle_frame_transformation_quaternion: [0.0, 0.0, 0.0, 1.0]
 * /gx5/ros_mscl_node/filter_speed_lever_arm: [0.0, 0.0, 0.0]
 * /gx5/ros_mscl_node/filter_velocity_zupt: True
 * /gx5/ros_mscl_node/filter_velocity_zupt_topic: /moving_vel
 * /gx5/ros_mscl_node/gnss1_antenna_offset: [0.0, -0.7, -1.0]
 * /gx5/ros_mscl_node/gnss1_data_rate: 2
 * /gx5/ros_mscl_node/gnss1_frame_id: gnss1_antenna_wgs84
 * /gx5/ros_mscl_node/gnss2_antenna_offset: [0.0, 0.7, -1.0]
 * /gx5/ros_mscl_node/gnss2_data_rate: 2
 * /gx5/ros_mscl_node/gnss2_frame_id: gnss2_antenna_wgs84
 * /gx5/ros_mscl_node/gpio1_behavior: 0
 * /gx5/ros_mscl_node/gpio1_feature: 0
 * /gx5/ros_mscl_node/gpio1_pin_mode: 0
 * /gx5/ros_mscl_node/gpio2_behavior: 0
 * /gx5/ros_mscl_node/gpio2_feature: 0
 * /gx5/ros_mscl_node/gpio2_pin_mode: 0
 * /gx5/ros_mscl_node/gpio3_behavior: 0
 * /gx5/ros_mscl_node/gpio3_feature: 0
 * /gx5/ros_mscl_node/gpio3_pin_mode: 0
 * /gx5/ros_mscl_node/gpio4_behavior: 0
 * /gx5/ros_mscl_node/gpio4_feature: 0
 * /gx5/ros_mscl_node/gpio4_pin_mode: 0
[... 93 lines cut ...]
rostopic hz /gx5/nav/filtered_imu/data 
subscribed to [/gx5/nav/filtered_imu/data]
average rate: 1007.002
	min: 0.000s max: 0.011s std dev: 0.00192s window: 944

```
```

subscribed to [/gx5/nav/status]
subscribed to [/gx5/nav/dual_antenna_status]
subscribed to [/gx5/nav/filtered_imu/data]
subscribed to [/gx5/nav/heading]
subscribed to [/gx5/nav/heading_state]
subscribed to [/gx5/nav/odom]
           topic                  rate     min_delta   max_delta   std_dev    window
====================================================================================
/gx5/nav/status                1.002e+03   6.914e-06   0.02996     0.001612   1041  
/gx5/nav/dual_antenna_status   499.0       8.106e-06   0.03336     0.002508   1041  
/gx5/nav/filtered_imu/data     996.1       5.007e-06   0.02928     0.002466   522   
/gx5/nav/heading               1.026e+03   5.96e-06    0.01073     0.001365   522   
/gx5/nav/heading_state         998.5       7.868e-06   0.01165     0.001366   1034  
/gx5/nav/odom                  999.8       5.007e-06   0.02884     0.003201   1034  
```

Thank you.

robbiefish, 2021-11-02:
Thank you very much for the information. I see the same behavior that you posted on the `master` branch. Were your tests done on master, or on the "example/different_rates" branch I linked to?

As a sanity check could you check that no other `ros_mscl` nodes are running by checking the output of `rosnode list` and then try running the following commands?

In one terminal:
```
$ git clone -b "example/different_rates" https://github.com/robbiefish/microstrain_inertial.git ~/microstrain_test_ws/src/microstrain_inertial
$ cd ~/microstrain_test_ws
$ source /opt/ros/melodic/setup.bash
$ catkin_make
$ source devel/setup.bash
$ roslaunch ros_mscl microstrain.launch
```

In a second terminal
```
$ source ~/microstrain_test_ws/devel/setup.bash
$ rostopic hz /gq7/nav/status /gq7/nav/dual_antenna_status /gq7/nav/filtered_imu/data /gq7/nav/heading /gq7/nav/heading_state /gq7/nav/odom
```

robbiefish, 2021-12-09:
Closing this issue due to inactivity. Please reopen if more help is required

closed 2021-12-09 by robbiefish