LORD-MicroStrain/microstrain_inertial #79: libros_mscl.so: undefined symbol in ROS Melodic

dwhite18, 2021-11-19:
I'm using the 3DMGQ7-GNSS sensor on a clearpath Husky UGV with a Jetson Xavier running Ubuntu 18.04 LTS. I am trying to run 'roslaunch ros_mscl microstrain.launch' and i keep getting this error.

[ INFO] [1637347321.374461281]: Publishing IMU data.
[ INFO] [1637347321.378945511]: Publishing Magnetometer data.
[ INFO] [1637347321.381899625]: Publishing GNSS1 data.
[ INFO] [1637347321.389770445]: Publishing GNSS2 data.
[ INFO] [1637347321.399111634]: Publishing Filter data.
[ INFO] [1637347321.463575671]: Setting spin rate to <200>
[ INFO] [1637347321.489508540]: Starting Data Parsing
/opt/ros/melodic/lib/ros_mscl/ros_mscl_node: symbol lookup error: /opt/ros/melodic/lib/libros_mscl.so: undefined symbol: _ZNK4mscl20RTKDeviceStatusFlags15controllerStateEv
[gq7/ros_mscl_node-1] process has died [pid 17450, exit code 127, cmd /opt/ros/melodic/lib/ros_mscl/ros_mscl_node __name:=ros_mscl_node __log:=/home/administrator/.ros/log/5a798b74-4963-11ec-8ece-48b02d3a858d/gq7-ros_mscl_node-1.log].
log file: /home/administrator/.ros/log/5a798b74-4963-11ec-8ece-48b02d3a858d/gq7-ros_mscl_node-1*.log

robbiefish, 2021-11-19:
Hi @dwhite18,

I am unfortunately unable to reproduce this error in a similar environment. However, I have seen similar errors when linking against a debug version of MSCL. Are you installing your own version of MSCL?

Also, the branch you are using has been deprecated in favor of the [ros](https://github.com/LORD-MicroStrain/microstrain_inertial/tree/ros) branch. You can install this branch from the buildfarm by running the following commands:
```
sudo apt-get update
sudo apt-get install ros-melodic-microstrain-inertial-driver
```

Then the equivalent of your `roslaunch` command above would be:
```
roslaunch microstrain_inertial_driver microstrain.launch
```

robbiefish, 2021-12-09:
Closing this issue due to inactivity. Please reopen if more help is required

closed 2021-12-09 by robbiefish