LORD-MicroStrain/microstrain_inertial #80: launch error

realxuyang, 2021-11-22:
[image]

realxuyang, 2021-11-22:
we move microstrain_inertial_driver, microstrain_inertial_msgs , and microstrain_inertial_examples in ~/catkin_ws/src, then roslaunch microstrain_inertial_driver microstrain.launch，but always this error.

robbiefish, 2021-11-22:
Hi @realxuyang 

That error suggests that the `catkin_make` command did not succeed. Could you check if `catkin_make` succeeded without errors? If it failed, could you please post the errors?

There are several reasons for why this might have happened, but my first guess would be that this is a problem with missing submodules. When you cloned the repo did you add the `--recursive` flag? The whole command would look this this:
```
git clone --recursive https://github.com/LORD-MicroStrain/microstrain_inertial.git
```

More documentation on submodules can be found [here](https://github.com/LORD-MicroStrain/microstrain_inertial#submodules)

realxuyang, 2021-11-22:
> Hi @realxuyang
>
> That error suggests that the `catkin_make` command did not succeed. Could you check if `catkin_make` succeeded without errors? If it failed, could you please post the errors?
>
> There [...]

Thank you very much, we'll try it as soon as possible!

realxuyang, 2021-11-23:
> git clone --recursive https://github.com/LORD-MicroStrain/microstrain_inertial.git
sorry, we meet this bug when git cloning：
fatal: clone of 'https://github.com/LORD-MicroStrain/microstrain_inertial_driver_common.git' into submodule path 'microstrain_inertial_driver/microstrain_inertial_driver_common' failed

realxuyang, 2021-11-23:
We catkin_make succeed after downloading zip of the source code. but when roslaunch, we met a new bu
[image]
g

realxuyang, 2021-11-23:
> 我们catkin_make 下载了源代码的zip 后就成功了。但是在roslaunch的时候，我们遇到了一个新的bug，我们的设备是3DM-GX3-25
> 
> [image]

we sove this problem, but when we launch, rostopic list have the 3 topics, but we cant rostopic echo them, and when we echo them to txt, we got 0 bytes

robbiefish, 2021-11-23:
I am glad to hear that you got the node up and running. I am unfortunately unable to reproduce the error with the same device. However, I have seen similar problems with rostopic when `ROS_MASTER_URI` and either `ROS_IP` or `ROS_HOSTNAME` are set to the wrong values. For more information, [this article](https://answers.ros.org/question/90536/ros-remote-master-can-see-topics-but-no-data/) might be helpful.

In my case, I am running rostopic and the node on the same computer, so I only have `ROS_MASTER_URI` set to `http://localhost:11311`

Could you please post the output of the following command?
```bash
env | grep -E 'ROS_MASTER_URI|ROS_IP|ROS_HOSTNAME'
```

realxuyang, 2021-11-23:
> I am glad to hear that you got the node up and running. I am unfortunately unable to reproduce the error with the same device. However, I have seen similar problems with rostopic when `ROS_MASTER_URI` and either `ROS_IP` or `ROS_HOSTNAME` are set to the wrong values. For more information, [this article](https://answers.ros.org/question/90536/ros-remote-master-can-see-topics-but-no-data/) might be helpful.
> 
> In my case, I am running rostopic and the node on the same computer, so I only have `ROS_MASTER_URI` set to `http://localhost:11311`
> 
> Could you please post the output of the following command?
> 
> ```shell
> env | grep -E 'ROS_MASTER_URI|ROS_IP|ROS_HOSTNAME'
> ```

[image]

realxuyang, 2021-11-24:
> > I am glad to hear that you got the node up and running. I am unfortunately unable to reproduce the error with the same device. However, I have seen similar problems with rostopic when `ROS_MASTER_URI` and either `ROS_IP` or `ROS_HOSTNAME` are set to the wrong values. For more information, [this article](https://answers.ros.org/question/90536/ros-remote-master-can-see-topics-but-no-data/) might be helpful.
> > In my case, I am running rostopic and the node on the same computer, so I only have `ROS_MASTER_URI` set to `http://localhost:11311`
> > Could you please post the output of the following command?
> > ```shell
> > env | grep -E 'ROS_MASTER_URI|ROS_IP|ROS_HOSTNAME'
> > ```
> 
> [image] I ran rostopic and the node on the same computer too. i can echo the other topics(/rosout ) but cant echo /gx3/imu/data, /gx3/mag and /gx3/device/status is the bug from src.cpp or microtrain.launch? After we were launch, the lights on the sensor didn't flash

we find that this bug may have been caused by the imu not entering data acquisition mod, We start the imu under windows, then unplug the cable and plug it into the ubuntu, then we can do the launch and echo

robbiefish, 2021-11-29:
hi @realxuyang,

That is strange behavior, but it sounds like you are getting data now. Would you consider this issue resolved?

robbiefish, 2021-11-29:
Okay, I did not think to check before, but can you verify that your user is part of the `dialout` group? I was able to see similar behavior to what you are describing if my user is not part of that group. An easy way to check is to run the `groups` command and check that `dialout` is listed somewhere in the output

If you are not part of the dialout group, you can add yourself to the group by running the following command: (if for some reason `$USER` is not defined, replace that with your username)
```bash
sudo usermod -aG dialout $USER
```
After you run that command you will need to restart the computer for those changes to take affect. You can verify that you are now in the `dialout` group by running `groups` after you restart

robbiefish, 2021-12-09:
Closing this issue due to inactivity. Please reopen if more help is required

robbiefish, 2022-02-15:
Hi @yunxdai,

Sorry to hear that you are having this issue. I am assuming that you have already tried the steps suggested in this issue, specifically, ensuring that your user is added to the `dialout` group?

@realxuyang, did you ever end up solving your issue? You might have some more insight than me on what solved this if so.

yunxdai, 2022-02-16:
> Hi @yunxdai,
>
> Sorry to hear that you are having this issue. I am assuming that you have already tried the steps suggested in this issue, specifically, ensuring that your user is added to the [...]

Yes, I have checked that the user is in ```dialout``` group.

realxuyang, 2022-02-16:
> 

> Yes, I have checked that the user is in `dialout` group.

hi, I haven't fully solved this problem, can we use wechat to discuss more conveniently?

yunxdai, 2022-02-16:
> 

Sure! my wechat id is "yunxdai1999".

closed 2022-03-20 by github-actions[bot]