LORD-MicroStrain/microstrain_inertial #92: Does  3DMGX5-AHRS has different euler frame with 3DMGX5-25?

BizzareNathan, 2021-12-21:
I got a 3DMGX5-AHRS and i want to use it as the imu for LIO-SAM, but after several tests, i found that the settings of the euler coordinate system is x(roll),y(pitch),z(yaw),which is not same system as shown in the picture.

So i'd like to know if these two imu really have two different euler system ?

Thanks in advance.
[image]

nathanmillerparker, 2021-12-21:
If you set "use_enu_frame" to true, the axis for inertial data frame is reported as x out the connector, z up, y completing the right hand rule.  That doesn't quite match what you have so....

Did you set any of these launch parameters?:

<param name="filter_sensor2vehicle_frame_selector" value="1" type="int" />

    <rosparam param="filter_sensor2vehicle_frame_transformation_euler">      [0.0, 0.0, 0.0]</rosparam>
    <rosparam param="filter_sensor2vehicle_frame_transformation_matrix">     [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]</rosparam>
    <rosparam param="filter_sensor2vehicle_frame_transformation_quaternion"> [0.0, 0.0, 0.0, 1.0]</rosparam>

Also, please provide your serial number so we can double check your calibration report.

Thank you,

Nathan

BizzareNathan, 2021-12-22:
Hi Nathan,

My serial number is 6253.123790.
After I found that, I use your software MIP monitor to observe the change of Euler angles.
Since there are too many pictures to show, I will tell you my situation in form of roll, pitch, yaw，x out the connector,z down, y completing the right hand rule as the frame(marked on the AHRS).

Original     -0.0538     0.0931     -2.4816
rotate -x    -1.0757     -0.0681   -2.4067
rotate +x    1.0013     -0.0006    -2.5197
rotate -y     0.00186   -0.4985    -2.5035
rotate +y   -0.0135      0.6230    -2.4611
rotate -z     0.0536      0.0375     -3.1247
rotate +z     0.0375     0.08624   -2.0679

[image]
[image]

To make the yaw angle counterclockwise positive (as in the figure), I tried flipping AHRS up and down, but still got the yaw angle clockwise positive.

BizzareNathan, 2021-12-22:
After your reminder, I learned the concept of the NED coordinate system, and if I understand correctly, the Euler angles output by AHRS are based on the NED coordinate system, and the acceleration is based on the coordinate system marked on the device. For the definition of yaw, in particular, I think it is defined here as north by east as positive, so I get a negative result.

So I set the coordinate system to "use_enu_frame" to true in the launch file and got the result you mentioned, after reorienting the x-axis and y-axis, I got the result I wanted.
```
<rosparam param="filter_sensor2vehicle_frame_transformation_euler">      [3.14159, 0.0, 0.0]</rosparam>
```

During the calibration process, I found that the roll angle was around -3.14 in the initial state, so I added an offset of [3.14159 0 0] to the launch file for the Euler angle, and finally, I got the initial values of roll and pitch around 0.

closed 2021-12-22 by BizzareNathan

zyx-99, 2022-04-21:
> After your reminder, I learned the concept of the NED coordinate system, and if I understand correctly, the Euler angles output by AHRS are based on the NED coordinate system, and the acceleration [...]

请问答主LIOSAM调通了嘛，我也是用此款IMU想做LIOSAM，但RPY的外参数矩阵始终调不出，可以交流一下嘛？

redheli, 2024-09-04:
> 请问答主LIOSAM调通了嘛，我也是用此款IMU想做LIOSAM，但RPY的外参数矩阵始终调不出，可以交流一下嘛？

I tried, can not
shall use original `microstrain_mips`, can download from https://github.com/TixiaoShan/LIO-SAM/issues/128#issuecomment-706556404
it contains midification of `imu_correct` topic