LORD-MicroStrain/microstrain_inertial #95: LIO-SAM INTEGRATION WITH GQ7

dsb6063, 2021-12-27:
Does 3DMGX5-AHRS has different euler frame with 3DMGX5-25? #92

I have GX-25 also. I use the author's custom microsense driver code for the rotation matrix. With the GQ7, I would like to use the parameters config file. The author's provides a parameter file in the LIO-SAM code below.

extrinsicTrans: [-2.155, 0.0, -0.83]
 extrinsicRot: [1, 0, 0,
                0, 1, 0,
                0, 0, 1]
 extrinsicRPY: [1, 0, 0,
                0, 1, 0,
                0, 0, 1]

Iam having problem aligning the acceleration matrix. I seem to be able to rotation the sensor to lidar frame, but after reader issue #92, it seems that I can use the Enu to true then rotate 90 degrees around the z axis to obtain the Ouster LiDAR frame. Is the correct and is the best route to obtain the sensor to  Lidar frame
The main point here is to use the micosense frame work instead of LIO-SAM config file.

dsb6063, 2021-12-28:
[image]

Below is LIO-SAM Author's custom microsense driver /imu_correct topic. The above picture shows the orientation from sensor to vehicle (lidar).  The linear and angular accelerations x and z are negated. The custom driver works and trying to repeat the process in the params.yaml file in the current driver version.

[image]

dsb6063, 2021-12-28:
Default settings for /config/params file:

extrinsicRot: [-1, 0, 0,
                  0, 1, 0,
                  0, 0, -1]
  extrinsicRPY: [0,  1, 0,
                 -1, 0, 0,
                  0, 0, 1]

I initially found the author's dataset to work, but when I needed to configure my own dataset, it did not work as expected. 

The above rotations use the microstrain gx-25 orientation of south ,left, down (aux plug at bottom) or NED depending on (aux plug at top). Using the south, left, down system, the following frame:

 extrinsicRot row 1 of (-1,0,0) rotates x axis to the north (forward)
 extrinsicRot row 2 of (0,1,0) keeps same y axis
 extrinsicRot row 2 of (0,0,-1) flips z axis to up

IMU alignment. LIO-SAM transforms IMU raw data from the IMU frame to the Lidar frame, which follows the ROS REP-105 convention (x - forward, y - left, z - upward). To make the system function properly, the correct extrinsic transformation needs to be provided in "params.yaml" file. The reason why there are two extrinsics is that my IMU (Microstrain 3DM-GX5-25) acceleration and attitude have different coordinates. Depend on your IMU manufacturer, the two extrinsics for your IMU may or may not be the same. 

 extrinsicRPY was configured using the 3D Maxtrix Calculator starting with the  extrinsicROT from above AND rotate (-90 deg) or -1.5708 to end up with 

  extrinsicRPY: [0,  1, 0,
                        -1, 0, 0,
                        0, 0, 1]

this aligns the RPY with the new convention of convention (x - forward, y - left, z - upward).

At this point, one should be able to keep the default settings with the microstrain driver with NED and LIO-SAM params will correct the the imu settings to match Ouster's Lidar frame. 

Side note:

I attached the author's snippet of custom implementation of the microsense driver to produce the "imu_correct" in the dataset. Therefore, if you use the "imu_correct", set the extrinsicRPY and  extrinsicRot  matrices to the identity matrix.

robbiefish, 2022-01-03:
Hi @dsb6063 is this still an issue?

dsb6063, 2022-01-04:
No. I had to review the settings a few times and compare with the use case.

closed 2022-01-04 by dsb6063