LORD-MicroStrain/microstrain_inertial #96: more flexible param setting on launch

adamsj-ros, 2021-12-28:
Currently launch args for `microstrain_launch.py` are used to override params. It would be nice to use a single launch arg that is a `param.yaml` path with a default file included in the package.  This way only a single arg needs to be exposed but allows for a complete resetting of all params available.

robbiefish, 2022-01-03:
Hi @adamsj-ros, this has been implemented close to what you are requesting in [feature/ros_params_file](https://github.com/LORD-MicroStrain/microstrain_inertial/tree/feature/ros_params_file) for ROS, and in [feature/ros2_params_file](https://github.com/LORD-MicroStrain/microstrain_inertial/tree/feature/ros2_params_file) for ROS2.

Couple caveats:
* The [default yaml file](https://github.com/LORD-MicroStrain/microstrain_inertial_driver_common/blob/feature/params_file/config/params.yml) is in a format that will work with ROS so we only need to maintain one copy between versions, so copying it to use as a starting point for ROS2 will not work. Instead, you should start with [this empty yml file](https://github.com/LORD-MicroStrain/microstrain_inertial/blob/feature/ros2_params_file/microstrain_inertial_driver/config/empty.yml)
* The default yaml file is always loaded, but it is loaded first, so the file specified by `params_file` will override any settings specified in the default yaml file

Does this accomplish what you are looking for?

adamsj-ros, 2022-01-04:
That looks to be the right track for what I'm suggesting. I don't have direct access to a device to completely test at the moment but it seems good. You probably want to complete the launch arguments with a description for each. When I look at the launch file they are obvious but it's still nice to have the descriptions for the `-s` option.

robbiefish, 2022-01-04:
Good feedback, descriptions have been added, and PRs have been opened. Please feel free to leave any additional comments on the corresponding PRs