LORD-MicroStrain/microstrain_inertial #110: Option to stamp headers with ros::time::now()

sei-jmattson, 2022-01-20:
One can optionally stamp ROS message headers with collected-time or device-time.  Can you add an option for ros-time? Or consider a pull-request?

I feed a gx5-45 imu topic at 400Hz to a google cartographer node.  Cartographer's imu_tracker has a time comparison that is super sensitive, and faults when the times don't match.

Using `to_ros_time(<collected-time | device-time)` often causes a slight offset; more frequently at higher data rates.  https://github.com/LORD-MicroStrain/microstrain_inertial_driver_common/blob/267552013b4a38e65322bceb8a5afdab54b27c67/src/microstrain_parser.cpp#L136

I modify the `microstrain_parser.cpp` to make all `...header.stamp = ros_time_now(node_);`, the discrepancies disappear and cartographer is happy. 

Here is an example of the time offset that cartographer sees...often very small.
```
[FATAL] [1642703280.516943908]: F0120 13:28:00.000000 17821 imu_tracker.cc:40] Check failed: time_ <= time (637783000805140749 vs. 637783000805140746)
```

robbiefish, 2022-01-20:
@nathanmillerparker This would be fairly easy to implement, and could also be useful if users want to run the driver with other data in simulation such as playing back a bag. Do you have any objections?

nathanmillerparker, 2022-01-21:
@robbiefish Sounds good to me... should be done for all data streams though.

robbiefish, 2022-01-21:
@sei-jmattson This should be implemented in [`feature/ros_use_ros_time`](https://github.com/LORD-MicroStrain/microstrain_inertial/tree/feature/ros_use_ros_time) for ROS and [`feature/ros2_use_ros_time`](https://github.com/LORD-MicroStrain/microstrain_inertial/tree/feature/ros2_use_ros_time) for ROS2. Can you test the correct branch for the ROS version you are using with the parameter `use_ros_time` set to true and let us know if that works for you?

closed 2022-01-25 by robbiefish