LORD-MicroStrain/microstrain_inertial #120: Filter data messages are not being published on ROS 1.

dimensional-difficulties, 2022-01-31:
I am trying to obtain acceleration and orientation estimates provided by my Microstrain 3DM-GX5-25's adaptive Kalman filter.  When running microstrain_inertial_driver to collect data, no /nav/filter messages are being published, even though the driver outputs the following INFO messages at startup: 

[ INFO] [1643658687.785366709]: checking if we should publish filter data 1 1
[ INFO] [1643658687.785459385]: Publishing Filter data.

The only messages being published are

/device/status
/imu
/imu/data
/mag
/moving_ang
/nav/dual_antenna_status
/nav/heading
/nav/heading_state
/nav/odom
/nav/status
/rosout
/rosout_agg
/tf

I have publish_filter set to True and the filter_data_rate set to 500 (same as the IMU polling rate) in my yml file.  

Strangely, the ROS2 drivers do output the filtered data I need, but I need this to work in the ROS1 ecosystem.

robbiefish, 2022-01-31:
Hi @dimensional-difficulties, I do not believe that we publish any data on the `nav/filter` topic. Do you mean `nav/filtered_imu/data` perhaps?

It is very strange that the ROS2 version is outputting the data while ROS is not. Could you please provide me with a couple pieces of information, so I can attempt to reproduce?
1. The params file you used to launch
2. The command you used to launch the ROS node (and the command you used to launch the ROS2 node if you still have it)
3. The version of the node you are using

dimensional-difficulties, 2022-01-31:
Thank you for your quick reply.  Yes, I did mean the nav/filtered_imu/data messages and have edited my original question to reflect this.  

I have pasted the params file here: https://pastebin.com/erjXJLWZ

The command I used to launch the ROS node is roslaunch microstrain_inertial_driver microstrain.launch params_file:=~/ros_1_DAQ/imu_params.yml

The command I used to launch the ROS2 node is ros2 launch microstrain_inertial_driver microstrain_launch.py poll_rate_hz:=500 imu_data_rate:=500 filter_data_rate:=500

The version of microstrain_inertial_driver I have is 2.4.1 (2022-01-06).

robbiefish, 2022-01-31:
Thanks for the information.

I have set up my own environment using version 2.4.1 of the driver. I launched the node using your launch file using a similar command to what you provided, but was unable to reproduce this bug as I am still seeing data on `/nav/filtered_imu/data`

A few more questions:
1. Is the topic `/nav/filtered_imu/data` showing up in `rostopic list` and then not publishing any messages, or is it not showing up at all?
2. Have you modified the code in any way? In theory this should not be possible without code modifications, as the publisher will always be setup and published on (even if the data is blank) if `publish_filter` is set to true. Specifically the lines in question would be [`microstrain_publishers.cpp:106`](https://github.com/LORD-MicroStrain/microstrain_inertial_driver_common/blob/267552013b4a38e65322bceb8a5afdab54b27c67/src/microstrain_publishers.cpp#L106) and [`microstrain_parser.cpp:950`](https://github.com/LORD-MicroStrain/microstrain_inertial_driver_common/blob/267552013b4a38e65322bceb8a5afdab54b27c67/src/microstrain_parser.cpp#L950)
3. What firmware version is your device running? You can check by running `rosservice call /device_report`
4. This is probably not related, but in case this is affecting anything, what ROS version are you using and what architecture are you running on? I ran my tests on ROS noetic on an x86_64 machine.

dimensional-difficulties, 2022-01-31:
The topic `/nav/filtered_imu/data` is not showing up in rostopic list.

I have not modified the code in any way, although I did delete the whole "microstrain_inertial_rqt" folder because catkin_make seems unable to install it.  Just to be sure, I just cloned, installed, and ran the driver again and saw the same result.  

Line 106 of `microstrain_publishers.cpp` reads `  filtered_imu_pub_ = create_publisher<ImuMsg>(node_, "nav/filtered_imu/data", 100);`, and line 950 of `microstrain_parser.cpp` reads `publishers_->filtered_imu_pub_->publish(publishers_->filtered_imu_msg_);`.  I am seeing the "Publishing Filter data." message upon starting the node, so the if-statement in `microstrain_publishers.cpp` containing the `filtered_imu_pub_ = create_publisher<ImuMsg>(node_, "nav/filtered_imu/data", 100);` is evaluating to "true".  I could add some info messages around line 950 of `microstrain_parser.cpp` to make sure the relevant code is firing if you'd like, but I have not done this yet.  

Are you testing this with a 3DM-GX5-25 (AHRS)?  Perhaps this problem is related to this specific model.  Another person reported this issue on answers.ros.org several years ago, and they said they were using a 3DM-GX5-25: https://answers.ros.org/question/290590/microstrain-3dm-gx5-25-not-streaming-filtered-data/

robbiefish, 2022-01-31:
Okay, that makes sense. Thanks for the additional information. I am also testing with a 3DM-GX5-25. Here is the output of my call to the `/device_report` service:
```
model_name: "3DM-GX5-25"
model_number: "6253-4220"
serial_number: "6253.103495"
options: "8g,300dps"
firmware_version: "1.1.76"
success: True
```

I think that adding some debug messages would be the next logical step. I have added some debug logs that I think would be helpful in the [`debug/filtered_imu`](https://github.com/LORD-MicroStrain/microstrain_inertial_driver_common/tree/debug/filtered_imu) branch of the `microstrain_inertial_driver_common` repo. Can you change the `microstrain_inertial/microstrain_inertial_driver/microstrain_inertial_driver_common` submodule to that branch, rebuild and let me know what the logs look like? I added some logs around the publish statement which at your requested rate will end up being very verbose, so you may need to redirect the output to a file so you can see everything.

I am specifically interested in seeing if you see the following logs once at configure time:
```
[ INFO] [1643667621.862321999]: !!! Publishing to topic nav/filtered_imu/data
[ INFO] [1643667621.864175224]: !!! Topic on resulting publisher /nav/filtered_imu/data
```

And if you are seeing the following logs repeat at runtime:
```
[ INFO] [1643667621.901885352]: ??? Publishing filtered IMU
[ INFO] [1643667621.901930031]: ??? Published filtered IMU
```

dimensional-difficulties, 2022-01-31:
I forgot to mention that the /device_report service didn't work, but I got the firmware from the Windows-based MSCL GUI.  But with this debug branch, the rossservice call /device_report command ouputs this: 

model_name: "3DMGX5-AHRS"
model_number: "6253-4220"
serial_number: "6253.127153"
options: "8g,300dps"
firmware_version: "1.1.76"
success: True

I ran the debug version, and here's what it outputs:

[ INFO] [1643670331.244137073]: !!! Publishing to topic device/status
[ INFO] [1643670331.247179288]: !!! Topic on resulting publisher /device/status
[ INFO] [1643670331.249913319]: Publishing IMU data.
[ INFO] [1643670331.250010376]: !!! Publishing to topic imu/data
[ INFO] [1643670331.252276988]: !!! Topic on resulting publisher /imu/data
[ INFO] [1643670331.252392506]: Publishing Magnetometer data.
[ INFO] [1643670331.252472728]: !!! Publishing to topic mag
[ INFO] [1643670331.254612985]: !!! Topic on resulting publisher /mag
[ INFO] [1643670331.254754501]: Publishing Filter data.
[ INFO] [1643670331.254836198]: !!! Publishing to topic nav/odom
[ INFO] [1643670331.257585934]: !!! Topic on resulting publisher /nav/odom
[ INFO] [1643670331.257738298]: !!! Publishing to topic nav/status
[ INFO] [1643670331.260088652]: !!! Topic on resulting publisher /nav/status
[ INFO] [1643670331.260196322]: !!! Publishing to topic nav/heading
[ INFO] [1643670331.261839040]: !!! Topic on resulting publisher /nav/heading
[ INFO] [1643670331.261938670]: !!! Publishing to topic nav/heading_state
[ INFO] [1643670331.263847957]: !!! Topic on resulting publisher /nav/heading_state
[ INFO] [1643670331.263982788]: !!! Publishing to topic nav/filtered_imu/data
[ INFO] [1643670331.266366695]: !!! Topic on resulting publisher /nav/filtered_imu/data
[ INFO] [1643670331.266483888]: !!! Publishing to topic nav/dual_antenna_status
[ INFO] [1643670331.268600071]: !!! Topic on resulting publisher /nav/dual_antenna_status
[ INFO] [1643670331.304682044]: Setting spin rate to <1000.000000> hz
[ INFO] [1643670331.307122566]: Resuming the device data streams
[ INFO] [1643670331.313338708]: Starting Data Parsing
[ INFO] [1643670331.316450799]: ??? Publishing filtered IMU
[ INFO] [1643670331.316509731]: ??? Published filtered IMU
[ INFO] [1643670331.316728827]: ??? Publishing filtered IMU
[ INFO] [1643670331.316754518]: ??? Published filtered IMU
[ INFO] [1643670331.316835193]: ??? Publishing filtered IMU
[ INFO] [1643670331.316862373]: ??? Published filtered IMU
[ INFO] [1643670331.316958202]: ??? Publishing filtered IMU

and so-on forever.

dimensional-difficulties, 2022-01-31:
So as I said in my previous reply, the code run from this debug/filtered_imu branch of the microstrain_inertial_driver_common repo works.  So perhaps this is a version control issue?  Running `git status` in the microstrain_inertial folder (the one I had before cloning the debug version) yields 
`On branch ros`     
`Your branch is up to date with 'origin/ros'.`.  
That's the branch I should be on, right?

robbiefish, 2022-02-01:
Hmmmm this is a real head scratcher. Those differences look mostly accurate for that branch between the branches you mentioned. Here is the diff according to [github](https://github.com/LORD-MicroStrain/microstrain_inertial_driver_common/compare/debug/filtered_imu..78eca64) which at a glance appears to match up with your diff.

You are correct that the ros branch is what you should be on when using ROS. In theory, the 3DM-GX5-25 and 3DM-GX5-AHRS should be the same (I think), but I will try to get my hands on one of those to test this out ASAP.

In the meantime, you mentioned a few comments ago that you
> cloned, installed, and ran the driver again and saw the same result.

A couple questions about that:
1. Do you know the commands you used to clone the repo during this process?
2. Did you do all of your testing previous to the testing on the `debug/filtered_imu` branch on the `ros` branch or on the `2.4.1` tag?
3. When you tested on ROS2, did you clone the repo a second time, or just switch to the `ros2` branch?
4. Have you tried installing the driver from the buildfarm? The latest on the buildfarm should also be `2.4.1`
    ```
    sudo apt-get update && sudo apt-get install ros-noetic-microstrain-inertial-driver
    ```

robbiefish, 2022-02-01:
No worries at all, while it would be preferable if everyone cloned the repo with `--recursive`, I realize it is an easy thing to miss, which is why the [`CMakeLists.txt`](https://github.com/LORD-MicroStrain/microstrain_inertial/blob/ros/microstrain_inertial_driver/CMakeLists.txt#L11) should (in theory) check out the submodules if they were not checked out on clone.

If I could bother you for a few more pieces of information to try to prevent this in the future:
1. What version of git is installed on your computer? `git --version`
2. Can you try cloning the repo again **without** the `--recursive` tag, then going into the `microstrain_inertial` directory and run `git submodule update --recursive --init` and see if the issue is still present when you rebuild?

If not, no worries, just please let me know so I can close the issue

dimensional-difficulties, 2022-02-01:
Certainly, I owe you at least that much.

My Git version is 2.25.1

If I install the driver without the recursive tag and then run `git submodule update --recursive --init`, the /nav/filtered_imu/data messages ARE published.

robbiefish, 2022-02-01:
Huh that is still very puzzling as in theory that is exactly what the `CMakeLists.txt` file should be doing. Also I have the same exact git version, so again very confusing.

Either way, thank you very much for the help. I am going to close this issue as it appears resolved, but if this issue pops up again, or you think of any additional information, please feel free to reopen

closed 2022-02-01 by robbiefish