LORD-MicroStrain/microstrain_inertial #122: cannot start imu driver node

ozangerger, 2022-02-04:
Hi, 

We have recently purchased an imu model  3DM-GQ7. I'm trying to setup an environment with ros driver. It worked only for the first time where I was able to echo imu topics and now it is not working anymore.

env: native linux 18.04, usb-to-rs232 cable, ros-melodic

What I did so far:

set the baudrate to 115200,removed startup config, so it is always idle when i launch driver node, using sensor connect
identified the serial device port
after launching the node I can see constant led light turning into yellow-green blinking light
I get the communication error at the end

you can find the launch logs below:

started core service [/rosout]
process[microstrain_inertial_driver-2]: started with pid [4859]
[ INFO] [1643988018.469484976]: Using MSCL Version: 63.1.0
[ INFO] [1643988018.473192030]: Attempting to open serial port </dev/ttyUSB0> at <115200>
[ INFO] [1643988018.474888046]: Setting to Idle: Stopping data streams and/or waking from sleep
[ WARN] [1643988018.725377618]: It looks like the device is streaming. Waiting a second before setting to idle again
[ INFO] [1643988019.725715841]: Setting to Idle: Stopping data streams and/or waking from sleep
[ INFO] [1643988019.785494194]: 
      #######################
      Model Name:    3DM-GQ7
      #######################
    
[ INFO] [1643988019.856752111]: Setting IMU data to stream at 100 hz
[ INFO] [1643988019.878973613]: Setting Declination Source
[ INFO] [1643988019.926913010]: Setting GNSS0 data to stream at 2 hz
[ INFO] [1643988020.040903613]: Setting GNSS1 antenna offset to [0.000000, -0.700000, -1.000000]
[ INFO] [1643988020.213147183]: Setting GNSS1 data to stream at 2 hz
[ INFO] [1643988020.385999569]: Setting GNSS2 antenna offset to [0.000000, 0.700000, -1.000000]
[ INFO] [1643988020.860155234]: Setting RTK dongle enable to 0
terminate called after throwing an instance of 'mscl::Error_Communication'
  what():  Failed to communicate with the device.
[microstrain_inertial_driver-2] process has died [pid 4859, exit code -6, cmd /home/ozan/microstraim_imu_driver/devel/lib/microstrain_inertial_driver/microstrain_inertial_driver_node __name:=microstrain_inertial_driver __log:=/home/ozan/.ros/log/f5b142de-85cd-11ec-b3d4-287fcfb0b300/microstrain_inertial_driver-2.log].
log file: /home/ozan/.ros/log/f5b142de-85cd-11ec-b3d4-287fcfb0b300/microstrain_inertial_driver-2*.log

Best regards,
Ozan

robbiefish, 2022-02-04:
Hi @ozangerger,

A few questions:
1. You noted that it worked the first time. Between the first time and subsequent runs did you change any launch parameters?
2. Your logs also appear to be missing the serial number of your device which should be printed right after the model name. Was this omitted on purpose or was it just not printed?

@nathanmillerparker Do you have any insights into this?

ozangerger, 2022-02-04:
I have changed baudrate from 460800 to 115200. I think it is not able to publish at low baudrate due to number of messages. 

Is there a way to remove some signals from configuration similar to the one in sensor connect? I'm asking this since I've checked the sensor publish configuration from sensor connect and it seems that all signals are added to the list. I only need few messages at the beginning (orientation, gyro, accel, gnss1 and gnss2 positions, no covariance or anything else) maybe you can help me on that. 

I removed the serial number on purpose but I can share if you really need it.

robbiefish, 2022-02-04:
Okay, I do not need the serial number, I just wanted to check if that was another issue.

With the way the driver is architect ed right now it is possible to remove some signals, but only in groups unlike SensorConnect which allows more fine-grain control. We are planning to add more fine-grain support in the future, see #121, but still need to work out what that will look like.

If all you want is raw IMU and GNSS data, you could set [`publish_filter`](https://github.com/LORD-MicroStrain/microstrain_inertial_driver_common/blob/main/config/params.yml#L144) to False which would disable all filter output. But aside from that, you will not be able to trim the data down more than that, while still keeping the message you need without modifying the source. For an example of someone modifying the source to remove some messages they did not want see #73.

So just to clarify, if you set baud rate back to 460800 does the device work properly, and you do not get the errors, or do you still see the error in the log at that higher baudrate?

ozangerger, 2022-02-07:
In fact, I would need filtered output rather than raw measurement for the beginning. Is it possible to output only filtered data? By the way, increasing baudrate back to 460800 seem to be working so far. 

As you've stated, we definitely need a fine-granular solution as in sensorconnect. I hope you can prioritize it, as our devices has lower baudrates, so the information we can send thru serial with high frequency sampling is quite limited.

robbiefish, 2022-02-07:
Understood, thanks for the feedback. It is definitely high on our list.

Yes, you can definitely configure the node to only publish filtered data and not raw data. You can do this by setting [`publish_filter`](https://github.com/LORD-MicroStrain/microstrain_inertial_driver_common/blob/main/config/params.yml#L144) to True, and then setting [`publish_imu`](https://github.com/LORD-MicroStrain/microstrain_inertial_driver_common/blob/main/config/params.yml#L94), [`publish_gnss1`](https://github.com/LORD-MicroStrain/microstrain_inertial_driver_common/blob/main/config/params.yml#L111), [`publish_gnss2`](https://github.com/LORD-MicroStrain/microstrain_inertial_driver_common/blob/main/config/params.yml#L121) to False

robbiefish, 2022-02-07:
Actually, @nathanmillerparker just reminded me that you can actually achieve the behavior you are looking for by configuring the device via SensorConnect to stream only the data you are looking for, and then turn [`device_setup`](https://github.com/LORD-MicroStrain/microstrain_inertial_driver_common/blob/main/config/params.yml#L48) to False.

This will cause the driver to run and publish whatever data you have setup in SensorConnect, but the driver will not configure the device at all, so if you go this route, make sure you have configured everything via SensorConnect including antenna offsets, etc.

Also in this mode, the params I previously called out will no longer configure the device, but will control whether the data is published on the ROS network. So make sure you still keep `publish_filter` set to True.

chedanix, 2022-02-07:
I had the same exact same error. Thanks to this thread, I got it to work after decreasing the data_rate frequency. I will increase my baud rate when it's needed.

ozangerger, 2022-02-08:
I guess in the light of the information you provided, we can close the ticket for now. 

Just few more updates from yesterday:

If I add desired signals and enable sampling from sensor connect interface, ros driver is not able to set device status to idle and therefore cannot connect to device. So, it doesn't work.

If I set publish_imu to false, filtered imu signals are also published as all zeros (linear accel and body rates). In short, it is not working as intended.

ozangerger, 2022-02-08:
The baud rate was 460k but then reverted back to 115k for some reason (And I did save setting to NVM). The config was essentially generated by ros driver. I think 460k should be more than enough but we always need to make sure that we are able to connect the device using sensor connect. However, in most cases it is not practical due to a need for two seperate devices using windows and linux.

robbiefish, 2022-02-17:
@ozangerger, I am not sure that this will solve all of your problems, but more fine grain control of data rates has been implemented in the following PRs: [ROS](https://github.com/LORD-MicroStrain/microstrain_inertial/pull/131), [ROS2](https://github.com/LORD-MicroStrain/microstrain_inertial/pull/132), with more information on exactly what has changed in the [submodule PR](https://github.com/LORD-MicroStrain/microstrain_inertial_driver_common/pull/15).

If possible, could you try running on [feature/ros_different_data_rates](https://github.com/robbiefish/microstrain_inertial/tree/feature/ros_different_data_rates) and let me know if this solves your problem, or if you run into any problems?

closed 2022-08-28 by github-actions[bot]