LORD-MicroStrain/microstrain_inertial #126: acceleration / gyro offset parameters needed in param.yaml file

ozangerger, 2022-02-10:
Right now, the only way to add offset values to the sensor is through sensor cloud interface. Is it possible to add this parameters to the yaml file as parameters? It will be easier to tune these offset values while working with the ros environment.

device:  3DM-GQ7

Regards,
Ozan

robbiefish, 2022-02-10:
Hi @ozangerger, although we do not allow accel and gyro offsets to be configured via the params file, you can configure them via ROS services. If you want to manually configure acceleration offsets, and gyro offsets you can do that by calling the `set_gyro_bias`, and `set_accel_bias` services with your desired offsets.

An easier method of setting gyro bias would be to call the `capture_gyro_bias` service and not move the device for 10 seconds.

More documentation on available services can be found on [our wiki page](http://wiki.ros.org/microstrain_inertial_driver).

chedanix, 2022-02-21:
Hi @robbiefish, if my GPS is not installed at the center of my robot, I will need to set the gyro bias and the accel bias, correct?  Which frame (sensor or vehicle) should I use for these biases?

How about the sensor2vehicle_offset? Do I need to set that as well? (I have more questions about this offset. I may open up another issue to track it.)

closed 2022-03-27 by github-actions[bot]