LORD-MicroStrain/microstrain_inertial #133: NatSatFix for the vehicle center (or at least the sensor origin)

chedanix, 2022-02-21:
**Is your feature request related to a problem? Please describe.**
Currently, there are only NatSatFix topics for each individual antenna. Since the antennas are mounted far away from my robot center, I cannot directly use these lat-long values.

**Describe the solution you'd like**
Ideally, I'd like to get the NatSatFix for the vehicle center, which is defined by the user via the sensor-to-vehicle-offset ROS service. If that's not possible, I'd like to get the NatSatFix for the GPS unit.

**Alternative ways***
I thought about writing the code to "offset" the lat-long values from one of the antenna. But that wouldn't take advantage of having two antennas, which can provide higher accuracy after filtering.

robbiefish, 2022-02-22:
Hi @chedanix, it sounds like you are looking for the [filtered LLH position](https://s3.amazonaws.com/files.microstrain.com/GQ7+User+Manual/external_content/dcp/Data/filter_data/data/mip_field_filter_llh_pos.htm). The filtered LLH position takes into account the `set_sensor2vehicle_offset`, and `set_sensor2vehicle_rotation` when outputting position. Currently, the driver does not output the filtered LLH position in a NavSatFix message, but it is output on the `/nav/odom` topic as follows:

`odom.pose.pose.position.x` -> Lat (degrees)
`odom.pose.pose.position.y` -> Lon (degrees)
`odom.pose.pose.position.z` -> Alt (meters)

Please note that if you set `use_enu_frame` to True, the message will be output as follows:
`odom.pose.pose.position.x` -> Lon (degrees)
`odom.pose.pose.position.y` -> Lat (degrees)
`odom.pose.pose.position.z` -> Alt (meters)

We realize that this is a little confusing to populate an Odometry message with LLH position, and are looking at possible alternatives.

chedanix, 2022-02-22:
Hi @robbiefish, I am glad hear that the filtered LLH position is already published in /nav/odom.

Does the filtered LLH position gives the location of the "sensor origin" in the link below?
https://s3.amazonaws.com/files.microstrain.com/GQ7+User+Manual/user_manual_content/specifications/Mechanical.htm

My particular application requires the latitude and longitude of the vehicle frame. Can the TF transformation gives me this value directly? If not, I can write my own code to perform this operation.

robbiefish, 2022-02-22:
> Does the filtered LLH position gives the location of the "sensor origin" in the link below? https://s3.amazonaws.com/files.microstrain.com/GQ7+User+Manual/user_manual_content/specifications/Mechanical.htm

Yes, that is correct. 
 
> My particular application requires the latitude and longitude of the vehicle frame. Can the TF transformation gives me this value directly? If not, I can write my own code to perform this operation.

Unfortunately, the GQ7 does not have a concept of an offset to a different reference point on a vehicle. Thinking more about this, I am not sure if the TF can help you to transform the LLH to your vehicle frame. The most common use case for transforms in ROS (at least from what I understand) is to transform position data between [ROS frames](https://www.ros.org/reps/rep-0105.html) which are in meters.

Are you also interested in getting velocity or acceleration from the GQ7 at the vehicle frame? If so, you will also need to do some more complex transformations to that data.

Can I ask for more information on your specific usecase? A diagram or picture of how you have the sensor mounted on the robot would be helpful.

chedanix, 2022-02-23:
My existing robot layout only allows the physical GPS center to be located at the robot center. Therefore, the sensor origin will be roughly 3.5cm away from the robot center. At the moment, I believe this offset is tolerable, but I will need to do some tests to confirm.

I am interested in getting velocity from the GQ7 at the vehicle frame. But then again, the 3.5cm offset could be tolerable.

Yes of course, I can provide more info of my use case. Below is a marketing video of the robot. Please let me know if you are still unclear about the sensor mounting location. I can send a picture via email.
https://vimeo.com/638634436/1fa6d4e9cf

nathanmillerparker, 2022-02-24:
Hi @chedanix,

Thank you for the video, this gives us more context.  

For the velocity, you would need to be spinning rather quickly for the 3.5cm lever arm to have an appreciable effect.  

For position, you have a few choices:

1. If your application allows, accept the additional position error.
2. Use the relative position topic, which reports position in the NED frame, then rotate the sensor2vehicle frame lever arm from the vehicle frame to the NED frame using the quaternion in the same topic, and then add it to the reported relative position.
3. Use a bit more complex math to do the same as 2, but in LLH.

Which one of these do you think you'd like to do?  Some additional details could be provided for 2 or 3 if needed.

chedanix, 2022-02-24:
Although I see the appeal in method 2, the customers will want the actual path traveled by the robot during a crash test. Therefore, we will provide them the data in the lat-long format, which are independent from a pre-defined reference point, unlike x-y positions. Let me know if this doesn't make sense.

I only have a rough idea on implementing method 3. It involves the Haversine formula with small angle approximation. I would like to learn about how you would implement this.

Thanks!

nathanmillerparker, 2022-02-24:
Here's some pseudocode I pulled together from some code I had sitting around.  The only thing missing is the quaternion transformation for a vector, but ROS has that.  Hope it helps.

#define EARTH_R  6378137.0         //m
#define EARTH_f  1.0/298.257223563 //dimensionless

double calculate_R_n(double lat)
{
 double sin_lat = sin(lat*M_PI/180.0);
 
 return EARTH_R/sqrt(1.0 - (2.0*EARTH_f - EARTH_f*EARTH_f)*sin_lat*sin_lat); 
}

double calculate_R_m(double lat, double r_n)
{
 double sin_lat = sin(lat*M_PI/180.0);
 
 return r_n*((1.0 - (2.0*EARTH_f - EARTH_f*EARTH_f))/(1.0 - (2.0*EARTH_f - EARTH_f*EARTH_f)*sin_lat*sin_lat));  
}

double calculate_delta_lat_to_delta_north_factor(double r_m)
{
 return 1.0/atan(1.0/r_m); 
}

double calculate_delta_lon_to_delta_east_factor(double lat, double r_n)
{
 return 1.0/atan(1.0/(r_n*cos(lat*M_PI/180.0)));
}

double R_n = calculate_R_n(curr_latitude);
double R_m = calculate_R_m(curr_latitude, R_n);

sensor2vehicle_offset_ned = q_transform_vector(q_ned2body.inv(), sensor2vehicle_offset_body);
 
height = curr_height - sensor2vehicle_offset_ned[2];
lat    = (1.0/calculate_delta_lat_to_delta_north_factor(R_m))*(sensor2vehicle_offset_ned[0])*180.0/M_PI  + curr_latitude; 
lon    = (1.0/calculate_delta_lon_to_delta_east_factor(curr_latitude, R_n))*(sensor2vehicle_offset_ned[1])*180.0/M_PI  + curr_longitude; 
  
//Bound latitude
if(lat > 90.0)
{
 lat = 90.0 - (lat - 90.0); 
}
else if(lat < -90.0)
{
 lat = -90.0 - (lat + 90.0);     
}
  
//Bound longitude
if(lon > 180.0)
{
 lon -= 360.0; 
}
else if(lon < -180.0)
{
 lon += 360.0;     
}

closed 2022-03-27 by github-actions[bot]