LORD-MicroStrain/microstrain_inertial #134: Cannot find the topic "rtk/status"

chedanix, 2022-02-25:
**Describe the bug**
I would like to get the RTK status such as Differential Fix and Integer. The website below mentions "rtk/status" as a published topic in section 5.2, but I am not getting this topic.
http://wiki.ros.org/microstrain_inertial_driver

**To Reproduce**
Steps to reproduce the behavior:
1. roslaunch the microstrain.launch file
2. rostopic list
3. rtk/status isn't listed

**Expected behavior**
rtk/status should be listed with `rostopic list`

**Environment (please complete the following information):**
 - OS: Ubuntu 16.04
 - Architecture: arm
 - ROS Version: Kinetic
 - Sensor(s): 3DM-GQ7

**Modifications**
None

**Launch Parameters**
I tried to look for something like "publish_rtk_status" in the yaml file, but no luck

robbiefish, 2022-02-25:
Hi @chedanix,

Sorry about the confusion. Do you have [`rtk_donble_enable`](https://github.com/LORD-MicroStrain/microstrain_inertial_driver_common/blob/1fe1e809824fd2492c38d95ee9b8bc5808ab66b2/config/params.yml#L131) set to True in the yml file? That is the config option that controls whether or not RTK data will be published. Please note that we have two different versions of the RTK dongle and you may get data on different topics depending on the version you are using. If you are using a v1 dongle, you will get data on the topic `rtk/status_v1`, and if you are using a v2 dongle, you will get data on the topic `rtk/status`.

Also, since it looks like you are looking for whether the GQ7 is getting differential corrections/integer fix, that information is actually published for each GNSS antenna on the topics `gnss1/aiding_status` and `gnss2/aiding_status`. A summary of what is published on those topics can be found [here](http://docs.ros.org/en/api/microstrain_inertial_msgs/html/msg/GNSSAidingStatus.html). Those topics will be published if [`publish_filter`](https://github.com/LORD-MicroStrain/microstrain_inertial_driver_common/blob/1fe1e809824fd2492c38d95ee9b8bc5808ab66b2/config/params.yml#L144) is set to true

chedanix, 2022-02-25:
Thanks @robbiefish 

Obtaining Fix status separately from gnss1 and gnss2 works for me.

closed 2022-02-25 by chedanix

robbiefish, 2022-02-28:
@chedanix

I am glad that will work for you. @nathanmillerparker, pointed out a few important subtleties that I did not know about that you may want to know:

1. The information in the aiding status message isn’t directly from the receiver, but is actually calculated by the filter.  It has been grouped with the GNSS info as that seemed like the best place.
1. The RTK fix flag only needs to be asserted for 1 receiver for the filter to have an RTK fix, it does not require the both receivers have the flag asserted.
1. A receiver reporting differential corrections flag as 1, but RTK fix flag as 0 means it is in RTK float mode… which is about 10 cm of position accuracy.