LORD-MicroStrain/microstrain_inertial #136: Using 3DM-GX5-10 IMU unit in Ubunto 20.04 & ros2 foxy

kevinhan75, 2022-03-14:
Hello.
I have some problem when I use 3DM-GX5-10 IMU unit in Ubuntu 20.04 & ros2 foxy. (not GNSS/AHRS)
(I used a usb2serial (Bus 001 Device 021: ID 067b:2303 Prolific Technology, Inc. PL2303 Serial Port))

I installed 'sudo apt install ros-foxy-microstrain-inertial-driver', 'sudo apt install ros-foxy-microstrain-inertial-msgs', 'sudo apt install ros-foxy-microstrain-inertial-rqt', 'sudo apt install ros-foxy-microstrain-inertial-examples'.

When I executed like below,
Terminal 1 : ros2 launch microstrain_inertial_driver microstrain_launch.py port:=/dev/ttyUSB0
Terminal 2 : ros2 lifecycle set /microstrain_inertial_driver configure

I got the error like that
-------------------------------------------------------------------
kevinhan@kevinhan-ubuntu:~$ ros2 launch microstrain_inertial_driver microstrain_launch.py port:=/dev/ttyUSB0

[INFO] [launch]: All log files can be found below /home/kevinhan/.ros/log/2022-03-14-14-13-07-199871-kevinhan-ubuntu-133864

[INFO] [launch]: Default logging verbosity is set to INFO

[INFO] [microstrain_inertial_driver_node-1]: process started with pid [133876]

[microstrain_inertial_driver_node-1] [INFO]: Using MSCL Version: 63.1.0

[microstrain_inertial_driver_node-1] [INFO]: Attempting to open serial port </dev/ttyUSB0> at <115200>

[microstrain_inertial_driver_node-1] [INFO]: Setting to Idle: Stopping data streams and/or waking from sleep

[microstrain_inertial_driver_node-1] [INFO]: 

[microstrain_inertial_driver_node-1]       #######################

[microstrain_inertial_driver_node-1]       Model Name:    3DM-GX5-10

[microstrain_inertial_driver_node-1]       Serial Number: 6255.113743

[microstrain_inertial_driver_node-1]       #######################

[microstrain_inertial_driver_node-1]     

[microstrain_inertial_driver_node-1] [INFO]: Setting IMU data to stream at 100 hz

[microstrain_inertial_driver_node-1] [INFO]: Note: Device does not support the declination source command.

[microstrain_inertial_driver_node-1] [ERROR]: **Failed to set sensor2vehicle frame transformation with euler angles!

[microstrain_inertial_driver_node-1] [ERROR]: Failed to read configuration for node

[microstrain_inertial_driver_node-1] [ERROR]: Failed to configure node base
-------------------------------------------------------------------

How can I solve this issue?

robbiefish, 2022-03-15:
Hi @kevinhan75

You can resolve this issue by:
1. Creating a `params.yml` file with the following content:
    ```yml
    microstrain_inertial_driver:
      ros__parameters:
        filter_sensor2vehicle_frame_selector : 0
    ```
1. Launch the node with:
    ```bash
    ros2 launch microstrain_inertial_driver microstrain_launch.py port:=/dev/ttyUSB0 params_file:=<Path to params file>/params.yml
    ```

Note that you could also add the port option (and all other parameters you may need to tune) to the `params.yml` and remove it from your launch command like so:
```yml
microstrain_inertial_driver:
  ros__parameters:
    port: /dev/ttyUSB0
    filter_sensor2vehicle_frame_selector : 0
```
For a more comprehensive list of parameters see the [params.yml](https://github.com/LORD-MicroStrain/microstrain_inertial_driver_common/blob/main/config/params.yml) on Github

kevinhan75, 2022-03-15:
@robbiefish
Thank you for help.
My issue is solved.

closed 2022-03-15 by kevinhan75